zigpy/zha-device-handlers #2452: [Device Support Request] Smoke detector TS0601 by _TZE200_m9skfctm

mistraller, 2023-07-02:
### Problem description

ZHA is only reporting one entity binary sensor "Smoke"

Battery status is missing, and zigbee2mqtt is reporting smoke PPM as well (but not clear if that is valid data)

Paired the device several times but only one binary sensor shown.

### Solution description

battery entity and smoke ppm

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0500",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_m9skfctm",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_smoke.TuyaSmokeDetector0601"
}
</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.6.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "6.1.34",
    "supervisor": "2023.06.4",
    "host_os": "Home Assistant OS 10.3",
    "docker_version": "23.0.6",
    "chassis": "vm",
[... 281 lines cut ...]
    }
  }
}

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

doreks, 2023-10-23:
How is status of this issue?

FlorianLaunay, 2024-01-11:
I am also interested by this feature. Is anyone have a working custom quirk with battery level reporting?

closed 2024-11-01 by github-actions[bot]

hdcoffeeboy, 2024-11-02:
Still hoping for a resolution on this. Battery level reporting would be excellent.

idumzaes, 2024-11-11:
The battery entity isn't working on this quirk. Only the Smoke entity works, which is a win already.
Has anyone been able to implement the battery into quirk yet??
Thanks in advance!!

FlorianLaunay, 2025-04-04:
Issue was closed automatically and we cannot reopen. I guess we must open a new one.