zigpy/zha-device-handlers #2457: [Device Support Request] _TZE204_dtzziy1e TS0601 24GHz Millimeter Wave Tuya Zigbee Human Presence Sensor with switch (relay)

Nelevit, 2023-07-04:
### Problem description

When adding to ZHA only device exists, no entities.

### Solution description

Device is added with all related entities (sensor, switch). If possible add sensitivity controls.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>
[image]
[image]
[image]
[image]
[image]
[image]
</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_dtzziy1e",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.6.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Kiev",
    "os_name": "Linux",
    "os_version": "6.1.34",
    "supervisor": "2023.06.4",
    "host_os": "Home Assistant OS 10.3",
[... 286 lines cut ...]

[0x4EA2:1:0xef00] Unknown cluster command 2 b'\x00G\t\x02\x00\x04\x00\x00\x00\x00'
[0x4EA2:1:0xef00] Unknown cluster command 2 b'\x00H\x01\x04\x00\x01\x01'
[0x4EA2:1:0xef00] Unknown cluster command 2 b'\x00I\t\x02\x00\x04\x00\x00\x01o'
[0x4EA2:1:0xef00] Unknown cluster command 2 b'\x00J\t\x02\x00\x04\x00\x00\x00\x00'
[0x4EA2:1:0xef00] Unknown cluster command 2 b'\x00K\x01\x04\x00\x01\x00'
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Successfully loaded a custom ts0601_motion.py quirk (with additional MODELS INFO in MmwRadarMotion(CustomDevice) class) but it exposes only one binary sensor and relay isn't clicking when detected/clear.

Also as you can see on the board photo it has a IR led. Why is it there? Would it be possible to use it?

javicalle, 2023-07-04:
Can you attach the logs with the quirk applied? Logs from quirk would be more descriptive.
Also, with the quirk applied, catch the logs from the device powerup (when wires are connected). Tuya devices usually reports its status and give to us some DPs information.
Any link to vendor site or device capabilities?

Nelevit, 2023-07-05:
@javicalle Thanks a lot for your reply!

Sorry it takes me so long to post those logs. I can't reproduce loading the quirk now :/

### Logs

<details><summary>Logs</summary>

```python
Logger: zhaquirks
Source: custom_zha_quirks/ts0601_motion.py:6
First occurred: 3:06:31 PM (1 occurrences)
Last logged: 3:06:31 PM

Unexpected exception importing custom quirk 'ts0601_motion'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 454, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 605, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1120, in load_module
  File "<frozen importlib._bootstrap_external>", line 945, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 721, in _load
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_motion.py", line 6, in <module>
    from zigpy.profiles import zgp, zha
ImportError: cannot import name 'zgp' from 'zigpy.profiles' (/usr/local/lib/python3.11/site-packages/zigpy/profiles/__init__.py)
```

</details>

Would you please help to figure it out?

Here is a [link to device page on AliExpress](https://a.aliexpress.com/_mMmxk40)

Thanks in advance!
Nestor

javicalle, 2023-07-05:
Take the `ts0601_motion.py` version from here:
https://github.com/zigpy/zha-device-handlers/raw/release/0.0.74/zhaquirks/tuya/ts0601_motion.py

Add your model, save changes and restart HA.

Check here to enable the debug logs:
* https://www.home-assistant.io/integrations/zha/#debug-logging

Nelevit, 2023-07-05:
@javicalle Thanks for your help :)

Its alive! :D

I've experimented with it and 2 classes were able to load:

- MmwRadarMotion

Single entity - human presence sensor which works ok

<details><summary>MmwRadarMotion_NAS-PD07 Diagnostics</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.7.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Kiev",
    "os_name": "Linux",
    "os_version": "6.1.34",
    "supervisor": "2023.07.1",
    "host_os": "Home Assistant OS 10.3",
    "docker_version": "23.0.6",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "scheduler": {
      "version": "v0.0.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
[... 649 lines cut ...]
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

[NeoMotion_NAS-PD07.txt](https://github.com/zigpy/zha-device-handlers/files/11962024/NeoMotion_NAS-PD07.txt)

txt files files contain logs while resetting.

Hope it helps

JeroenVanOort, 2023-08-31:
I think the label just concerns adding `_TZE204_dtzziy1e` to the existing MmwRadarMotion cluster. That's what I did and it does work for the basic function of detecting occupancy and illuminance, but I can't control the relay or the parameters that should make the relay switch.

Does anybody know where to get started on this? There is some information available from another project: https://github.com/Koenkk/zigbee-herdsman-converters/issues/5930

evgenln, 2023-09-13:
@JeroenVanOort I did quirk for it's brother _TZE204_clrdrnya, it's same options device. But it probably will not really usable for you as I running zha as component and exposed all options/target distance sensor it it bundles. Anyway it contains all datapoints so at least you will be able to change values via Mange ZigBee device.

JeroenVanOort, 2023-09-15:
Thank you very much!

Have you been able to get the relay working? For me, the relay never switches, neither automatic nor manual. Maybe my unit is just broken, but I don't know.

evgenln, 2023-09-15:
Yes, it is working. 

I have the full UI due to custom changes I did in ZHA, with default ZHA you will not see controls 
but you can try to do it manually from the device page via the Manage Zigbee device.

Open Manage Zigbee device and select MmwRadarV3ManufCluster
Select breaker_mode attribute and write 0 to it (it will set the device to Standart mode with the ability to manipulate relay)
then select the breaker_status attribute and write 0 to set it off or 1 to set it on (polarity depends on one more attribute breaker_polarity but I am not sure if it is implemented in your device)

RiRomain, 2024-01-02:
I tried with the attached zip file by replacing the device info, but I only have occupancy showed, does someone know how to correct it and have the .py file that fit the _TZE204_dtzziy1e ?

[ts0601_presence.py.txt](https://github.com/zigpy/zha-device-handlers/files/13810279/ts0601_presence.py.txt)

bouboun59, 2024-01-09:
> Working fine with the following:
> 
> ```
> import math
> from typing import Dict
> 
> from zigpy.profiles import zgp, zha
> from zigpy.quirks import CustomDevice
> import zigpy.types as t
> from zigpy.zcl.clusters.general import (
>     AnalogInput,
>     AnalogOutput,
>     Basic,
>     GreenPowerProxy,
>     Groups,
>     Ota,
>     Scenes,
>     Time,
> )
> from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing
> 
> from zhaquirks.const import (
>     DEVICE_TYPE,
>     ENDPOINTS,
>     INPUT_CLUSTERS,
>     MODELS_INFO,
>     OUTPUT_CLUSTERS,
>     PROFILE_ID,
> )
> from zhaquirks.tuya import NoManufacturerCluster, TuyaLocalCluster, TuyaNewManufCluster
> from zhaquirks.tuya.mcu import (
>     DPToAttributeMapping,
>     TuyaAttributesCluster,
>     TuyaMCUCluster,
> )
> 
> ZONE_TYPE = 0x0001
> 
> 
> class TuyaMmwRadarSelfTest(t.enum8):
>     """Mmw radar self test values."""
> 
>     TESTING = 0
>     TEST_SUCCESS = 1
>     TEST_FAILURE = 2
>     OTHER = 3
>     COMM_FAULT = 4
>     RADAR_FAULT = 5
> 
> 
> class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
>     """Tuya local OccupancySensing cluster."""
> 
> 
> class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
>     """Tuya local AnalogInput cluster."""
> 
> 
> class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
>     """Tuya local IlluminanceMeasurement cluster."""
> 
> 
> class TuyaMmwRadarSensitivity(TuyaAttributesCluster, AnalogOutput):
>     """AnalogOutput cluster for sensitivity."""
> 
>     def __init__(self, *args, **kwargs):
>         """Init."""
>         super().__init__(*args, **kwargs)
>         self._update_attribute(self.attributes_by_name["description"].id, "Sensitivity")
>         self._update_attribute(self.attributes_by_name["min_present_value"].id, 1)
[... 478 lines cut ...]
>             5: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
>                 INPUT_CLUSTERS: [
>                     TuyaMmwRadarFadingTime,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>             242: {
>                 PROFILE_ID: zgp.PROFILE_ID,
>                 DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         }
>     }
>     ```
> ```

Thanks a lot, it's work fine with my TS060 _TZE204_clrdrnya, but I don't understand why i don't see the radar target  distance in my sensors
[image]
so i see in the advanced parameters
[image]

could you explain why ? And you, you can see the radar target distance

tommetje, 2024-01-16:
> I tried with the attached zip file by replacing the device info, but I only have occupancy showed, does someone know how to correct it and have the .py file that fit the _TZE204_dtzziy1e ?
> 
> [ts0601_presence.py.txt](https://github.com/zigpy/zha-device-handlers/files/13810279/ts0601_presence.py.txt)
> 

@RiRomain Do you see more than Illuminance and Occupancy in ZHA?

RiRomain, 2024-01-16:
I only see illuminance and occupancy, but didn't think that the sensor can handle anything more (appart from the settings).
[image]

tommetje, 2024-01-17:
Wow nice, GUI controls woohoo! :)
I tried it and it works! 

My detectors are picking up movements of neighbors so for me the entry controls are essential.  Currently, Entry Sensitivity and Entry Distance Indentation are sadly missing in the quirk update. 
I currently have no time nor knowledge (yet :) ) to adjust your quirk, so I will revert back and keep an eye out for possible updates in this thread. 

Thanks for sharing your quirk @RiRomain !

mariovw, 2024-02-21:
Hi @evgenln ,

I have _TZE204_clrdrnya. I load the quirk [ts0601_presence.py.txt](https://github.com/zigpy/zha-device-handlers/files/13810279/ts0601_presence.py.txt)
 but I only get the presence and iluminance but not the Controls to adjust the parametres.

Do you know what could be the issue?

Many thanks!

tommetje, 2024-02-21:
Sometimes it helps to reload the ZHA integration, just in case you have not tried this. 
You can also try to remove the devices and add them again. I had the same problem and this fixed things for me.
[image]

mariovw, 2024-02-21:
thanks @tommetje for your quick answer.

Unfortunally I followed your instructions several time with no results. Any other idea?

Many thanks!

tommetje, 2024-02-21:
> thanks @tommetje for your quick answer.
> 
> Unfortunally I followed your instructions several times with no results. Any other idea?
> 
> Many thanks!

I updated the existing quirk, after this I did not see the changes. 

From what I remember, I also tried removing removing the affected devices + removing the quirk folder + HA restart + ZHA reload. After this I added the folder of the updated quirk and gave it a different name, again HA restart + adding devices again + ZHA reload (if needed) . 

Not sure if all the steps are needed but this is what I tried. Eventually I got the updated quirk working.

mariovw, 2024-02-22:
thks @tommetje, 

I'll try to play more the weekend!

Best,

matsjrosenberg, 2024-03-07:
Hello, I have a _TZE204_sbyx0lm6 (the 5.8G version as opposed to the 2.4G) and have tried all quirks above, changing the device specs in the code, and only randomly gotten only the illuminance and presence to work. I am not a coder so can't unfortunately contribute to this, but would really appreciate if somebody could continue developing this quirk further. Many thanks!

frawau, 2024-03-16:
I have a _TZE204_clrdrnya device and with @RiRomain  python script, I get the occupancy to work fine ... Yeah! Thank you.

But

Luminance is not working, and neither Ota nor Time are working. Then there is the the question of the relay.

According to [@LuisAlbertoFB in this thead](https://github.com/Koenkk/zigbee2mqtt/issues/18677) it is possible to get it all working.  I tried a few things, but so far no luck... Anyone has a clue on how to port this?

OakfieldDrive, 2024-04-06:
Hi all,
I got it to work on _TZE204_clrdrnya using the quirk shared in the zip file above. however, the illuminance figures are not within the expected range. e.g, I am currently in a room, in a cloudy day; would expect to have something between 15 to 30 lx but instead, it shows 1,522lx. Anyone else has the same problem?

macf0x, 2024-05-05:
Was there any luck getting the relay to work?

cHeeSaW, 2024-05-21:
I got it working for _TZE204_dtzziy1e with the code from @RiRomain's comment. The illuminance value does not show up at first, i have to restart HA every time i add a new device of this type, then it shows up.
Anyone working on the missing values and relay?

macf0x, 2024-06-30:
Same. I get following controls and Occupancy Sensor. No lux.

Detection Delay
Fading Time
Max Range
Min Range
Sensitivity

Firmware Update is also visible

Relay isn't exposed and doesn't align to Occupancy. That all I really need.

nnae06, 2024-08-13:
I got everything working except the relay. In my case, I don't use the relay function.

Can this be added to ZHA. This ZHA quirks prevent me from updating to 2024.08 ->. Many of my automations at home use these devices to function.

"""Tuya mmw radar occupancy sensor."""

import math
from typing import Dict

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import NoManufacturerCluster, TuyaLocalCluster, TuyaNewManufCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

ZONE_TYPE = 0x0001

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""

    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaMmwRadarSensitivity(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for sensitivity."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self._update_attribute(self.attributes_by_name["description"].id, "Sensitivity")
        self._update_attribute(self.attributes_by_name["min_present_value"].id, 1)
        self._update_attribute(self.attributes_by_name["max_present_value"].id, 9)
        self._update_attribute(self.attributes_by_name["resolution"].id, 1)
[... 467 lines cut ...]
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

nnae06, 2024-08-13:
[image]

matsjrosenberg, 2024-08-14:
Thank you for this @nnae06 . Tried to copy paste this into a new customised quirk, and then reinstalled the device. It still insists on picking up the original zhaquirks.tuya.ts0601_motion.MmwRadarMotionGPP instead of the customised one. Can anybody help explaining how I make the custom quirk override the default one? Any specific file naming convention to be followed? The hardware specs correspond to what is in the customised quirk so this should not be the issue. And all my other customised device quirks work well.

nnae06, 2024-08-14:
I will post a finished file later in the day. Maybe can help.

nnae06, 2024-08-14:
[ts0601_radar.py.zip](https://github.com/user-attachments/files/16614977/ts0601_radar.py.zip)

matsjrosenberg, 2024-08-17:
[ts0601_radar.zip](https://github.com/user-attachments/files/16644017/ts0601_radar.zip)
Added support for TS0601 by _TZE204_sbyx0lm6. Occupancy and Lux but no relay.

macf0x, 2024-09-26:
Attempted to update latest quirk with the full DP's from zigbee2mqtt herdsmen ([ ](https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt/blob/68468dc630f19fdbea826538eddfaeafd964a1be/mtg035_radar_240701.js#L24))

More experimenting is required as I don't fully understand quirks, but posting this as it might help someone else.

```
 tuyaDatapoints: [
                [1, 'presence', tuya.valueConverter.trueFalse1],
                [2, 'radar_sensitivity', tuya.valueConverter.raw],
                [3, 'shield_range', tuya.valueConverter.divideBy100],
                [4, 'detection_range', tuya.valueConverter.divideBy100],
                [6, 'equipment_status', tuya.valueConverter.raw],
                [9, 'target_distance', tuya.valueConverter.divideBy100],
                [101, 'entry_filter_time', tuya.valueConverter.divideBy10],
                [102, 'departure_delay', tuya.valueConverter.raw],
                [103, 'cline', tuya.valueConverter.raw],
                [104, 'illuminance_lux', tuya.valueConverter.divideBy10],
                [105, 'entry_sensitivity', tuya.valueConverter.raw],
                [106, 'entry_distance_indentation', tuya.valueConverter.divideBy100],
                [107, 'breaker_mode', tuya.valueConverterBasic.lookup({'standard': tuya.enum(0), 'local': tuya.enum(1)})],
                [108, 'breaker_status', tuya.valueConverterBasic.lookup({'OFF': tuya.enum(0), 'ON': tuya.enum(1)})],
                [109, 'status_indication', tuya.valueConverterBasic.lookup({'OFF': tuya.enum(0), 'ON': tuya.enum(1)})],
                [110, 'illuminance_threshold', tuya.valueConverter.divideBy10],
                [111, 'breaker_polarity', tuya.valueConverterBasic.lookup({'NC': tuya.enum(0), 'NO': tuya.enum(1)})],
                [112, 'block_time', tuya.valueConverter.divideBy10],
                [113, 'parameter_setting_result', tuya.valueConverter.raw],
                [114, 'factory_parameters', tuya.valueConverter.raw],
                [115, 'sensor', tuya.valueConverterBasic.lookup({
                    'on': tuya.enum(0), 'off': tuya.enum(1), 'occupied': tuya.enum(2), 'unoccupied': tuya.enum(3)})],
                [116, 'distance_report', tuya.valueConverterBasic.lookup({
                    'off': tuya.enum(0), 
                    'on': tuya.enum(1)
```
[ts0601_radar_new.py.txt](https://github.com/user-attachments/files/17150541/ts0601_radar_new.py.txt)

dala318, 2024-10-01:
> [ts0601_radar_new.py.txt](https://github.com/user-attachments/files/17150541/ts0601_radar_new.py.txt)

Thanks a lot! Could you please add a TS0601 "_TZE204_iaeejhvf" to TuyaMmwRadarOccupancyVariant1GPP. I have added it locally and it show up as I would think is supported by the current quirk.
[image]

Looking forward to see if the breaker part of device can get solved as well.
[zha-10bd056a4de4cbece683ed97f85638a7-_TZE204_iaeejhvf TS0601-dac28b9c161b2bebc10f7ebf01608fb2.json](https://github.com/user-attachments/files/17202120/zha-10bd056a4de4cbece683ed97f85638a7-_TZE204_iaeejhvf.TS0601-dac28b9c161b2bebc10f7ebf01608fb2.json)

dala318, 2024-10-01:
One more improvement suggestion (I would love to contribute more actively but have yet not passed the threshold of understanding the quirks and how to debug/try/error/improve).

Currently the number sliders for the device settings show up as "Controls" in the device in Home Assistant, I think they would be better placed under "Configuration". The small benefit that I see of that is that they would then not show up in the default dashboard in Home Assistant. (You likely don't want to change those on a daily basis).

Yes, I can make them hidden to achieve the same so no major issue, but just a proposal :-)

closed 2025-01-16 by TheJulianJES