zigpy/zha-device-handlers #2459: [Device Support Request] Aquara Vibration Sensor - Tilt angles are missing

Emanuele-Spatola, 2023-07-05:
### Problem description

The sensor is correctly added to HA, and vibration sensor is correctly working, but the tilt angles are missing.
Those informations are available with Zigbee2mqqt.
Is it possible to have them in ZHA?

### Solution description

3 new sensors should be added:
tiltX, tiltY, tiltZ

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheJulianJES, 2023-07-05:
Listen for `zha_event` in the "developer tools", then shake the sensor.

Emanuele-Spatola, 2023-07-06:
Thanks @TheJulianJES.
I can see the tilt angles in the event data, but there are some strange things going on:
1. The device stops sending events after a few minutes. I need to remove it and pair it again to make it work again.
2. After only one day the battery level has gone from 100% to 0% (but the device still works)
3. The tilt angles are either 0, 45 or 90 degrees plus or minus 1: only values I've seen are 0,1,45,46,89,90.
this is behaviour is also confirmed here: https://www.reddit.com/r/homeassistant/comments/12j6r9x/comment/jg1qsul/
I appears that with zigbee2mqqt or deconz the tilt angles can have different values:
- https://community.home-assistant.io/t/using-xiaomi-aqara-vibration-sensor-for-controlling-rgb/364438
- https://www.reddit.com/r/homeassistant/comments/k3bitl/aqara_sensors_in_home_assistant_with_deconz_i_can/
- https://community.home-assistant.io/t/determine-window-state-tilted-open-with-one-sensor/137515/5

Should I rename this issue as a BUG?

slipstyle, 2023-08-22:
I can confirm the same behaviour for my Aqara Vibration Sensors and ZHA. 
ZHA and HA can see vibration is detected, but that is it... 

I can also see Vibration Strength, Current Orientation, Tilt and Drop events in the logbook, but I see no ability to see any tilt angles, the strength of the vibration, etc., none of those are available as sensor data in HA.

[image]

```javascript
event_type: zha_event
data:
  device_ieee: 00:15:8d:00:08:f3:4e:31
  unique_id: 00:15:8d:00:08:f3:4e:31:1:0x0500
  device_id: 35976d0dfeb5c514c041e44a07db6923
  endpoint_id: 1
  cluster_id: 1280
  command: current_orientation
  args:
    rawValueX: 64626
    rawValueY: 1
    rawValueZ: 228
    X: 90
    "Y": 0
    Z: 0
  params: {}
origin: LOCAL
time_fired: "2023-08-22T13:44:49.416719+00:00"
context:
  id: 01H8EQVAG8VA9C0HPY99QG8TYH
  parent_id: null
  user_id: null
```

```javascript
event_type: zha_event
data:
  device_ieee: 00:15:8d:00:08:f3:4e:31
  unique_id: 00:15:8d:00:08:f3:4e:31:1:0x0500
  device_id: 35976d0dfeb5c514c041e44a07db6923
  endpoint_id: 1
  cluster_id: 1280
  command: Drop
  args: {}
  params: {}
origin: LOCAL
time_fired: "2023-08-22T13:44:53.963509+00:00"
context:
  id: 01H8EQVEYB6T5BG03PC6CWQ9XH
  parent_id: null
  user_id: null
``` 

```javascript
event_type: zha_event
data:
  device_ieee: 00:15:8d:00:08:f3:4e:31
  unique_id: 00:15:8d:00:08:f3:4e:31:1:0x0500
  device_id: 35976d0dfeb5c514c041e44a07db6923
  endpoint_id: 1
  cluster_id: 1280
  command: Tilt
  args:
    degrees: 85
  params: {}
origin: LOCAL
time_fired: "2023-08-22T13:44:49.413405+00:00"
context:
  id: 01H8EQVAG59Z2N6BS9YSD8TDP6
  parent_id: null
  user_id: null
```

```javascript
event_type: zha_event
data:
  device_ieee: 00:15:8d:00:08:f3:a4:80
  unique_id: 00:15:8d:00:08:f3:a4:80:1:0x0500
  device_id: 0e7ba05c817360c92020dc137af0bbaf
  endpoint_id: 1
  cluster_id: 1280
  command: vibration_strength
  args:
    strength: 17
  params: {}
origin: LOCAL
time_fired: "2023-08-22T13:50:30.166733+00:00"
context:
  id: 01H8ER5Q8P41SMGA4K5SRFN5QF
  parent_id: null
  user_id: null
```

RaceNJason, 2024-03-13:
Anyone make a ZHA quirk that works for this yet? I'm in the same boat where I don't want to switch/use 2mqtt just for this sensor's xyz values. I'm ok with implementing a quirk if it never gets implemented in the base code...

closed 2024-09-16 by github-actions[bot]

Andrew-T-Smith, 2025-11-01:
@RaceNJason So uhh what are the chances on that quirk being done and just not uploaded? Or did you find a better workaround than MQTT?