zigpy/zha-device-handlers #2461: [Device Support Request] ThirdReality Multifunction Light 3RSNL02043Z

llhappier, 2023-07-06:
### Problem description

Hi, 
I have a multifunction night light. It use private motion cluster and I want it work correctly.
By the way, how can i remove the start-up color temprature? Thanks.

### Solution description

I think it should it should change private cluster(0xfc00) to ias zone(0x0500) .

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

 "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4877, maximum_buffer_size=127, maximum_incoming_transfer_size=242, server_mask=11264, maximum_outgoing_transfer_size=242, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0102",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0008",
            "0x0012",
            "0x0300",
            "0x0400",
            "0x1000",
            "0xfc00"
          ],
          "output_clusters": [
            "0x0019"
          ]
        }
      },
      "manufacturer": "Third Reality, Inc",
      "model": "3RSNL02043Z"
    },

### Diagnostic information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Asia/Hong_Kong",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.07.1",
    "host_os": "Home Assistant OS 10.2",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {},
[... 543 lines cut ...]
    }
    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.COLOR_DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    BasicCluster.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    MultistateInput.cluster_id,
                    Color.cluster_id,
                    IlluminanceMeasurement.cluster_id,
                    ThirdRealityMotionCluster,
                    MotionCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            }
        }
    }

llhappier, 2023-07-07:
In Custom quirk, I also try

darkxst, 2023-07-07:
Does this device work at all for you? I tried without any custom quirk, but it seems to go into a boot loop, going unavailable every 60secs or so.

llhappier, 2023-08-03:
Sorry, I'm a bit busy lately.
This problem is that I didn't call my Test script correctly and I fix this problem.
The steps are as described on the homepage, thanks.

closed 2023-08-03 by llhappier