zigpy/zha-device-handlers #2462: [Device Support Request] 2023 Aqara Light Sensor T1 Brightness Sensor

Anto79-ops, 2023-07-07:
### Problem description

Hi, purchased a few of these sensors

2023 Aqara Light Sensor T1 Brightness Sensor on Ali express:

https://www.aliexpress.com/item/1005005012531138.html

Touted as 2023 with model number is GZCGQ11LM

The device pairs and works correctly (with SkyConnet)

[image]

However, a couple of feature may be missing.

1. No battery level entity
2. Interval updating (as reported by Z2M user that this device can support)

[image]

### Solution description

possible update the library driver for create a quirk?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheJulianJES, 2023-07-07:
You need to attach the device signature or device diagnostic information.

Anto79-ops, 2023-07-07:
ah, yes how did I forget that. Sorry about that. Here's the diagnostic information (file download)

[zha-35e4f47bd1aec2897f2fa1af0eeb1329-LUMI lumi.sen_ill.agl01-4af9c66c77f298fff514ff0407f47e2b.json.txt](https://github.com/zigpy/zha-device-handlers/files/11985011/zha-35e4f47bd1aec2897f2fa1af0eeb1329-LUMI.lumi.sen_ill.agl01-4af9c66c77f298fff514ff0407f47e2b.json.txt)

asucrews, 2023-08-01:
Z2M Page & code block if that helps.  Onces this is added i should be able to run ZHA :)

 https://www.zigbee2mqtt.io/devices/GZCGQ11LM.html

`    {
        zigbeeModel: ['lumi.sen_ill.agl01'],
        model: 'GZCGQ11LM',
        vendor: 'Xiaomi',
        description: 'Aqara T1 light intensity sensor',
        fromZigbee: [fz.battery, fz.illuminance, fz.aqara_opple],
        toZigbee: [tz.GZCGQ11LM_detection_period],
        meta: {battery: {voltageToPercentage: '3V_2850_3000'}},
        exposes: [e.battery(), e.battery_voltage(), e.illuminance(), e.illuminance_lux(),
            e.numeric('detection_period', exposes.access.ALL).withValueMin(1).withValueMax(59).withUnit('s')
                .withDescription('Time interval in seconds to report after light changes')],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
            await endpoint.read('aqaraOpple', [0x0000], {manufacturerCode: 0x115f});
        },
        ota: ota.zigbeeOTA,
    },`

TheJulianJES, 2023-08-01:
https://github.com/zigpy/zha-device-handlers/issues/1473

Can you check what the device reports for "Power Source"?

Anto79-ops, 2023-08-01:
@TheJulianJES 
[image]

TheJulianJES, 2023-08-01:
Thanks. Please download and install this file as a custom quirk: https://raw.githubusercontent.com/zigpy/zha-device-handlers/98bd5229c6f233f27d8960b1021707fc50e9f962/zhaquirks/xiaomi/aqara/illumination.py  -> SAVE using your browser / CTRL + S or CMD + S in most cases).

Instructions for installing: https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

(For that, you just have to create the `custom_zha_quirks` folder in your `/config` directory and add the zha config section to your `configuration.yaml`, then drop the downloaded file in that folder and restart Home Assistant.
If everything works correctly, the sensor should/might work now.
Let me know if that custom quirk works correctly, so we can get it merged into ZHA (so you can remove that custom quirk and it'll keep on working).

Anto79-ops, 2023-08-01:
Sounds good. I will report back in a few hours.

Cheers!

TheJulianJES, 2023-08-01:
Note: It should only change the "power source" to "battery" and actually show the battery entity.

If that works, I'll also look at adding the update interval attribute.

TheJulianJES, 2023-08-01:
Here's a quirk with the added detection interval: https://raw.githubusercontent.com/zigpy/zha-device-handlers/5ef2d80955f1ba205bcebb3e3c73891d8371adfc/zhaquirks/xiaomi/aqara/illumination.py

Apparently you can set it from 1 to 59 (in seconds).

Do note that you'll have to set it on the `OppleCluster` using:
-  "Manage Zigbee device -> clusters" -> select `OppleCluster`
- -> select `detection_interval` -> click "Read Zigbee attribute" to read current interval

It'll display in the first text field.
To set a new interval, change the value in the first text field (and always leave the lower text field empty!), then click "Write Zigbee attribute".

Let me know if this works (and if the battery entity works).

Anto79-ops, 2023-08-01:
Perfect!

I will try the quirk with interval report out.

By the way I don't need to delete or repair the device correct?

TheJulianJES, 2023-08-01:
Yeah, you shouldn't need to re-pair it. It should pick up the quirk (if everything is working) and display it under "Power Source" (and that should change from "Mains" to "Battery or Unknown").

When reading/writing the detection interval, you might have to wake up the device by pressing the button on it.

Anto79-ops, 2023-08-02:
Hey! it worked exacty as you said it would.

the battery entity is showing up now, albeit it shows unknown, I know eventually that will report a %

[image]

But the attribute is reporting correctly battery voltage, size etc so I know that will eventually report a %

[image]

Now as for the reporting interval, the default was 5

[image]

but I changed it to 59 seconds. Here the screen shot after I re-read the the value after waking it up again

[image]

to make sure it is obeying the interval, I watch the light value and indeed it reported every 1 minute

Woo hoo! now, less clutter on my ZHA network.

Thank again @TheJulianJES

Anto79-ops, 2023-08-02:
and as expected....the battery % came after about 20 minutes

:)
[image]

TheJulianJES, 2023-08-02:
Awesome! The quirk won't make it in time for 2023.8.0 (we just pushed a new version some hours ago).

Expect it to be officially available in 2023.9.0.
For now, you can keep using your custom quirk, but you should remove it for future versions which include the quirk (to not cause any issues in the future).

Anto79-ops, 2023-08-02:
That sounds great!

I often join the beta core, so I'll make sure to remove it before I do the beta update but also I'll keep out for ZHA core updates too.

Thanks all for the help.

TheJulianJES, 2023-08-02:
It won't break instantly, even if the quirk is merged. The custom quirk will take precedence over the integrated one though.
And if something major ever changes in the libraries, that custom quirk *could break*.

closed 2023-08-02 by Anto79-ops

kristofferR, 2023-08-15:
Turns out that it is VERY easy to pair the Aqara T1 sensor halfway, leaving it broken. After pairing it and deleting it like 5 times, it finally paired properly and it works fine now.

PR #2504 (2023-08-02): Add Aqara T1 illumination sensor `lumi.sen_ill.agl01`
## Proposed change

Adds support for the Aqara T1 illumination sensor `lumi.sen_ill.agl01`.

This adds:
- the power source
- adds the battery entity
- adds the `OppleCluster` with a `detection_interval` attribute (1 to 59 seconds)

Tested and confirmed working in: #2462

## Additional information

Fixes #2462

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works