zigpy/zha-device-handlers #2463: Tuya Zigbee Energy Meter Smart Power Consumption Monitor

Martin123456777, 2023-07-07:
### Problem description

After added meter to Zigbee network, no entities for metering are visible in HA.
Item: https://www.aliexpress.com/item/1005005418122329.html?spm=a2g0o.order_list.order_list_main.31.24511802FVo01V

Only LQI and RSSI entities added.

Item is recognized as TS0601 from _TZE200_bcusnqt8

### Solution description

I want to use this meter in Energy panel. I have 3 pieces, for 3 phases.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_bcusnqt8",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.7.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Prague",
    "os_name": "Linux",
    "os_version": "6.1.34",
[... 389 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Thank you :-)

Martin123456777, 2023-07-11:
Hi, i found same device, which is supported on zigbee2mqtt: https://www.zigbee2mqtt.io/devices/SPM01.html
May be helpfull. It is SPM01

Thank you
Martin

nicolasknto, 2023-07-25:
Hello, I also bought three devices from Zemismart SPM01-D2TZ on Aliexpress and I intend to use them in Home Assistant to measure the consumption of two showers and a boiler.
I use the ZHA in the HA and no electrical measurement entity appears.
Devices return model TS0601 _TZE200_bcusnqt8.
I would greatly appreciate it if anyone could expedite creating the custom quirk code to allow full use of these devices.
thank you very much I'm waiting
Nicolas

razqqm, 2023-08-22:
please add SPM01 in ZHA

Cseeu, 2023-08-28:
I have the same issue... Anyone that can pleaaaase add SPM01 into ZHA?

Hedda, 2023-08-29:
For reference, blakadder has posted a nice review of the hardware in his blog here -> https://blakadder.com/zemismart-spm01/

sectune, 2023-09-14:
I am also very interested in having it supported in zha. I have a three phase Zigbee version of SPM01 which I am not using currently and am offering to send it to developers free of charge. Please contact me in case you are willing to give it a shot.

mmastria, 2023-09-18:
I have 16 of SPM01 device in my house, categorizing energy consumptions. I'm using with Zigbee2MQTT, but I'm having several instability issues, and I want to switch to ZHA. I have a few other non-recognized devices, but SPM01 is a critical one, I can't loose my energy monitoring.

Please add support for this device or, If there's a quirk somewhere from this or a similar device, please let me know.

Thanks, 
Mastria.

agocsdaniel, 2023-09-22:
@sectune Check my PR. Please test it with your 3-phase unit. I only have 1-phase ones.

sectune, 2023-09-23:
> @sectune I'm afraid the 3-phase variant is different, but I can't test it

Can I somehow get in touch with you e.g. to send you the 3-phase? (free of charge ofc)

agocsdaniel, 2023-10-01:
@sectune There's probably no need for that. Check the latest commit in the PR. I tried something, I hope it will work for the SPM02.

nemesix29, 2023-12-01:
I'm very enthusiastic about this support

ckadluba, 2024-01-08:
I'm also looking forward to Zemismart SPM01 being supported with ZHA. It seems to work with Zigbee2MQTT but I cannot use this since I have problems getting it (Z2M) to run with my Nabu Casa SkyConnect dongle. :(

BlairC1, 2024-03-03:
This Quirk worked for me - https://github.com/zigpy/zha-device-handlers/pull/2603/commits/719490a2fda71ca790252e952224583cb090aa6e

mrehanabbasi, 2024-03-13:
> This Quirk worked for me - https://github.com/zigpy/zha-device-handlers/pull/2603/commits/719490a2fda71ca790252e952224583cb090aa6e

Any similar one for SPM02?

lboue, 2024-04-03:
> This Quirk worked for me - [719490a](https://github.com/zigpy/zha-device-handlers/commit/719490a2fda71ca790252e952224583cb090aa6e)

It works for mee too:
[image]

mhalano, 2024-06-27:
Just installed an SPM-02 that meters 3 phases and didn't work out-of-the-box, so I'm excited about the fix for this issue.

juapem, 2024-07-05:
I am also interested on making the Zemismart 3 phase meter available in ZHA

Teki84, 2024-08-25:
Hello, will it ever be possible to integrate into ZHA? Thank you.

homerojleal, 2024-09-06:
Yes please +1 for support of SPM01

toarman82, 2025-01-29:
Yo también lo necesito. Por favor,  actualizar en ZHA

eriktack, 2025-02-09:
Im having problem getting it to report energy, it jsu sits at 0. The other values are reported correctly though. Also, the power value is correctly showing as negative if i place it in the wrong direction on the wire.

Any tips on what to do? I already tried to removing it from z2m and readding it, but still the same - it does not seem to increase energy usage.

This is how the state looks: 
{
    "ac_frequency": 50,
    "current": 0,
    "energy": 0,
    "last_seen": "2025-02-09T13:13:18+01:00",
    "linkquality": 98,
    "power": 0,
    "power_factor": 100,
    "produced_energy": 0,
    "voltage": 231.8
}

alexvhr, 2025-06-19:
> I have a question: Why aren’t the quirks merged into ZHA when they already work? Do you know the correct process to get my changes merged into the main branch?
I have no idea :) 
> Thanks for your advice!
Welcome!