zigpy/zha-device-handlers #2469: [Device Support Request]

liamlain, 2023-07-11:
### Problem description

No entities after pair/add Zigbee device  3 gang switch TS0601 by _TZE200_kyfqmmyl with ZHA

### Solution description

hope someone can help to create a custom quirk that i can use for these 3 gang switch

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000a",
        "0xef00"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_kyfqmmyl",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.7.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Asia/Kuala_Lumpur",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.07.1",
    "host_os": "Home Assistant OS 10.3",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
[... 379 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

javicalle, 2023-07-28:
If I'm not wrong, that device can be added to the `TuyaSingleSwitchTI` quirk:
* https://github.com/zigpy/zha-device-handlers/blob/fb00099826980eb52e1541764c17fcf5c8a306d8/zhaquirks/tuya/ts0601_switch.py#L22

Copy the current `ts0601_switch.py` file in your local quirk folder and annd your device in the `MODELS_INFO` part:
```python
        MODELS_INFO: [
            ("_TZE200_7tdtqgwv", "TS0601"),
            ("_TZE200_kyfqmmyl", "TS0601"),
        ],
```

liamlain, 2023-07-28:
Hi, thanks for helping.i just a newbie of home assistant.
for the code u provide is it just copy and replace in ts0601_switch.py?

im sorry to ask  this kind of stupid question.hopefully u can guide me. thank you

liamlain, 2023-07-29:
hi, this is my steps to add the quirks:
1. create a folder name zha_quirks in  \config\ zha_quirks
2. copy ts0601_switch.py to   zha_quirks folder
3. in configuration add this:
```
 zha:
    enable_quirks: true
    custom_quirks_path: /config/zha_quirks/
```
4. restart HA and readd the devide.but result still the same no control light entities

this is the quirks i added ur provided:

```
"""Tuya DP based switches."""
from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    MoesSwitchManufCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)

class TuyaSingleSwitchTI(TuyaSwitch):
    """Tuya single channel switch time on in cluster device."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000,0x0004, 0x0005,0x000a, 0xef00]
        # output_clusters=[0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=81 device_version=1 input_clusters=[0, 4, 5, 10, 61184] output_clusters=[25]>
        MODELS_INFO: [("_TZE200_7tdtqgwv", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Time.cluster_id,
                    Groups.cluster_id,
[... 715 lines cut ...]
                    MoesSwitchManufCluster,
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

```

liamlain, 2023-07-30:
Here i attached diagnostics :

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.7.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Asia/Kuala_Lumpur",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.07.1",
    "host_os": "Home Assistant OS 10.3",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "laundry": {
      "version": "1.0.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "browser_mod": {
      "version": "2.3.0",
      "requirements": []
    },
    "hass_agent_mediaplayer": {
      "version": "2022.5.6",
      "requirements": []
    },
    "roborock": {
      "version": "1.0.9",
      "requirements": [
        "python-roborock==0.30.1",
        "ical==4.5.4"
      ]
    },
    "hass_agent_notifier": {
      "version": "2022.10.25.1",
      "requirements": []
    },
    "localtuya": {
      "version": "5.2.1",
      "requirements": []
    },
    "midea_ac": {
      "version": "0.2.3",
      "requirements": [
        "msmart==0.2.3",
        "pycryptodome",
        "pycryptodomex",
        "click"
      ]
    },
    "dwains_dashboard": {
      "version": "3.4.3",
      "requirements": []
    },
    "alexa_media": {
[... 318 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

javicalle, 2023-07-31:
If you don't have done yet, remove the device from HA, wait a couple of minutes and pair the device again.

liamlain, 2023-08-01:
> If you don't have done yet, remove the device from HA, wait a couple of minutes and pair the device again.

tested,but still no luck

[image]

[image]

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000a",
        "0xef00"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_kyfqmmyl",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

```

javicalle, 2023-08-01:
The quirk is not loading.
Any error in the HA startup logs?

liamlain, 2023-08-01:
here the log file after i tested add the deivce and restart HA

```
Logger: zhaquirks
Source: /usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py:456
First occurred: 12:48:39 AM (1 occurrences)
Last logged: 12:48:39 AM

Unexpected exception importing custom quirk 'ts0601_switch'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 454, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 605, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1120, in load_module
  File "<frozen importlib._bootstrap_external>", line 945, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 721, in _load
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 936, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1074, in get_code
  File "<frozen importlib._bootstrap_external>", line 1004, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/zha_quirks/ts0601_switch.py", line 42
    }
    ^
SyntaxError: unmatched '}'
```

javicalle, 2023-08-01:
I see the error. Code from comment has been fixed. You just need to add the `1: {` endpoint like this:

```python
    signature = {
        MODELS_INFO: [
            ("_TZE200_kyfqmmyl", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }
```

Save changes, restart HA a pair the device again.

liamlain, 2023-08-01:
Yes!!finally work,thanks!!

[image]

PR #2966 (2024-02-11): Add Tuya `_TZE200_kyfqmmyl` switch variant
New Moes 3 gang switch TS0601 by _TZE200_kyfqmmyl

## Proposed change

New `TuyaTripleSwitch` variation for `_TZE200_kyfqmmy`

## Additional information

Fixes: #2469 (already closed)
Fixes https://github.com/zigpy/zha-device-handlers/issues/716

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2024-02-11 by TheJulianJES