zigpy/zha-device-handlers #2473: [BUG] ts0601_valve.py: unit of measurement L (liters) doesn't match device class 'energy'

nicknol, 2023-07-13:
### Bug description

the quirk for the garden watering valve ts0601_valve causes a warning in the logs about using a unit that is not matching the device class:
`Entity sensor.bewasserung_garten_1z_summation_delivered (<class 'homeassistant.components.zha.sensor.SmartEnergySummation'>) is using native unit of measurement 'L' which is not a valid unit for the device class ('energy') it is using; expected one of ['kWh', 'MWh', 'GJ', 'Wh', 'MJ']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+zha%22`

it is about the entity containing the delivered water, which of course should be liters.

In the code I saw the import:
`from zigpy.zcl.clusters.smartenergy import Metering`
may it have caused the device class energy?

### Steps to reproduce

it happens each time when I restart homeassistant and the quirk is executed. 
you'll find the warning in the logs

### Expected behavior

No warning about mismatching units in the logs.

And: I stumbled upon these warnings, because I am searching for the reason of wrongly reported summation values

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

nicknol, 2023-10-12:
wondering whether there is a chance to get it fixed :)

almendesb, 2024-01-10:
I want to know how fix too!

closed 2024-07-15 by github-actions[bot]