zigpy/zha-device-handlers #2475: [BUG] nous plug TS011F by _TZ3000_2putqrmw "unknown cluster command"

dierochade, 2023-07-14:
### Bug description

I have got two of these plugs, they work as expected. 
Although i get persistently these error messages: 

Logger: zigpy.zcl
Source: runner.py:179
First occurred: July 13, 2023 at 00:51:36 (40 occurrences)
Last logged: 00:51:47

[0x322A:1:0xe001] Unknown cluster command 225 b'\x02\xfa\xf8A,\n\x06\x00\x00\x03\x06\x00\x00\x00'
[0x3170:1:0xe001] Unknown cluster command 225 b"\n\xe9\xf8A,R\x04\x00\x00N\x04\x00\x00I\x04\x00\x00D\x04\x00\x00?\x04\x00\x00:\x04\x00\x006\x04\x00\x001\x04\x00\x00,\x04\x00\x00'\x04\x00\x00\x01"
[0x3170:1:0xe001] Unknown cluster command 225 b'\niJC,W\x04\x00\x00R\x04\x00\x00N\x04\x00\x00I\x04\x00\x00D\x04\x00\x00?\x04\x00\x00:\x04\x00\x006\x04\x00\x001\x04\x00\x00,\x04\x00\x00\x01'
[0x322A:1:0xe001] Unknown cluster command 225 b'\x03|JC,\x11\x06\x00\x00\n\x06\x00\x00\x03\x06\x00\x00\x00'

### Steps to reproduce

run zha, wait some time, showing up multiple times

### Expected behavior

cluster command should be dismissed or integrated properly 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_2putqrmw",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>
[... 562 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

TheJulianJES, 2023-07-30:
You can ignore them for now. They're probably special "Tuya messages" that ZHA can't parse (and aren't really relevant).

dierochade, 2023-08-07:
That is not nice, cause it gives a lot of spam warnings, but ok. Thanks for this project btw.

closed 2024-02-10 by github-actions[bot]