zigpy/zha-device-handlers #2476: [Device Support Request] Full implementation of BOSCH Radiator Thermostat II (BTH-RA)

dierochade, 2023-07-15:
### Problem description

I use skyconnect+zha.
I am able to **pair the device** using zha.permit with install code and source-ieee.

It then shows up with **4 entitys**: 

1. [button.bosch_rbsh_trv0_zb_eu_identify]

        BOSCH RBSH-TRV0-ZB-EU Identify
        unknown	device_class: identify
        friendly_name: BOSCH RBSH-TRV0-ZB-EU Identify

2. [climate.bosch_rbsh_trv0_zb_eu_thermostat]

        BOSCH RBSH-TRV0-ZB-EU Thermostat
        heat	hvac_modes: off, heat
        min_temp: 5
        max_temp: 30
        current_temperature: 24.6
        temperature: 21
        system_mode: [<SystemMode.Heat: 4>]/heat
        occupied_cooling_setpoint: 2300
        occupied_heating_setpoint: 2100
        friendly_name: BOSCH RBSH-TRV0-ZB-EU Thermostat
        supported_features: 1

3. [sensor.bosch_rbsh_trv0_zb_eu_battery]

        BOSCH RBSH-TRV0-ZB-EU Battery
        100	state_class: measurement
        battery_voltage: 3.1
        unit_of_measurement: %
        device_class: battery
        friendly_name: BOSCH RBSH-TRV0-ZB-EU Battery

4. [sensor.bosch_rbsh_trv0_zb_eu_hvac_action]

      BOSCH RBSH-TRV0-ZB-EU HVAC action
      unknown	friendly_name: BOSCH RBSH-TRV0-ZB-EU HVAC action

**I have got two main issues:**

1. The state of climate.bosch entity is always "heating". If I change the target temparature I can hear the motor of the device closing/opening the valve, but it doesnt change the state.

If I do set the state manually in the developer tools, it changes but switches back after some seconds.

2. Most attributes are missing

From the integration via z2m it is known that there are many more attributes, some are importat/useful:

window open, boost, position of the valve, temperature offset, disabling of internal temp sensor ++

As far as I understand, these attributes can be manipulated via a set command in z2m....

**Additional wish**

The device got a nice led color ring that can show status. I do not know if the led can be manipulated or is controlled locally by the device solely

### Solution description

Quirk?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>
[... 462 lines cut ...]
<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

https://www.zigbee2mqtt.io/devices/BTH-RA.html

https://github.com/Koenkk/zigbee2mqtt/issues/14926

https://zigbee.blakadder.com/Bosch_BTH-RA.html

pdsccode, 2023-10-07:
After finally switching to a SkyConnect, I want to ditch my Bosch SHC but without a working quirk, I cannot fully control the thermostat.

Please prioritize this. If needed, I might be able to provide neccessary information about the device.

Samueras, 2023-10-18:
Same from me, If Information are neede, I'm happy to help, for the Bosh Thermostat 2 and the BOSCH Smart Home Roomthermostat

JohndeWaal, 2023-10-21:
I tried to connect the BOSCH Radiator Thermostat II today. On multiple attempts it connects each time, but most times without any entities. It only connected once with entities (battery 100% HVAC), but they didn't work.

Samueras, 2023-10-21:
It worked for me. What you do is first you use the zha.permit zu give it the adress and installcode that is inside the battery lid. That you need to go in HA to adding new Zigbee devices. Then you reset the Themostat. (Hold the button while insterting the batteries till it blinks orrange.) Lastly press the button ont the thermostat once more so it enters pairing mode.

JohndeWaal, 2023-10-21:
> It worked for me. What you do is first you use the zha.permit zu give it the adress and installcode that is inside the battery lid. That you need to go in HA to adding new Zigbee devices. Then you [...]

Did this each time. I’ve got 53 working zigbee devices, but this one isn’t working. I bought it to replace a Tuya thermostat, even that worked okay (only battery drained every 2 months).
I ordered a Bosch controller now, hope it works with Home Assistant.

Samueras, 2023-10-21:
I actually want to move awy from bosh for the don't hold the Set temperatur but hat up and cool down for about 2 degree all the time
[image]

ulrichkamp61, 2023-10-24:
> hey don't hold the Set temperatur but hat up and cool down for about 2 degree all the 
Same for me, it only holds the temperature, when I frequently update the Set temperature, which is `Occupied heating setpoint`. 

Also, the Z2MTT dialog says for element `Remote temperature` ->  `Setting this will disable the internal temperature sensor until batteries are removed!`, which is not right. I can always change the remote temperature, and as far as I understood, this setting interferes with the heating mode and with the Set temperature. So, it seems there are some small bugs in the Bosch Thermostat Z2MTT implementation. 

With some tricks (setting up the thermostat target values about every 30 minutes reflecting the room temperature as needed) I got the thermostat working for me, but it is not very handy and intuitive.

dierochade, 2023-10-26:
@Samueras @ulrichkamp61 @everybodyelse

Regarding the overshooting temperature: I believe this is intentional, to compensate that close to the heater the temp will rise very fast but not in the rest of the room. I have some trv's with external sensors that do not show this kind of wave overshooting the limit set. 

regarding the remote temp: As far as i can see, home assistant tracks "remote" and "current" temperature, but it is the same value all the time. So current is just mirroring remote values. I have not tried to deactivate the remote temp (because it will just keep the last state forever till reboot of home assistant), but it might end in mirroring an unavailable state - and the documentation on z2m means you have to reset the trv to enable measurement again? 

I run these trvs on z2m now, which works as expected. But there is major setback in my installation: The device (without hub) has a build in schedule that cannot be wiped (as i do not know how, at least). this would be annoying but could be overcome with a simple time based automation. but it calls this scheduled temperature also every time when switched to auto mode… This has nothing to do with zigbee but the firmware itself. 
I was able to fix this in node red, but it is a real pain…

For any one interested, I made a rather big flow in node red (based on z2m)  with this key capabilities:

-run different schedules for workday and holiday with a 15 min schedule (manual timetables with array of targettemps though)
-ability to change temperatures (triggered by external event/manual switch) specified trv and time(range), e.g. heat now for x hours, -heat guest room today, heat living room this evening etc
-block schedule for single trv after manual change for a given period of time (e.g. 90 min) to prevent imminent overwrite.
-boost functionality up to 30 mins (6*5 min)
- away mode
- window sensor
- external temp sensor
- lovelace interface  (incomplete, cause i still think how to best use the ability to change temperatures) 
display brightness auto dim at night

I would be interested in collaboration and further joint development, although it is still work in progress and documentation limited..just pm me

jclsn, 2023-11-06:
Here is all the relevant information I could gather. Maybe OTA would also be relevant for updates, but the endpoint is quite complicated.

```
Signature:

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4617, maximum_buffer_size=127, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0020",
        "0x0201",
        "0x0204",
        "0x0b05"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "BOSCH",
  "model": "RBSH-TRV0-ZB-EU",
  "class": "zigpy.device.Device"
}

Endpoints:

	Basic:
		manufacturer: BOSCH
		model: RBSH-TRV0-ZB-EU
		generic_device_class: GenericDeviceClass.Lighting ???
		generic_device_type: GenericDeviceType.LightBulb ???

	Thermostat (Endpoint id: 1, Id: 0x0201, Type: in)

	Attributes:
		ctrl_sequence_of_oper (id: 0x001b)
			values:
				- ControlSequenceOfOperation.Heating_Only
				- ControlSequenceOfOperation.Heating_With_Reheat
				- ControlSequenceOfOperation.Cooling_Only
				- ControlSequenceOfOperation.Cooling_With_Reheat

		local_temperature (id: 0x0000) 
			read-only: yes
			format: 2050 (20.5°)

		local_temperature_calibration (id: 0x0010) 
			read-only: no
			format: 25 (2.5°)
			range: -50 to 50

		max_heat_setpoint_limit (id: 0x0016)
			read-only: no
			format : 500 (5°)
			range: 500 - 3000

		min_heat_setpoint_limit (id: 0x0015)
			read-only: no
			format : 500 (5°)
			range: 500 - 3000

		number_of_daily_transitions (id: 0x0022)
			read-only: yes
			format: integers
			
		number_of_weekly_transitions (id: 0x0021)
			read-only: yes
			format: integers

		occupied_cooling_setpoint (id: 0x0011)
			read-only: no
			format : 500 (5°)
			range: 500 - 3000

		occupied_heating_setpoint (id: 0x0012)
			read-only: no
			format : 500 (5°)
			range: 500 - 3000

		setpoint_change_source (id: 0x0030)
			read-only: yes
			values: 
				- SetpointChangeSource.External
				- SetpointChangeSource.Schedule

		start_of_week (id: 0x0020)
			read-only: yes
			values: 
				- StartOfWeek.Monday
				- StartOfWeek.Tuesday
				- StartOfWeek.Wednesday
				- StartOfWeek.Thursday
				- StartOfWeek.Friday
				- StartOfWeek.Saturday
				- StartOfWeek.Sunday

		system_mode (id: 0x001c)
			read-only: no
			values:
				- SystemMode.Heat
				- SystemMode.Cool

	Commands:
		clear_weekly_schedule (id: 0x0003)

	PowerConfiguration (Endpoint id: 1, Id: 0x0001, Type: in)

		battery_alarm_mask (id: 0x0035)
			read-only: no
			format: integer
			values:
				- 1
				- 0
			
		battery_alarm_mask (id: 0x003e)
			read-only: yes
			format: integer

		battery_percentage_remaining (id: 0x0021)
			read-only: yes
			format: integer

		battery_quantity (id: 0x0033)
			read-only: yes
			value: 2

		battery_size (id: 0x0031)
			read-only: yes
			value: BatterySize.AA

		battery_volt_min_thres (id: 0x0036)
			read-only: yes
			value: 22

		battery_volt_min_thres (id: 0x0036)
			read-only: yes
			value: 22

		battery_volt_thres1 (id: 0x0037)
			read-only: yes
			value: 23

		battery_volt_thres2 (id: 0x0038)
			read-only: yes
			value: 27

		battery_volt_thres3 (id: 0x0039)
			read-only: yes
			value: 29

		battery_voltage (id: 0x0020)
			read-only: yes
			format: 27 (2.7 V)
				

	UserInterface (Endpoint id: 1, Id: 0x0204, Type: in)

		keypad_lockout (id: 0x0001)
			values:
				- KeypadLockout.No_lockout
				- KeypadLockout.Level_1_lockout

		temperature_display_mode (id: 0x0000)
			values:
				- TemperatureDisplayMode.Metric
				- TemperatureDisplayMode.Imperial

```

jclsn, 2023-11-06:
> I actually want to move awy from bosh for the don't hold the Set temperatur but hat up and cool down for about 2 degree all the time [image]

Interesting! I don't have this issue. My heating setpoint is at 21°C and it stops at 21°C. 

What is your value for ``ctrl_sequence_of_oper ``?

coxtor, 2023-11-14:
I've just purchased three bosch thermostats and was able to pair them using the QR-Code with ZHA. However I only have a couple of sensors there: temperature, mode (heat an off) , hvac(unknown) and battery. LQI and RSSI are not available. 
There should be a whole bunch more or am I wrong?

jclsn, 2023-11-17:
@coxtor Yes, I am currently writing a quirk to support it. It should be ready in the coming weeks. This thermostat actually supports even valve opening from 1-100%. Really nice!

BernhardMaier, 2023-11-17:
@jclsn Very good 👏  I'm really looking forward to being able to use more functionalities.

Samueras, 2023-11-17:
> @coxtor Yes, I am currently writing a quirk to support it. It should be ready in the coming weeks. This thermostat actually supports even valve opening from 1-100%. Really nice!

You just became my personal Hero.

dierochade, 2023-11-17:
> @coxtor Yes, I am currently writing a quirk to support it. It should be ready in the coming weeks. This thermostat actually supports even valve opening from 1-100%. Really nice!

If I can contribute with any log/data from running these on z2m, just let me know. Will gladly assist.

jclsn, 2023-11-17:
> If I can contribute with any log/data from running these on z2m, just let me know. Will gladly assist. 

No need. Everything is in the Z2M repo. I just need to find time. We have a lot to do at work atm.

wallenium, 2023-11-17:
@jcsln willing to donate a beer. 12 thermos are waiting for full support.

jclsn, 2023-11-18:
@wallenium There are better beers if you ask me ;) 

So I have implemented the endpoints and they react correctly

https://github.com/jclsn/zha-device-handlers/commit/621eabd98b7ab821c26afdf28ed76044597a1856

If you want to try it out, add it as a custom quirk as described [here](https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226), then click on the three dot menu -> manage devices and make the setting on the ``BoschThermostat`` or ``BoschUserInterface``.

I still have to expose the endpoints correctly to home assistant. Bosch's ``system_mode`` and ``operating_mode`` endpoints are very different from the normal Zigbee specification and therefore what ZHA expects. That is why you get an error when trying to turn the thermostat off. Unfortunately the Zigpy code is not very well documented and I have to rely at the developers pointing me in the right direction on how to do this.

wallenium, 2023-11-18:
@jclsn You are right. But none which has a connection to this city ;)

Short question: Seems to work. Do you know the setting for temperature offset? Need to set 1.5 degrees offset... ;)
Is there a list of possible parameters? Orientation of display would be nice, too.

jclsn, 2023-11-18:
@wallenium The ``local_temperature_calibration`` attribute works without the quirk. 1.5°C should be an integer value of ``15``. 

Regarding the beer, there are plenty of good other beers from private breweries in Hamburg: Ratsherren, Wildwuchs, Gröninger

wallenium, 2023-11-18:
You are right. Favorite is Ratsherren.

I will try. 
Possible to add this set point to device management page?

jclsn, 2023-11-18:
``-15`` correctly sets -1.5°C for me

pdsccode, 2023-11-19:
Just wanted to say thanks @jclsn for working on this. I installed the quirk and like expected, it works for everything that worked before. For me, the only thing I really need added is the off mode. Though it seems there is no real "off" in the Bosch implementation. At least I did not find any working one.

jclsn, 2023-11-19:
@pdsccode No, there is this attribute I called ``operating_mode`` with ``manual``, ``pause`` and ``schedule``. I will probably have to create a virtual attribute somehow that encompass those and also ``heat`` and ``cool`` from ``system_mode``. Usually those would all be under system mode and be exclusive. 

Anyway, ``pause`` is what you are looking for!

pdsccode, 2023-11-19:
Ah, good to know. But it seems I am either blind or too dumb. I can not find the attribute you mention.
My states look like the following:
[image]

And my device page look like this:
[image]

May I have missed something?

jclsn, 2023-11-19:
Yes, seems like I will have to implement the entities from the ZHA side.

JericoPablo, 2023-11-29:
can anyone tell me here how i could use the remote temp? I can put a value in, but i would like to use a value from my sensor, like sensor.kitchen_roomthermostate_temp. I have some GoveeLife/Govee thermostates for my rooms which report me the temp of the room. Much accurater then the bosch one which is directly under the window / next to heater.

jclsn, 2023-11-29:
@JericoPablo I recommend not doing that. There is a note in Zigbee2MQTT code that touches have to take out the batteries and reset the thermostat once you have set this attribute. Therefore I will not implement this functionality. You will be able to do all of this with Home Assistant, once the thermostat is fully integrated.

jclsn, 2023-11-29:
I btw just wanted to say that I currently don't have the time and energy to fully implement this. I have been trying over the course of the last thee weeks, but I am just getting sporadic responses from the maintainers, which is not enough, as I am not an experienced Python or HA developer. 

If you want your thermostat to fully work, just use Zigbee2MQTT. It is a far more active project and all functionalities are implemented there!

coxtor, 2023-11-29:
Thanks for the effort. For now i have switched back to z2m and the thermoststrs are working great.

jclsn, 2023-11-30:
@JericoPablo You can also use
 
https://community.home-assistant.io/t/advanced-heating-control/469873

It has a field for an external temperature sensor and much more

exceldatei, 2023-12-19:
Hello all,

I tried to get the quirk implemented with your advise but I failed.

What I did is the following:

1. Creating a folder in the main HA directory which is called zhaquirks/bosch
2. Downloaded the Python file from jclsn and imported to the created folder
3. Restart HA
4. Deleted Bosch thermostat from ZHA
5. Add device in ZHA
6. Set Bosch thermostat to pairing mode

This is the result I am getting:

[image]

Any ideas, what I miss?

Thank you!

cadwizzard, 2023-12-25:
> Hello all,
> 
> I tried to get the quirk implemented with the advise of @jclsn but I failed.
.............

You cant just pair it, You need to go to developer options, services.
Find service: Zigbee Home Automation: Permit

Tick the box for QR code and input the code from the QR code on the TRV under the batteries. 

I've paired fine like that (I scan with a phone, save to google keep and then just do the rest by copying it back on a PC :) - but any way you get the exact code into the QR code option will work

JericoPablo, 2024-02-05:
> > I actually want to move awy from bosh for the don't hold the Set temperatur but hat up and cool down for about 2 degree all the time [image]
> 
> Interesting! I don't have this issue. My heating setpoint is at 21°C and it stops at 21°C.
> 
> What is your value for `ctrl_sequence_of_oper `?

first of all thanks for the answer for my other question, but since im not that often on github i just saw it.

Im facing the same "issue" about overshooting. Sometimes it even goes below the temp i set for half degree and starts then to heat.

Its not a huge issue for me but if there is any fix for this i would of course be happy 😄 sometimes one of my TRV gets stuck exactly on the temp i set and it continuesly heats but doesnt goes above the temp. I guess thats how its supposed to be? But the rest of them always has this rollercoaster like thing

JanoschDeurer, 2024-02-06:
> first of all thanks for the answer for my other question, but since im not that often on github i just saw it.
>
> Im facing the same "issue" about overshooting. Sometimes it even goes below the temp i [...]

I noticed, that the internal time schedule is always playing around with the configured temperature and therefore I cot similar fluctuations. I did a dirty fix where I set that thermostat to the target value every 30 min. That seems to fix the problem... Not 100% sure if this is the problem...

It seems to me that the here thought of implementation would fix a lot of these issues. I hope someone can spare the time soon, I'm very thankful for all your work 🙏

dierochade, 2024-02-07:
> It seems to me that the here thought of implementation would fix a lot of these issues. I hope someone can spare the time soon, I'm very thankful for all your work 🙏

take a look here, to see what has been changed in the device converter lately (1.35.1) to address the issue: [#6761](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6761) and [#6742](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6742).

Please take a look here too, about a possible change for the system mode of the trv:  [#6995](https://github.com/Koenkk/zigbee-herdsman-converters/issues/6995).

jclsn, 2024-02-11:
@dierochade I have started adding the cluster handlers to the core with the help of @dmulcahey. He is also pretty busy unfortunately. A few attributes are already working, but my coordinator is failing all the time. Just ordered a new one. Anyway, even if I did finish it soon, it would probably not get merged into the core before the winter is over...

exceldatei, 2024-02-16:
After getting the thermostat integrated I came across another issue:

Adressing the thermostats temperature is working fine. However, the HLK-action is not working properly:

[image]

The current temperature is far below the set temperature, however, it is still in heating mode. 

Do you have any suggestions?

dierochade, 2024-02-16:
> Do you have any suggestions?

you are just misinterpreting the information: heating means: manual mode, opposed to off (and auto). It does not say anything about the valve position. Take a look at the attributes of the climate entity under developer tools/state. 

I am not sure what is exposed in zha at the moment (migrated to z2m), but there you will see the relevant device states…

dierochade, 2024-02-17:
> am I right, that there is no state giving me the information about the valve position correct?

Technically this is true, but pi_heating demand is closely associated cause the device algorithm derives the valve position from this. 

It’s not linear though (only boost opens 100 for instance) and there is some more proprietary logic (as discussed on herdsman converters), but imho for normal use pi heating demand will do.

jclsn, 2024-02-17:
I will implement the ``pi_heating_demand`` in HA, but I don't think that changing it directly is a good idea. Probably best to let the TRV handle this internally, as it will do a better job.

SJankow, 2024-03-05:
@jclsn have you considered adding remote_temperature attribute to your quirk? 

```       
remote_temperature = ZCLAttributeDef(
            id=t.uint16_t(0x4040),
            type=t.int16s,
            is_manufacturer_specific=True,
        )
```

It can be used with some automation, to set the measured temperature from external sensor, like that:

```
- id: 'XXXXXXXXXXXX'
  alias: External Temperature
  trigger:
  - platform: state
    entity_id: sensor.aquara2
  - platform: time_pattern
    minutes: /30
  action:
  - service: zha.set_zigbee_cluster_attribute
    data:
      ieee: XX:XX:XX:XX:XX:XX:XX:XX
      endpoint_id: 1
      cluster_id: 513
      attribute: 16448
      cluster_type: in
      manufacturer: 4617
      value: '{{  (states(''sensor.aquara2'')|float(0) * 100.0)|int(0)
        }}'
``` 

BTW `pi_heating_demand` looks like a perfect sink for PID controllers like [HASmartThermostat](https://github.com/ScratMan/HASmartThermostat). Results with external sensor are pretty good, but it still space for improvements.
[image]

berwinter, 2024-03-31:
I have a Bosch Room Thermostat II - 230V (RBSH-RTH0-ZB-EU), which has similar endpoints and attributes. I used zha_toolkit to scan for all availble attributes and made them available through a quirk:
https://github.com/berwinter/zha-device-handlers/commit/94c5719060fcddd01b481ae829d0d07bad710ec3

I still need to figure out the name/function of those addional attributes and their allowed values. For now I just named them according to their IDs (e.g. attr_0x4022):
[image]

Maybe someone with a Bosch Controller can help to get the mapping?

I don't think we are going to need Entities for all attributes, but at least have the attributes mapped and accessible from ZHA to control the device setup.

We can do the same for the Radiator Thermostat II if someone could perform a scan in zha_toolkit and share the report.

mrrstux, 2024-04-22:
Hi! Many thanks for starting this quirk!

Here is an updated version that converts from Bosch's operating_mode to HA system_mode. It is my first attempt at messing around with ZHA quirks, so it might not be perfect.
[rbsh_trv0_zb_eu.zip](https://github.com/zigpy/zha-device-handlers/files/15068331/rbsh_trv0_zb_eu.zip)

What I tested and works:
 - switch thermostat to Heat or Off from the HA thermostat card
 - change operating_mode from "HA Manage Zigbee device" (this results in the thermostat card updating its state)

jclsn, 2024-04-24:
@mrrstux Thank you, I will try it out soon. Pretty busy atm

mrrstux, 2024-04-30:
Here is a version [quirk_v2_rbsh_trv0_zb_eu.zip](https://github.com/zigpy/zha-device-handlers/files/15170904/quirk_v2_rbsh_trv0_zb_eu.zip) that displays extra controls for:

- Preset { Normal, Boost }
- Window open { switch: On/Off }
- Operating mode: { Pause, Manual, Schedule }
- Displayed Temperature { Measured, Target }
- Display Orientation { Normal, Flipped }
- Display ON time (screen timeout) [5-30]
- Display brightness [0-10]

Since I could not find specific strings inside HA for these, one would need to name/rename each in HA.
Also, a newer HA version might be needed since this quirk version is based on the v2 API - https://github.com/zigpy/zigpy/discussions/1312

@jclsn Sorry to hear that you encounter issues. Maybe they go away with this new version. If not, a more complete trace might help with the investigation.

mrrstux, 2024-05-15:
Latest version: https://github.com/zigpy/zha-device-handlers/pull/2808:
1. Bosch Radiator Thermostat II
 - system_mode to/from operating_mode
 - operating mode: { select: Pause, Manual, Schedule }
 - control sequence of operation: { select: Heating, Cooling }
 - boost { switch: On/Off }
 - window open { switch: On/Off }
 - displayed temperature { select: Measured, Target }
 - display orientation { select: Normal, Flipped }
 - display on time (screen timeout) [5-30]
 - display brightness [0-10]
 - remote temperature
 - pi heating demand

2. Bosch Room Thermostat II 230v
 - system_mode to/from operating_mode
 - operating mode: { select: Pause, Manual, Schedule }
 - control sequence of operation: { select: Heating, Cooling }
 - boost { switch: On/Off }
 - window open { switch: On/Off }
 - display on time (screen timeout) [5-30]
 - display brightness [0-10]
 - pi heating demand

Hedda, 2024-08-14:
Anyone here who would be willing to help the whole community by adding some easy to follow how-to instructions in the ZHA integration documentation for new users pairing (joining) Zigbee devices that require install_code and/or qr_code please see here:

https://www.home-assistant.io/integrations/zha#action-zhapermit

and

https://github.com/home-assistant/home-assistant.io/blob/current/source/_integrations/zha.markdown

That is, add some simpler step-by-step notes similar to these instructions copied from Zigbee2MQTT device documentation:

# Pairing Zigbee devices that require an install-code or QR-code

**_1. To pair this device you have to install the device via its installation code._** 
_**2. The installation code can be obtained by scanning the QR-code with your smartphone, (note that the QR code is often on the outside of the device but can sometimes be found on inside of the battery cover, a seperate sticker or in the manual in the package).**_ 
_**3. Then get the device into pairing mode.**_ 
_**4. In zigbee2mqtt navigate to "Settings" --> "Tools" and click on "Add install code".**_
_**5. Paste the code you got from the QR-code and confirm by clicking "OK" which will get zigbee2mqtt into pairing mode automatically.**_ 
_**6. Wait for your device to be joined**_

PS: Also link to the ZHA limitations section which state that install_code is only supported with Silabs and TI Zigbee Coordinator's:

https://www.home-assistant.io/integrations/zha#limitations

MSZ-Spyro, 2025-01-16:
Hello, im not sure if i undersand this properly, but all i get from this radiator on my HA is what shown on the ss, reading the docu, seems like it should have a lot more ?

[image]

mrrstux, 2025-01-16:
@MSZ-Spyro For better support, you need to deploy this (not yet merged) quirk https://github.com/zigpy/zha-device-handlers/pull/2808.

MSZ-Spyro, 2025-01-16:
> [@MSZ-Spyro](https://github.com/MSZ-Spyro) For better support, you need to deploy this (not yet merged) quirk [#2808](https://github.com/zigpy/zha-device-handlers/pull/2808).

I havn't done it before, so if i understand it correctly, i should put 

> zhaquirks/bosch/rbsh_trv0_zb_eu.py 

into for example 

> /config/zha_quirks/rbsh_trv0_zb_eu.py 

and add 
```
zha:
     enable_quirks: true
     custom_quirks_path: /config/zha_quirks/
```
to the config.yaml, restart hass and readd the TRV so it should have more options then ?

As a side question, do you know when this is planned to be merged ?

mrrstux, 2025-01-16:
@MSZ-Spyro "enable_quirks: true" might not be needed. The name of the config is configuration.yaml.

If it works for you, please leave a comment in the PR to hopefully improve the chances of it getting reviewed and merged sooner.

MSZ-Spyro, 2025-01-16:
Either it doesnt work or ive done something wrong, but by seeing __pycache__ folder in the custom quirks folder i would assume it loaded it. Nothig has changed for me unfortunately

[image]

[image]

mrrstux, 2025-01-16:
@MSZ-Spyro If the quirk was loaded, then you should see this in the "Zigbee info": "Quirk: zigpy.quirks.v2.CustomDeviceV2". If you have it, then I suggest you remove and re-add the TRV to your HA.

mrrstux, 2025-01-16:
@MSZ-Spyro I believe specific quirks for Danfoss exist in the HA installation. For your TRV, the custom quirk doesn't seem to be loaded. Please check HA logs, maybe there is a clue in there as to why that is.

MSZ-Spyro, 2025-01-16:
> [@MSZ-Spyro](https://github.com/MSZ-Spyro) I believe specific quirks for Danfoss exist in the HA installation. For your TRV, the custom quirk doesn't seem to be loaded. Please check HA logs, maybe there is a clue in there as to why that is.

My bad, i couldnt fint any errors in the log before, but i got this one 

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py", line 462, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/zha_quirks/rbsh_trv0_zb_eu.py", line 6, in <module>
    from zigpy.quirks.v2 import QuirkBuilder, ReportingConfig
ImportError: cannot import name 'ReportingConfig' from 'zigpy.quirks.v2' (/usr/local/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py)
```

Not sure why its not available, i technically could just remove the ReportingConfig from imports but should i ?

mrrstux, 2025-01-16:
@MSZ-Spyro I suspect your HA is from before 2024.11 release. If you do not want to upgrade, then yes, you can remove ReportConfig usage from this quirk.

MSZ-Spyro, 2025-01-16:
> [@MSZ-Spyro](https://github.com/MSZ-Spyro) I suspect your HA is from before 2024.11 release. If you do not want to upgrade, then yes, you can remove ReportConfig usage from this quirk.

Yup, it was 2024.10.4. Thank you all for help ! Updated to the newest and works like a charm. 

Only thing that doesnt seem to work 100% properly is the firmware - it started with "up to date" status and after few seconds went to unknown :/ 

[image]

[image]

mrrstux, 2025-01-16:
@MSZ-Spyro I suggest you to also update the firmware of the device - the latest one is 0x37041514,

MSZ-Spyro, 2025-01-17:
> [@MSZ-Spyro](https://github.com/MSZ-Spyro) I suggest you to also update the firmware of the device - the latest one is 0x37041514,

Im not a huge fan of updating things that work fine, and the firmware most probably is to work better with the native hub, but the firmware status shows "unknown" and i dont even know where to get the files from :/

[25 more replies]

closed 2025-08-24 by TheJulianJES