zigpy/zha-device-handlers #2477: [Device Support Request] Add Friends of Hue Zigbee Green Power EnOcean Switch

smartmatic, 2023-07-15:
### Problem description

Zigbee Information from NodOn EnOcean Friends Of Hue switch could not parsed

### Solution description

Add new ZHA device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```

Logger: bellows.ezsp.protocol
Source: runner.py:179 
First occurred: 09:58:40 (25 occurrences) 
Last logged: 10:03:19

Failed to parse frame gpepIncomingMessageHandler: b'7fd6d5003e1f71013e1f71016b02010000d5090000138a2d22a2ff00'
Failed to parse frame gpepIncomingMessageHandler: b'7cd6d6003e1f71013e1f71016b00000000ffffffffe0ffffffffff2e02c5f2a835585ccd1a43d08badddcf2ace49e86f22d7b0d609000004111011121314151617226062636465666768'
Failed to parse frame gpepIncomingMessageHandler: b'7fd6d7003e1f71013e1f71016b02010000d709000068c5233053ff00'
Failed to parse frame gpepIncomingMessageHandler: b'7fd7d8003e1f71013e1f71016b02010000d8090000100c9299c5ff00'
Failed to parse frame gpepIncomingMessageHandler: b'7fd6d9003e1f71013e1f71016b02010000d909000014baa9d58bff00'

```
`

Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:137 
First occurred: 09:58:41 (20 occurrences) 
Last logged: 10:03:19

Error doing job: Exception in callback ThreadsafeProxy.__getattr__.<locals>.func_wrapper.<locals>.check_result_wrapper() at /usr/local/lib/python3.11/site-packages/bellows/thread.py:110
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.11/site-packages/bellows/thread.py", line 111, in check_result_wrapper
    result = call()
             ^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bellows/ezsp/__init__.py", line 339, in frame_received
    self._protocol(data)
  File "/usr/local/lib/python3.11/site-packages/bellows/ezsp/protocol.py", line 180, in __call__
    result, data = self.types.deserialize(data, schema)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bellows/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
                  ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/types/struct.py", line 302, in deserialize
    value, data = field.type.deserialize(data)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/types/basic.py", line 851, in deserialize
    item, data = cls._item_type.deserialize(data)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/types/basic.py", line 190, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
`

`

Logger: zigpy.zcl
Source: runner.py:179 
First occurred: 00:51:21 (119 occurrences) 
Last logged: 10:03:19

[0x94AC:242:0x0021] Unknown cluster command 0 b'\x88T>\x1fq\x01\xd4\t\x00\x00\x17\x00\xac\x94\xff'
[0x94AC:242:0x0021] Unknown cluster command 0 b'\x88T>\x1fq\x01\xd5\t\x00\x00\x13\x00\xac\x94\xff'
[0x94AC:242:0x0021] Unknown cluster command 0 b'\x88T>\x1fq\x01\xd7\t\x00\x00h\x00\xac\x94\xff'
[0x94AC:242:0x0021] Unknown cluster command 0 b'\x88T>\x1fq\x01\xd8\t\x00\x00\x10\x00\xac\x94\xff'
[0x94AC:242:0x0021] Unknown cluster command 0 b'\x88T>\x1fq\x01\xd9\t\x00\x00\x14\x00\xac\x94\xff'

`

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Hedda, 2023-07-15:
@smartmatic which exact EnOcean Zigbee Green Power module is that specific FOH switch from NodOn based on + what is its model and device signature?

@nworbneb Do you have any EnOcean ZGP modules like PTM215ZE, PTM215Z, or PTM216Z to test with ZHA?

PS: For reference there are more ”Friends of Hue” ZGP switches and more mentioned in https://github.com/zigpy/zigpy/issues/341

smartmatic, 2023-07-16:
> @smartmatic which exact EnOcean Zigbee Green Power module is that specific FOH switch from NodOn based on + what is its model and device signature?
> 

The EnOcean Model is a PTM215Z.
The Device itself is this wall switch: https://nodon.fr/en/nodon/enocean-wall-switch/

Can you how help me how can i determine the exact model and device signature?

Hedda, 2023-07-16:
https://www.home-assistant.io/integrations/zha#how-to-add-support-for-new-and-unsupported-devices

When you open a new issue there is instructions on how to collect debug information, including device signature and diagnostic information.

Expand the collapsed sections under device signature and debug information on what details is asked for each:

## Debug information

### Device signature
You can find the device signature by going to the device page, clicking the three dots, clicking on "Manage Zigbee device", and then selecting the "Signature" tab at the top.

Copy the contents and paste your device signature in between the lines with the backticks below.

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

You can find the diagnostic information of the device by going to the device page, clicking the three dots, and then by clicking on "Download diagnostics".

Copy the contents of the downloaded text file and paste it in between the lines with the backticks below.

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

smartmatic, 2023-07-16:
The device did not pair successfully. Therefore i can not provide exact model or device signature. I have only the logging information at the moment.

Hedda, 2023-07-19:
> The device did not pair successfully. Therefore i can not provide exact model or device signature.

Sorry but if the device does not successfully pair then I believe that this still belongs at the zigpy level, i.e. -> https://github.com/zigpy/zigpy/issues/341

smartmatic, 2023-08-17:
> @smartmatic Forgot to ask if you can please also give feedback on which exact Zigbee Coordinator and firmware you tested with.

I am using a Sonoff Dongle Plus E with the latest Firmware 6.10.3 NOT the MultiPAN Firmware.

Yamacore, 2023-08-19:
Hello there, i dont know if im right here but i saw the topic including PTM 215ZE module or the newer EnOcean PTM 216Z.
i currently try to get this [switch](https://new.abb.com/products/de/2CKA006710A0011/6716-uj-84-500) to work with my skyconnect but it seems it just works in zigbee2mqtt atm.

thibaultmol, 2023-10-24:
It's really annoying that I constantly have to replace the batteries on my buttons but I'm still looking for this issue to be fixed. What exactly is the problem that's holding it back right now? (Genuinely asking)

Hedda, 2023-10-24:
> I'm still looking for this issue to be fixed. What exactly is the problem that's holding it back right now? (Genuinely asking)

Note that this requires ZGP (Zigbee Green Power) profile support which is a new feature (and not something broken to be fixed).

https://www.home-assistant.io/integrations/zha#limitations

  “**_ZHA does currently not support devices that can only use the ZGP (“Zigbee Green Power”) profile which is used in a few batteryless self-powered or energy harvesting devices, (such as for example; Philips Hue Click, Philips Hue Tap, and some “Friends of Hue” partnership switches)._**”

What is actually holding this back is there needs to be a volunteering developer with a personal interest in taking on this project.

Can you confirm that you tried adding/pairing/joining a native ZGP (Zigbee Green Power) device to [ZHA](https://www.home-assistant.io/integrations/zha/) and it did not pair/join?

(Note! Will need to use a Zigbee Coordinator based on modern hardware + firmware such as example EFR32MG21 or CC2652).

If symptoms is still that can not even pair/join a device "Friend of Hue" switch based on PTM215ZE/PTM215Z/PTM216Z then assume the issue is that [zigpy](https://github.com/zigpy/zigpy) and/or its radio libraries do not yet fully support native ZGP (Zigbee Green Power) devices.

Tip is there is an end-user feature request and end-users discussion here:

https://community.home-assistant.io/t/zha-integration-support-for-zgp-zigbee-green-power-devices-via-zigpy/277149/

(Maybe the community could start a crowdfunding collection on [Bountysource](https://en.wikipedia.org/wiki/Bountysource) or a similar for this feature to be of more interest?)

Anyway, the main developers/development discussion about native ZGP (Zigbee Green Power) device support can be found here:

https://github.com/zigpy/zigpy/issues/341

PS: I do not own any ZGP (Zigbee Green Power) devices myself, so I can not even test pairing/joining to check debug/diagnostics myself, and regardless of that, I do not have the coding skills to identify the missing pieces and even less so implement the code.

Hedda, 2023-11-12:
FYI, developers should know that there is now a pull request for zigpy to add proper ZGP (Zigbee Green Power) support and the developer of that PR is asking for help in reviewing his code patches there:

https://github.com/zigpy/zigpy/pull/1282

Note that it is still only an early draft with initial work so best if end-users do not post there unless already know how contribute with code testing without handholding (e.i. not posting questions there asking how test a fork/patch).

hesspoint, 2024-03-04:
Hi

I want to move about 20 FoH (zigbee green power) switches from my Hue hub to ZHA and consolidate my 4 zigbee networks. I now learned that I cannot move the switches which sucks. Any ETA when these will be supported by ZHA?

Thanks

Hedda, 2024-03-04:
No ETA as it is for free and open source software developed by volenteers in their spare time, however it sounded really close with PR https://github.com/zigpy/zigpy/pull/1282 but then heard a tree fell on his house so had to prioritize that first simply due to bad luck. Anyway, if you are an end user then discussion can be found here -> https://community.home-assistant.io/t/zha-integration-support-for-zgp-zigbee-green-power-devices-via-zigpy/277149/

closed 2024-09-07 by github-actions[bot]