zigpy/zha-device-handlers #2481: [Device Support Request] Aqara B1 Curtain Motor

MrHaila, 2023-07-16:
### Problem description

B1 does not follow the standards and needs a quirk. It seems like a similar issue to the Aqara E1 roller, which already has a quirk.

The current behavior is that the device pairs correctly and exposes three entities: `identify`, `number` and `cover`. `Identify` makes the device beep (works), `number` represents and sets the current position of the curtains (works, but the naming is off) and `cover` fails to work as expected. The current state of the cover is not correctly mapped to the UI.

`Cover`'s up and down commands do nothing and stop command stops the curtains correctly if they are moving, but gives an error in the UI: "Failed to call service cover/stop_cover. '>' not supported between instances of 'NoneType' and 'int'"
is

### Solution description

`zigbee-herdsman-converters` has support that could be used as a reference. Original commit: https://github.com/Koenkk/zigbee-herdsman-converters/commit/75a7d6640ab5276beddecd04bd1f356757cb7b8a

The essence is that the B1 exposes its current lift percentage via `current_value` in the `analog_output` cluster. It also seems that this value is inverted when compared to the standard? This seems similar to the E1 quirk.

In addition, the device does not support dedicated commands for opening or closing but instead needs those commands mapped to set the `current_value` to 0 or 100.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000d",
        "0x0013",
        "0x0102"
      ],
      "output_clusters": [
        "0x0003",
        "0x000a"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.curtain.hagl04",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.5.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Helsinki",
    "os_name": "Linux",
    "os_version": "6.1.21-v7",
    "supervisor": "2023.07.1",
    "host_os": "Home Assistant OS 10.1",
[... 340 lines cut ...]
          }
        },
        "out_clusters": {
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Additional information

I tried looking into developing a quirk myself but didn't get anywhere as this is really unfamiliar territory. Maybe someone else would have the kind motivation to use the above info to add support?

hastarin, 2023-07-29:
I just thought I'd let you know that I have a slightly different version of this. It's the powered version model ZNCLDJ11LM which appears to have a different signature to yours but has similar issues.

I've got an issue #2451 created and have started to look into creating a custom quirk when I can spare some time.

lukasz-burak, 2023-11-20:
@MrHaila Hey!
Did you solve this issue by any chance? I'm also trying to integrate Aqara B1 battery curtain motor without luck for few days.

MrHaila, 2023-11-24:
Unfortunately not. I've had the battery of the device die on me so figuring out this bit has been on hold 😭

lukasz-burak, 2023-11-24:
@MrHaila I was able to figure it out a bit, based on work from @hastarin.
The only downside is I'm not able to expose battery level to HA for some reason. I'll try to expose PR in some free time but for now you can try my custom quirk if you like.

`"""Aqara Curtain Driver B1 device."""
from __future__ import annotations

from typing import Any

import logging

from zigpy import types as t
from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    AnalogOutput,
    Basic,
    Groups,
    Identify,
    MultistateOutput,
    PowerConfiguration,
    Scenes,
    Time,
)

from zhaquirks import CustomCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.xiaomi import LUMI, XiaomiCustomDevice, XiaomiPowerConfigurationPercent, BasicCluster

_LOGGER = logging.getLogger(__name__)

class AnalogOutputB1(CustomCluster, AnalogOutput):
    """Analog output cluster, used to relay current_value to WindowCovering."""

    def __init__(self, *args: Any, **kwargs: Any) -> None:
        """Init."""
        _LOGGER.debug("AnalogOutputB1 Cluster init")
        super().__init__(*args, **kwargs)
        self._update_attribute(
            self.attributes_by_name["max_present_value"].id, 100.0
        )  # max_present_value
        self._update_attribute(
            self.attributes_by_name["min_present_value"].id, 0.0
        )  # min_present_value

    def _update_attribute(self, attrid: int, value: Any) -> None:
        _LOGGER.debug("AnalogOutput update attribute %04x to %s... ", attrid, value)
        super()._update_attribute(attrid, value)
        if attrid == self.attributes_by_name["present_value"].id:
            self.endpoint.window_covering._update_attribute(  # pylint: disable=protected-access
                WindowCovering.attributes_by_name[
                    "current_position_lift_percentage"
                ].id,
                (100 - value),
            )

class WindowCoveringB1(CustomCluster, WindowCovering):
    """Window covering cluster, used to cause commands to update the AnalogOutput present_value."""

    stop_called = bool(False)

    def __init__(self, *args: Any, **kwargs: Any) -> None:
[... 95 lines cut ...]
                    Time.cluster_id
                ],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    AnalogOutputB1,
                    BasicCluster,
                    Identify.cluster_id,
                    MultistateOutput.cluster_id,
                    XiaomiPowerConfigurationPercent,
                    WindowCoveringB1,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Time.cluster_id
                ],
            },
        },
    }`

blair287, 2024-03-12:
Is this now all working with current quick? Is it working for open close stop and position as a cover and have battery shown? Thanks.

closed 2024-09-15 by github-actions[bot]

m0t0rh3ad, 2024-10-10:
I've got HA with ZBT-1 stick and ZHA configured. My Aqara b1 is not working with this config. Device been successfully interviewed, but nothing else working

jarkko-hautakorpi, 2025-01-15:
Also same here, lumi.curtain.hagl04. Idenfify works and the device beeps. Open close not working. I ran it from the motor button from 0 to 100% to calibrate ends, but still no luck.