zigpy/zha-device-handlers #2483: [Device Support Request] Tuya Curtain Motor

AlterSu24, 2023-07-20:
### Problem description

Can be paired successfully but showed nothing to control with. No entities or sensors were created.

### Solution description

Full support of the device. Open/Close/Stop and controlling of open percentage.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[image]

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_xu4a5rhj",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.6.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
[... 471 lines cut ...]
```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

I bought this LS82 motors from taobao. And this motor seems to be the same or similar with DS82 motor which should be supported by ZHA. (https://zigbee.blakadder.com/Tuya_DS82.html)

So maybe this can also be supported by ZHA?

Any tips would be helpful, and I am willing to offer more detail informations.

Please help, thanks.

besiktas97, 2023-07-20:
I have a zigbee blind motor _TS0601_TZE200_cpbo62rn, device can be added without issues, but cant see entities or controlling it.
As signature from ZHA, I can see the following:

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_cpbo62rn",
  "model": "TS0601",
  "class": "zigpy.device.Device"

javicalle, 2023-07-27:
> I have a zigbee blind motor _TS0601_TZE200_cpbo62rn, device can be added without issues, but cant see entities or controlling it.

I would suggest to add your device to the `TuyaZemismartSmartCover0601_3` quirk:
* https://github.com/zigpy/zha-device-handlers/blob/fb00099826980eb52e1541764c17fcf5c8a306d8/zhaquirks/tuya/ts0601_cover.py#L163

Or maybe the inverted one:
* https://github.com/zigpy/zha-device-handlers/blob/fb00099826980eb52e1541764c17fcf5c8a306d8/zhaquirks/tuya/ts0601_cover.py#L209

AlterSu24, 2023-07-28:
@javicalle My zigbee cover motor is _TZE200_xu4a5rhj
where should I add my device?

AlterSu24, 2023-07-28:
class TuyaZemismartSmartCover0601_2(TuyaWindowCover):
    """Tuya Zemismart curtain cover motor."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # input_clusters=[0x0000, 0x000a, 0x0004, 0x0005, 0xef00]
        # output_clusters=[0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=81 input_clusters=[0, 10, 4, 5, 61184] output_clusters=[25]>
        MODELS_INFO: [
            ("_TZE200_3i3exuay", "TS0601"),

Just and my device info here? Or should I do anything else?

I tried to add  ("_TZE200_xu4a5rhj", "TS0601"), and deleted my device then paired it again, but nothing changed.
Still can't see any entities.

AlterSu24, 2023-07-28:
It didn't load with any quirks.
Zigbbe Info:
IEEE: a4:c1:38:ea:63:85:c9:2c
Nwk: 0xfc9e
Device Type: Router
LQI: 56
RSSI: -86
上次出現: 2023-07-28T13:25:44
電力來源: Mains

javicalle, 2023-07-28:
@AlterSu24 I couldn't see your device signature in the first post because there are some format error.

In your case you need a new quirk for your device (a GPP one).
Can you try with this one?:

<details open>
<summary>TuyaCoverTS0601_var04</summary>

```python
class TuyaCover0601_var04(TuyaWindowCover):
    """Tuya blind cover motor."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_xu4a5rhj", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

You will need to add new imports in the `ts0601_cover.py` file:
```python
"""Tuya based cover and blinds."""
from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Identify, OnOff, Ota, Scenes, Time
```

besiktas97, 2023-07-28:
> _TZE200_cpbo62rn

Thanks this worked for me. Maybe you also know this one. Is it possible to do something with the position of the cover(s)?
When I am testing it, the cover doesn't open fully or close fully and cant set it to position with 50% for example?

AlterSu24, 2023-07-28:
Thanks for your reply, it can be identified as a cover now!

But when I try to control it, it has no response at all.

The log showed the cover was opened and closed, but the cover didn't move.

When I control it with the remoter, it also shows the state in the log.

Here is my device signature:

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_xu4a5rhj",
  "model": "TS0601",
  "class": "ts0601_cover.TuyaCover0601_var04"
}

And here is my quirk:

"""Tuya based cover and blinds."""
from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Identify, OnOff, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)

class TuyaCover0601_var04(TuyaWindowCover):
    """Tuya blind cover motor."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_xu4a5rhj", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

Thanks again.

AlterSu24, 2023-08-01:
My device has no physical switch.
You can control it only by pulling the cover by hand or using the remote controller.
[image]

So here is the code and what I did is:
1. Open the cover by remote controller
2. Close the cover by remote controller
3. Press open and close in HA
4. Open and close the cover by hand

[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/12224712/home-assistant.log)
2023-08-01 08:56:36.200 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4F95](TS0001): Device seen - marking the device available and resetting counter
2023-08-01 08:56:36.200 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x4F95](TS0001): Update device availability -  device available: True - new availability: True - changed: False
2023-08-01 08:56:36.318 DEBUG (MainThread) [bellows.uart] Data frame: b'03e3b5a9362a9017216ab52d06258d617e'
2023-08-01 08:56:36.318 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2023-08-01 08:56:36.319 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingSenderEui64Handler: [70:ac:08:ff:fe:78:a5:85]
2023-08-01 08:56:36.319 DEBUG (MainThread) [bellows.zigbee.application] Received incomingSenderEui64Handler frame with [70:ac:08:ff:fe:78:a5:85]
2023-08-01 08:56:36.319 DEBUG (MainThread) [bellows.uart] Data frame: b'13e3b1a9112a11b259944a25aa5596499cc85b10c8309874f657debad2543fecf9cdde6f8d44767e'
2023-08-01 08:56:36.320 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2023-08-01 08:56:36.320 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=0, clusterId=0, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_SOURCE_EUI64: 1024>, groupId=0, sequence=134), 124, -69, 0xfe25, 255, 255, b'\x91\xbd3.*\x00K\x12\x00\x00\x00']
2023-08-01 08:56:36.320 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=0, clusterId=0, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_SOURCE_EUI64: 1024>, groupId=0, sequence=134), 124, -69, 0xfe25, 255, 255, b'\x91\xbd3.*\x00K\x12\x00\x00\x00']
2023-08-01 08:56:36.321 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xFE25), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=134, profile_id=0, cluster_id=0, data=Serialized[b'\x91\xbd3.*\x00K\x12\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=124, rssi=-69)
2023-08-01 08:56:36.321 DEBUG (MainThread) [zigpy.zdo] [0xfe25:zdo] ZDO request ZDOCmd.NWK_addr_req: [00:12:4b:00:2a:2e:33:bd, <AddrRequestType.Single: 0>, 0]
2023-08-01 08:56:36.322 DEBUG (Thread-4) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42fa0d0130>, 'UPDATE devices_v12\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1690851396.321448, 'ieee': 70:ac:08:ff:fe:78:a5:85, 'min_update_delta': 30.0})
2023-08-01 08:56:36.324 DEBUG (Thread-4) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42fa0d0130>, 'UPDATE devices_v12\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1690851396.321448, 'ieee': 70:ac:08:ff:fe:78:a5:85, 'min_update_delta': 30.0}) completed
2023-08-01 08:56:36.324 DEBUG (Thread-4) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42fa0d0130>)
2023-08-01 08:56:36.325 DEBUG (Thread-4) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f42fa0d0130>) completed
2023-08-01 08:56:36.370 DEBUG (MainThread) [bellows.uart] Data frame: b'23e3b1a90d2a304cdc3132db555d3e39e0f527cd937e'
2023-08-01 08:56:36.370 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2023-08-01 08:56:36.371 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingRouteRecordHandler: [0xfe25, 70:ac:08:ff:fe:78:a5:85, 124, -69, []]
2023-08-01 08:56:36.371 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0xfe25, 70:ac:08:ff:fe:78:a5:85, 124, -69, []]
2023-08-01 08:56:36.372 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0xfe25, 70:ac:08:ff:fe:78:a5:85, 124, -69, [])
2023-08-01 08:56:36.372 DEBUG (Thread-4) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42fa0d0130>, 'INSERT INTO relays_v12 VALUES (:ieee, :relays)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays WHERE relays != :relays', {'ieee': 70:ac:08:ff:fe:78:a5:85, 'relays': b'\x00'})
2023-08-01 08:56:36.373 DEBUG (Thread-4) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f42fa0d0130>, 'INSERT INTO relays_v12 VALUES (:ieee, :relays)\n                        ON CONFLICT (ieee)\n                        DO UPDATE SET relays=excluded.relays WHERE relays != :relays', {'ieee': 70:ac:08:ff:fe:78:a5:85, [... cut ...]
2023-08-01 08:56:49.261 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD455](TS0002): Device seen - marking the device available and resetting counter
2023-08-01 08:56:49.261 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD455](TS0002): Update device availability -  device available: True - new availability: True - changed: False
2023-08-01 08:56:50.240 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD7CF](TS0002): Device seen - marking the device available and resetting counter

@javicalle Thanks for your reply.
If you need more information, please let me know.

besiktas97, 2023-08-01:
> > Thanks this worked for me. Maybe you also know this one. Is it possible to do something with the position of the cover(s)? When I am testing it, the cover doesn't open fully or close fully and cant set it to position with 50% for example?
> 
> Some Tuya cover can configure the 'travel time' (`calibration_time`) but not your model. Any error in your logs? Can enable the debug logs and attach the device operation from HA and from the physical switch?
> 
> > But when I try to control it, it has no response at all.
> > The log showed the cover was opened and closed, but the cover didn't move.
> > When I control it with the remoter, it also shows the state in the log.
> 
> Does your device respond to the phisycal switch? If not, the problem would be in the device connections. If yes, then enable the debug logs and attach the device operation from HA and from the physical switch.

Hey Javicalle,

I can see from the developer tools that it must have 15 supported features, see below screenshots:

[image]

What do you mean by device operation from HA and from the physical switch ?
With the physical switch its doing the same, like in HA pressing the open or close button.

The debug log files:

[home-assistant_zha_2023-08-01T13-31-13.897Z.log](https://github.com/zigpy/zha-device-handlers/files/12229837/home-assistant_zha_2023-08-01T13-31-13.897Z.log)

Thanks again Javicalle, appreciate your help!

closed 2024-02-07 by github-actions[bot]