zigpy/zha-device-handlers #2486: [Device Support Request]

Mr-Alicates, 2023-07-22:
### Problem description

I bought a [Garza smart TRV](https://www.amazon.es/gp/product/B09MKTBXJ5/ref=ppx_yo_dt_b_asin_image_o00_s00?ie=UTF8&psc=1).

I paired it initially with home assistant without any issue, but the device had no entities associated. The device was reported as a "TS0601" manufactured by "_TZE200_mz5y07w2".

I started investigating and found that the device is phisically very similar to this [zigbee device entry](https://zigbee.blakadder.com/RTX_ZB-RT1.html).

I also found that there is an existing [class for this device](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv_sas.py)

I downloaded the tuya quirks, modified it manually to include my device:

`
class Thermostat_TZE200_c88teujp(TuyaThermostat):
    signature = {
        MODELS_INFO: [
            ("_TZE200_c88teujp", "TS0601"),
            ("_TZE200_azqp6ssj", "TS0601"),
            ("_TZE200_yw7cahqs", "TS0601"),
            ("_TZE200_9gvruqf5", "TS0601"),
            ("_TZE200_zuhszj9s", "TS0601"),
            ("_TZE200_zr9c0day", "TS0601"),
            ("_TZE200_0dvm9mva", "TS0601"),
            ("_TZE200_h4cgnbzg", "TS0601"),
            ("_TZE200_exfrnlow", "TS0601"),
            ("_TZE200_9m4kmbfu", "TS0601"),
            ("_TZE200_3yp57tby", "TS0601"),
            ("_TZE200_mz5y07w2", "TS0601"),
        ],
`

Restarted HA, re-paired the device and all the device's functionality seems to be working:
* I have a working temperature offset slider 
* I have a working thermostat entity that reports current temperature, target temperature and action (OFF / Heating)
* An "HVAC action" sensor that does not seem to work as it is always reporting "unknown"
* I have a working battery sensor (at least is reporting 100% which makes sense because the device is brand new)

The only issue is that HVAC action sensor that seems to be stuck in "Unknown"

        
        

### Solution description

Adding official support for the device _TZE200_mz5y07w2.
It seems to be almost working with that existing quirk.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x000d",
[... 477 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2023-07-22:
The latest good working tuya TRV quirks is hosted here https://github.com/jacekk015/zha_quirks and shall being at the newest stand and all functions implanted.

Mr-Alicates, 2023-07-22:
Hello @MattWestb.

Thank you very much for pointing me to that repo. Using [this](https://github.com/jacekk015/zha_quirks/blob/main/trv_saswell.py) quirk I got it working fine.

Now the TRV displays a number slider for the temperature offset, a thermostat entity and an HVAC action sensor that seems to work fine since it displays idle and heating and the battery.

I also get 6 switches control that I have no clue what they do, but they seem to have a reaction on the valve.

What should I do now? Should I move this request to @jacekk015's repository?

MattWestb, 2023-07-22:
Great  that is working good for you !!
Plea ask Jacekk and i think hi is back home after his holiday trip and i hope hi have time explaining the functions for you.

Mr-Alicates, 2023-07-22:
I managed to find out what each switch does by peeking on the attributes of the device.
They seem to be:
Window detection
Child lock
Anti freeze
Limescale protection
Schedule mode
Away mode

Thanks again!

jacekk015, 2023-07-23:
Can't give the names to switches. ZHA Code limitation.