zigpy/zha-device-handlers #2488: [Device Support Request] Hue Ambiance Gradient Lightstrip

scottgerring, 2023-07-25:
### Problem description

The [Hue Ambiance Gradient lightstrip](https://www.philips-hue.com/de-de/p/hue-white---color-ambiance-gradient-ambiance-lightstrip-basis-set-2-meter/8719514339965) currently shows up as a single light, but is configurable to some degree in segments. 

### Solution description

Something like what deconz and [zigbee2mqtt ](https://github.com/Koenkk/zigbee2mqtt/issues/10674)are doing - it would be great to configure the different colours of the strip to get the gradient effect. 

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4107, maximum_buffer_size=82, maximum_incoming_transfer_size=128, server_mask=11264, maximum_outgoing_transfer_size=128, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 260,
      "device_type": "0x010d",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xfc01",
        "0xfc03"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Signify Netherlands B.V.",
  "model": "LCX004",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
 "data": {
    "ieee": "**REDACTED**",
    "nwk": 46461,
    "manufacturer": "Signify Netherlands B.V.",
    "model": "LCX004",
    "name": "Signify Netherlands B.V. LCX004",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4107,
    "power_source": "Mains",
    "lqi": 255,
    "rssi": -56,
    "last_seen": "2023-07-25T18:09:36",
[... 250 lines cut ...]
        }
      },
```

</details>

### Logs

<details><summary>Logs</summary>

```python

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python

```

</details>

th3cube, 2024-04-11:
Just moved to ZHA to realize that this still isn’t solved. In Z2M you can use the Gradient Features and Effects like Flame. In ZHA this expensive light is downgraded to a 10$ lightstrip. Moving back to Z2M

edgegit, 2024-07-02:
Surprised to find such little activity on this request. Would be great if I didn't have to switch to Z2M...

Trurl101, 2024-08-07:
The missing support for gradient lightstrips and the Tap Switch (Green Power Zigbee) are the only reasons why I can't get rid of my hue bridge.

Rijndael1998, 2024-09-25:
I got one myself yesterday and the gradient didn't work. I mean no pressure on the devs but I really would appreciate support for this. There's a [zigbee2mqtt issue](https://github.com/Koenkk/zigbee2mqtt/issues/10674) that has been solved. Maybe the solution could help with this?

Either way, I love you devs. That's for your amazing project.

thorbolo, 2024-09-27:
I am too really hoping this Ticket finally finds a way one the roadmap, pleeease :)

taarraas, 2024-11-11:
just configured 32 lamps into ZHA. signe lamp was last one, I would hate to move now to z2m :(

jw4949, 2024-12-25:
would also be interested in this working with ZHA

Orange-GT3, 2026-01-28:
I don't see the commands that you see for my gradient light strip:

[image]

Have you added your own quirk?

Orange-GT3, 2026-01-29:
> I think there are `LCX004` and `LCX007` variants that aren't listed,

Yes, I have the `LCX004` variant.

tkodev, 2026-02-23:
There are other devices not covered by these IDs

https://www.philips-hue.com/en-ca/p/hue-white-and-color-ambiance-play-gradient-light-tube-compact/046677803537

https://www.philips-hue.com/en-ca/p/hue-white-and-color-ambiance-play-gradient-light-tube-large/046677582500

Any chance they can be supported too? I can provide any logs, dumps, etc required if given instruction