zigpy/zha-device-handlers #2489: [BUG] Diffrent handling for _TZ3000_kmsbwdol TS130F

zolakt, 2023-07-25:
### Bug description

I bought 2 curtain switches, a few week apart. I bought identical ones.

The first one that I got works fine. I've added it to zha. It loaded from the quirk. I can calibrate it from the TuyaCoveringCluster.
Here is the signature of that device:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_kmsbwdol",
  "model": "TS130F",
  "class": "zhaquirks.tuya.ts130f.TuyaTS130Double_GP"
}
```

I just hooked up the second device, expecting everything will be the same, but it's not. Even though it's the same manufacturer and same model. It doesn't load if from the quirk. I can't see the TuyaCoveringCluster and I can't calibrate it. I can only see the WindowCovering cluster where I can't do any configuration. Also, open/close commands seem to be reversed.
There is the signature of that device:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102",
        "0xe001"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_kmsbwdol",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}
```

So, how can I load the second device using the same zhaquirks.tuya.ts130f.TuyaTS130Double_GP quirk as the first one, so I can calibrate it via the TuyaCoveringCluster?

### Steps to reproduce

1. Hooked up both devices
2. One gets loaded with a quirk, the other without

### Expected behavior

Both devices should be loaded using the zhaquirks.tuya.ts130f.TuyaTS130Double_GP quirk

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102",
        "0xe001"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_kmsbwdol",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

zolakt, 2023-07-26:
I got it working with a custom quirk.
The problem was that PROFILE_ID was not matching any of the signatures of the built in quirks.
If anyone needs it, here it is:

```
"""Device handler for loratap TS130F smart curtain switch."""
from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import TuyaZBExternalSwitchTypeCluster

from zhaquirks.tuya.ts130f import (
    TuyaWithBacklightOnOffCluster,
    TuyaCoveringCluster,
)

class TuyaTS130Double_GP_ESTC_2(CustomDevice):
    """Tuya double smart curtain roller shutter with Green Power and External Switch Type Cluster."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=0x0104, device_type=0x0202, device_version=1, input_clusters=[0, 4, 5, 6, 0x0102, 0xE001], output_clusters=[0x000a, 0x0019]))
        MODEL: "TS130F",
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    WindowCovering.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                # "profile_id": 0x0104, "device_type": "0x0202",
                # "in_clusters": ["0x0004","0x0005","0x0006","0x0102", 0xE001],
                # "out_clusters": []
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    WindowCovering.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=0x0061
                # input_clusters=[]
                # output_clusters=[0x0021]
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [0x0021],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWithBacklightOnOffCluster,
                    TuyaCoveringCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWithBacklightOnOffCluster,
                    TuyaCoveringCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

Could you guys maybe add this to your default ones, so we don't have to use a custom quirk?
It's completely the same as your TuyaTS130Double_GP_ESTC class. Only the PROFILE_IDs are slightly different.

zolakt, 2023-07-27:
Sure.
The only problem is that I don't know all the constants for the hex values, like: 0x0104, 0x0061, 0x0021.
I could leave them as hex, but I assumed you wanted to use constants.

javicalle, 2023-07-28:
That's the easy part 😉:
<details open>
<summary>Another Tuya TS130F Double cover</summary>

```python
class TuyaTS130Double_GP_ESTC_2(CustomDevice):
    """Tuya double smart curtain roller shutter with Green Power and External Switch Type Cluster."""

    signature = {
        MODEL: "TS130F",
        ENDPOINTS: {
            1: {
                # profile=0x0104, device_type=0x0202, 
                # input_clusters=[0, 4, 5, 6, 0x0102, 0xE001], 
                # output_clusters=[0x000a, 0x0019]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    WindowCovering.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                # "profile_id": 0x0104, "device_type": "0x0202",
                # "in_clusters": ["0x0004","0x0005","0x0006","0x0102", 0xE001],
                # "out_clusters": []
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    WindowCovering.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=0x0061
                # input_clusters=[]
                # output_clusters=[0x0021]
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWithBacklightOnOffCluster,
                    TuyaCoveringCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaWithBacklightOnOffCluster,
                    TuyaCoveringCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

zolakt, 2023-07-28:
@javicalle I've created a pull request [here](https://github.com/zigpy/zha-device-handlers/pull/2494)

zolakt, 2023-07-28:
Ok. Either way, good that it's sorted out. As long as my blinds are working, I don't care how :)
So this will be included in the next HA Core release, I assume?

javicalle, 2023-07-28:
Yes, it should usually go this way.
As the PR is already merge I will bet that will be in the next _minor_ HA Core release.
Remember to delete your local quirk once updated 😉

javicalle, 2023-07-28:
Could you close the PR and the issue?

Thanks in advanced.

closed 2023-07-28 by zolakt