zigpy/zha-device-handlers #2490: [Device Support Request] TS130F ZemiSmart no applying quirk

jfisbein, 2023-07-26:
### Problem description

I have a TS130F from ZemiSmart not covered by the existing TS130F quirk, so it's detected by HA but I cannot calibrate it, making it unuseful.

Here is the zigbee info from HA

### Solution description

Add support for this model/manufacturer in the quirk

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.7.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Madrid",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.07.1",
    "host_os": "Home Assistant OS 10.3",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "sonoff": {
      "version": "3.5.1",
      "requirements": [
        "pycryptodome>=3.6.6"
      ]
    },
    "spotcast": {
      "version": "v3.6.30",
      "requirements": []
    },
    "plant": {
      "version": "2.1.1",
      "requirements": []
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
[... 562 lines cut ...]
                "attribute_name": "installed_closed_limit_tilt"
              },
              "0x0015": {
                "attribute_name": "acceleration_time_lift"
              },
              "0x0018": {
                "attribute_name": "intermediate_setpoints_lift"
              }
            }
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

javicalle, 2023-07-27:
A new `TS130F` signature.
Maybe this quirk can do the job
<details open>
<summary>TuyaTS130F_var08</summary>

```python
class TuyaTS130F_var08(CustomDevice):
    """Tuya smart curtain TS130F (variation 08)."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0x0202, device_version=1, input_clusters=[0, 4, 5, 6, 10, 0x0102], output_clusters=[25]))
        MODEL: "TS130F",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    WindowCovering.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
```

</details>

You would need to add the new `Identify` import to the current `ts130f.py` quirk:
```python
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
```

gilbertorconde, 2023-08-30:
I have a TS130F ZemiSmart module (ZN-LC1E) but for the quirk to work had to be like this:
```python
class ZemismartTS130F(CustomDevice):
    """ZemiSmart smart curtain roller shutter."""

    signature = {
        # SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=0x0202, device_version=1, input_clusters=[0x0000, 0x0004, 0x00>
        MODEL: "TS130F",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Identify.cluster_id,
                    Scenes.cluster_id,
                    WindowCovering.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Identify.cluster_id,
                    Scenes.cluster_id,
                    TuyaCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }
```

closed 2024-03-04 by github-actions[bot]