zigpy/zha-device-handlers #2492: [Device Support Request]

gkawanami, 2023-07-28:
### Problem description

Hey to all!

I have recently paired a Tuya TS0601 with 8 gangs, bought at [Aliexpress](https://www.aliexpress.us/item/3256805462800641.html?spm=a2g0o.order_list.order_list_main.54.7f39caa4hs12SL&gatewayAdapt=glo2usa4itemAdapt), and using instructions in the [previously reported issue](https://github.com/zigpy/zha-device-handlers/issues/1634) the manufacturer was included and I was able to get 6 of the 8 buttons to work.  The last two are still not responsive thru HA. (Physically working, but not reported to HA).

Running
Home Assistant 2023.7.3
Supervisor 2023.07.1
Operating System 10.3
Frontend 20230705.1 - latest
Device: Micro PC -  Generic x86-64 
ZHA thru Sonoff dongle E (flashed)

### Solution description

How can I make the last two button work?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0006"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0006"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0006"
[... 1891 lines cut ...]
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            8: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

### Tuya Zb TZ3000 devices (TS0003 and TS0004) #####
```

</details>

javicalle, 2023-07-28:
I don't know how have you implemented the `TuyaOctapleSwitchTO_GP` quirk (there are no reference in the related issue) but to make the 8 buttons work, the `TuyaOnOffManufCluster` class needs to be extended adding the adicional DPs:
* https://github.com/zigpy/zha-device-handlers/blob/fb00099826980eb52e1541764c17fcf5c8a306d8/zhaquirks/tuya/mcu/__init__.py#L404

gkawanami, 2023-07-28:
Hey @javicalle !

Thanks for the answer.  
I'm a total mess with this thing....

I tried inserting lines 404 to the end at "####  New Edit by Gus ####" position but that didn't work

Where should I insert these lines?

javicalle, 2023-07-28:
Just for testing purposes you can try with:

<details open>
<summary>TuyaOnOffManufCluster_8</summary>

```python
class TuyaOnOffManufCluster_8(TuyaMCUCluster):
    """Tuya with On/Off data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        2: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=2,
        ),
        3: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=3,
        ),
        4: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=4,
        ),
        5: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=5,
        ),
        6: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=6,
        ),
        7: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=7,
        ),
        8: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=8,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        6: "_dp_2_attr_update",
        7: "_dp_2_attr_update",
        8: "_dp_2_attr_update",
    }
```

</details>

You would need at least also import the `DPToAttributeMapping` and `TuyaMCUCluster` classes:
```python
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    MoesSwitchManufCluster,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)
```

Also the `Dict` one:
```python
from typing import Dict
```

Then, your quirk will be something like:

<details open>
<summary>TuyaOctapleSwitchTO_GP</summary>

```python
class TuyaOctapleSwitchTO_GP(TuyaSwitch):
    """Tuya octaple channel switch time on out cluster device with GreenPowerProxy cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_wktrysab", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # input_clusters=[0x0000,0x0004,0x0005,0xEF00]
            # output_clusters=[0x000A,0x0019]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster_8,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            6: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            8: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

</details>

The `TuyaOnOffManufCluster_8` must be upper in file than the quirk.
Be sure to delete any `__pycache__` folder from your local quirk folder.

gkawanami, 2023-07-28:
Hi @javicalle 

Edited the quirk file (below) but still no response from the last two buttons.  

I inserted the dict reference, but I don't know if I got what to do with it. Put it just below thezhaquirks import, in the beginning of the file.

Restarted and still non responsive buttons.

 What am I doing wrong?

`"""Tuya DP based switches."""
from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    MoesSwitchManufCluster,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)

from typing import Dict

class TuyaSingleSwitchTI(TuyaSwitch):
    """Tuya single channel switch time on in cluster device."""

    signature = {
        # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4098
        #                       maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264
        #                       maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
        # device_version=1
        # input_clusters=[0x0000,0x0004, 0x0005,0x000a, 0xef00]
        # output_clusters=[0x0019]
        # <SimpleDescriptor endpoint=1 profile=260 device_type=81 device_version=1 input_clusters=[0, 4, 5, 10, 61184] output_clusters=[25]>
        MODELS_INFO: [("_TZE200_7tdtqgwv", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Time.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Time.cluster_id,
                    Groups.cluster_id,
[... 828 lines cut ...]
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            8: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
### Tuya Zb TZ3000 devices (TS0003 and TS0004) #####`

gkawanami, 2023-07-31:
Hi @javicalle 

I decided to remove the previous file (the ones that make the 4 and 6 gang switches work), and created a new file in my quirks folder with just the "simplified version" you sent me in your last message.  Restarted.
After that I uploaded the original quirk file and substituted the entire “8 gang” portion with your last message’s “simplified version”, restarted again.
Removed the modified file and re-uploaded the original quirk file.  Restarted.

The 3 methods made the same following error show up:

> Invalid entity ID at position 2: start
> type: logbook
> entities:
>   - light.sw_porta_de_cima_light_7
>   - light.sw_porta_de_cima_light_8
>   - start
>   - update.home_assistant_operating_system_update
> title: LOGS

None os the 8 gangs are "lit" or are clickable on integration/devices screen.  Pressing "reconfigure" shows no errors, but makes no difference in allowing me to click any of the gangs.
They only become lit and are clickable with the original quirk file (the one on my first post)

Please help!

javicalle, 2023-08-01:
Just to be sure: my 'simplified version' must be the full content of the quirk file. You can delete your current quirk file (or rename with a diferent extension) and put another file (the name doesn't matter) with that content.

Check for errors in the `homeassistant.log` file. If there are format errors in file the logs will tell to you. 
Put your logs and comments inside ` ``` ` marks:
````
```
All your code/comments/logs here
```
````

gkawanami, 2023-08-01:
Hi again @javicalle 

OK, renamed all the extensions to .inactive 
 Created one new "JaviCalleFile.py" with just the copied "simplified version".

Log:
```
2023-08-01 11:55:29.237 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-08-01 11:55:40.080 ERROR (MainThread) [zhaquirks] Unexpected exception importing custom quirk 'JaviCalleFile'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 454, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 605, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1120, in load_module
  File "<frozen importlib._bootstrap_external>", line 945, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 721, in _load
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 936, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1074, in get_code
  File "<frozen importlib._bootstrap_external>", line 1004, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/JaviCalleFile.py", line 84
    """Tuya octaple channel switch time on out cluster device with GreenPowerProxy cluster device."""
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
IndentationError: expected an indented block after class definition on line 83
2023-08-01 11:55:40.759 ERROR (bellows.thread_0) [bellows.uart] CRC error in frame b'51a8bd7e' (b'a8bd' != b'ab24')
```

None of the TS0601 (6 ou 8 gangs) are responding.
Clicking reconfigure results in:
```
The device reconfiguration failed. Additional information may be available in the logs.
```

javicalle, 2023-08-01:
The class is not loading because the code error.
I have updated the code but you need just to indent the `TuyaOctapleSwitchTO_GP` comment like this:
```python
class TuyaOctapleSwitchTO_GP(TuyaSwitch):
    """Tuya octaple channel switch time on out cluster device with GreenPowerProxy cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_wktrysab", "TS0601"),
        ],
        .../...
```

Save changes, restart HA and pair it again.

gkawanami, 2023-08-01:
Made the changes.

In the end, the resulting section is:
```

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        6: "_dp_2_attr_update",
        7: "_dp_2_attr_update",
        8: "_dp_2_attr_update",
    }

class TuyaOctapleSwitchTO_GP(TuyaSwitch):
    """Tuya octaple channel switch time on out cluster device with GreenPowerProxy cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_wktrysab", "TS0601"),
        ],

        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # input_clusters=[0x0000,0x0004,0x0005,0xEF00]
            # output_clusters=[0x000A,0x0019]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
```

Restarted, repaired.
The lights are now clickable at the devices screen; but clicking the last 2 buttons still don't get transmitted to the wall switch.

Logs:
```
2023-08-01 16:31:31.212 WARNING (MainThread) [zhaquirks] Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers
2023-08-01 16:31:32.420 ERROR (bellows.thread_0) [bellows.uart] CRC error in frame b'f4df5589fd7e3fa7ebcd02457e' (b'0245' != b'56e4')
2023-08-01 16:31:32.421 ERROR (bellows.thread_0) [bellows.uart] CRC error in frame b'1bc3b1a90d2aee2eebcdda86987e' (b'8698' != b'619b')```

@javicalle thanks for the help so far...  I know its taking you some time..  Anything else I could try?

TheJulianJES, 2023-08-01:
Z2M has different datapoint IDs for two of the buttons:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/77547a0888dbc5f955027c0bdfb11543d99bf202/src/devices/tuya.ts#L5104-L5113

I'd try changing those two in your quirk and see if it works.

gkawanami, 2023-08-01:
Hi @TheJulianJES 

Really don't know how to do that..

Tried this in @javicalle's Simplified Code, adjusted.  Still the non responsive  last two buttons.

Is this what I was supposed to do with the quirk?

```
class TuyaOnOffManufCluster_8(TuyaMCUCluster): 
    """Tuya with On/Off data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        2: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=2,
        ),
        3: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=3,
        ),
        4: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=4,
        ),
        5: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=5,
        ),
        6: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=6,
        ),
        7: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=0x65,
        ),
        8: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=0x66,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        6: "_dp_2_attr_update",
        0x65: "_dp_2_attr_update",
        0x66: "_dp_2_attr_update",
    }

```

TheJulianJES, 2023-08-01:
Not really familiar with the Tuya stuff, but from a quick look at it, you could try this:
```python
class TuyaOnOffManufCluster_8(TuyaMCUCluster): 
    """Tuya with On/Off data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        2: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=2,
        ),
        3: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=3,
        ),
        4: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=4,
        ),
        5: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=5,
        ),
        6: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=6,
        ),
        0x65: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=7,
        ),
        0x66: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=8,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        6: "_dp_2_attr_update",
        0x65: "_dp_2_attr_update",
        0x66: "_dp_2_attr_update",
    }
```

gkawanami, 2023-08-02:
Hey guys!

Well, JaviCalle.py is the file which contains exclusively the @javicalle simplified version, with the substitution provided by @TheJulianJES.  

After rebooting the logs showed:

```
2023-08-02 13:15:38.363 ERROR (MainThread) [zhaquirks] Unexpected exception importing custom quirk 'JaviCalle'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 454, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 605, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1120, in load_module
  File "<frozen importlib._bootstrap_external>", line 945, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 721, in _load
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/JaviCalle.py", line 23, in <module>
    class TuyaOnOffManufCluster_8(TuyaMCUCluster):
  File "/config/custom_zha_quirks/JaviCalle.py", line 26, in TuyaOnOffManufCluster_8
    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
                     ^^^^
NameError: name 'Dict' is not defined
2023-08-02 13:15:39.419 ERROR (bellows.thread_0) [bellows.uart] CRC error in frame b'f6de2f83fd7e1f240932fe768b29027e' (b'2902' != b'ae90')
```

Line 26 reads: 
```
    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
```

And this is the entire file:
```
"""Tuya DP based switches."""
from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    MoesSwitchManufCluster,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)

class TuyaOnOffManufCluster_8(TuyaMCUCluster): 
    """Tuya with On/Off data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        2: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            endpoint_id=2,
        ),
[... 134 lines cut ...]
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            8: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

After the reboot
All button are grey and non clickable....

javicalle, 2023-08-03:
Thanks @TheJulianJES !!!

@gkawanami, you left the `Dict` import from my file:
```python
"""Tuya DP based switches."""
from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    MoesSwitchManufCluster,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)

from typing import Dict

class TuyaOnOffManufCluster_8(TuyaMCUCluster): 
    """Tuya with On/Off data points."""
```

javicalle, 2023-08-05:
@gkawanami Did you manage to get it working without problems?

gkawanami, 2023-08-05:
Hey @javicalle 

I'm out of town.  I'll be back home in a couple of days.

Will surely test it as soon as I get home.

gkawanami, 2023-08-07:
Ok!!!

@TheJulianJES 
@javicalle 

JaviCalle file is now working great!  
I've deleted the eight gang section of the original file, and that got all the switches (3, 4 and 6 gangs) working thru the original file and the 8 gang switch is working thru the new file you both created!  

Problem solved!  Thanks to you both a lot!!!!!!

Is there any problem in having 2 or more files in the quirks folder?  Should I try to merge both files?

javicalle, 2023-08-07:
> Is there any problem in having 2 or more files in the quirks folder?

No, no problem at all as long as you don't repeat definitions of the same device in both files.
If you are not using any custom code from the 3, 4 and 6 gangs devices, you can remove the file from your `custom_quirk` folder and just keep the 8 gang file.

> Should I try to merge both files?

Next steps would be to create a PR to integrate that code to mainstream.
I'm not sure what to do with that new `TuyaOnOffManufCluster_8` class. We can just update the original `TuyaOnOffManufCluster` with the 7th and 8th DP, but not sure if these are quite Tuya 'standar' values.

Once the PR for your device is created and merged, you would remove the file with the next HA minor update (the nex 2023.X version after PR gets merged).

gkawanami, 2023-08-07:
Great!

Problem solved then!

Thanks very much for the support!

matthewcoates, 2023-10-02:
thank you guys soooo much i have been working on getting my 16ch tuya relay board working for months and just came across this, and with a little editing i got all 16ch to work
[ts0601_switch_8.txt](https://github.com/zigpy/zha-device-handlers/files/12784894/ts0601_switch_8.txt)

javicalle, 2023-10-03:
Could anyone test with that version of quirk (or just replace all the `DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,` with `DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,`:

<details open>
<summary>TuyaSwitchX8_GP</summary>

```python

class TuyaSwitchX8_GP(TuyaSwitch):
    """Tuya x8 channels switch with GreenPowerProxy cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_wktrysab", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # input_clusters=[0x0000,0x0004,0x0005,0xEF00]
            # output_clusters=[0x000A,0x0019]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            6: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            8: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

```

</details>

DISCLAIMER: This change will create all new switch entities for the channels and the light entities will be unavailable.

PR #2616 (2023-10-04): Add new Tuya multi switch devices (x8, x16)
## Proposed change

Adds new implementations for the Tuya x8 and x16 switch devices

## Additional information

Fixes #2492
Fixes #2536

## Checklist

- [ ] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
commit 10a17d3b (2023-10-04): Add new Tuya multi switch devices (x8, x16) (#2616)

Fixes #2492
Fixes #2536
closed 2023-10-04 by TheJulianJES