zigpy/zha-device-handlers #2493: [Device Support Request] Fingerbot (_TZ3210_dse8ogfy, TS0001)

brettjurgens, 2023-07-28:
### Problem description

I purchased this [Zigbee Fingerbot from aliexpress](https://www.aliexpress.us/item/3256805414231144.html) and it has limited support in home assistant.

I've attempted to write a custom quirk based on the code from [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters/blob/8181aefa9344bac21a95d2033bee3c8833ec61e1/src/devices/tuya.ts#L5175-L5216). I can read most of the values, but writing does not work.

### Solution description

I'd like to be able to write the `mode`, `down_movement`, and `up_movement` data points.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

[image]

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0103",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_dse8ogfy",
  "model": "TS0001",
  "class": "ts0001_switch.TuyaFingerbot"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.7.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Los_Angeles",
[... 577 lines cut ...]
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff.cluster_id,
                    TuyaFingerbotCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
```

</details>

brettjurgens, 2023-08-02:
I was able to get this to work by overriding the command method and having it use `TUYA_SEND_DATA` instead of the default `TUYA_SET_DATA`

<details><summary>Updated Quirk</summary>
from typing import Any, Dict, Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TUYA_SEND_DATA
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaEnchantableCluster,
)

class FingerBotMode(t.enum8):
    CLICK = 0x00
    SWITCH = 0x01
    PROGRAM = 0x02

class FingerBotReverse(t.enum8):
    UP_ON = 0x00
    UP_OFF = 0x01

class TuyaFingerbotCluster(TuyaEnchantableCluster, TuyaMCUCluster):
    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update({101: ("mode", FingerBotMode)})
    attributes.update({102: ("down_movement", t.uint16_t)})
    attributes.update({103: ("sustain_time", t.uint16_t)})
    attributes.update({104: ("reverse", FingerBotReverse)})
    attributes.update({105: ("battery", t.uint16_t)})
    attributes.update({106: ("up_movement", t.uint16_t)})
    attributes.update({107: ("touch_control", t.Bool)})

    async def write_attributes(self, attributes, manufacturer=None):
        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

    async def command(
        self,
        command_id: Union[foundation.GeneralCommand, int, t.uint8_t],
        *args,
        manufacturer: Optional[Union[int, t.uint16_t]] = None,
        expect_reply: bool = True,
        tsn: Optional[Union[int, t.uint8_t]] = None,
        **kwargs: Any,
    ):
        """Override the default Cluster command."""

        return await super().command(
            TUYA_SEND_DATA,
            *args,
            manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID,
            expect_reply=expect_reply,
            tsn=tsn,
            **kwargs,
        )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(TuyaOnOff.ep_attribute, "on_off"),
        # Mode
        101: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "mode",
            converter=lambda x: FingerBotMode(x),
        ),
        # Down Movement
        102: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "down_movement",
        ),
        # Sustain Time
        103: DPToAttributeMapping(TuyaMCUCluster.ep_attribute, "sustain_time"),
        # Reverse
        104: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "reverse",
            converter=lambda x: FingerBotReverse(x),
        ),
        # Battery
        105: DPToAttributeMapping(TuyaMCUCluster.ep_attribute, "battery"),
        # Up Movement
        106: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "up_movement",
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        101: "_dp_2_attr_update",
        102: "_dp_2_attr_update",
        103: "_dp_2_attr_update",
        104: "_dp_2_attr_update",
        105: "_dp_2_attr_update",
        106: "_dp_2_attr_update",
        107: "_dp_2_attr_update",
    }

class TuyaFingerbot(EnchantedDevice):
    signature = {
        MODELS_INFO: [("_TZ3210_dse8ogfy", "TS0001")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaOnOff.cluster_id,
                    TuyaFingerbotCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff.cluster_id,
                    TuyaFingerbotCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
</details>

xLama, 2023-08-08:
It seems down_movement does not work for me

https://github.com/zigpy/zha-device-handlers/assets/8139881/368d91db-0c40-4178-94ec-80bbbdace3b5

abmantis, 2023-09-04:
> It seems down_movement does not work for me. up_movement does
> WhatsApp.Video.2023-08-08.at.12.33.52.mp4

I had the same issue initially, but after setting both down and up, down started working.
Try setting up to 20 and down to 80 and test again (it only works after the second test in my case).

vineetk32, 2023-09-17:
I have had this device for a couple of months, and it has suddenly stopped working. On/off control has suddenly stopped working, and I cannot figure out why. 

It doesn’t look like a hardware issue - I had a spare unopened device that I paired to ZHA, and that doesn’t work either. Doesn't seem like connection issue since I can read/write the up/down movement attributes (or the touch_control attribute) just fine.

Any pointers to troubleshoot?

pickonedev, 2023-12-13:
I don't know if this is the write place to ask, but I will do it, maybe someone knows...

I have the same product and I use it by doing a ZbSend command like this `zbsend {"Device":"0xD3E6","Endpoint":1,"Send":{"Power":1}}`  and the device will do the job, but I really want to know a command to use in order to make the device only press 50% of the way down, anyone knows how can I find all the commands or how to do this?

Thanks

pickonedev, 2023-12-15:
Nobody? :-(

xxLeoxx93, 2023-12-29:
@pickonedev 
Have you tried setting these parameters: https://github.com/zigpy/zha-device-handlers/issues/2493#issuecomment-1705745634

btw: how are you sending the command? for me the device becomes unresponsive after pairing. When I sart paring and the blue light flashes it gets connected but once it stops flashing (it flashes for a minute or so even after paring) it doesn't respond anymore

xxLeoxx93, 2023-12-29:
> I have had this device for a couple of months, and it has suddenly stopped working. On/off control has suddenly stopped working, and I cannot figure out why.
>
> It doesn’t look like a hardware issue - [...]

Since I have an issue in the same direction: Maybe it's the new firmware. Not sure if you updated it but mine is new ordered and this could be a reason for it.

micium, 2024-02-24:
https://github.com/TheJulianJES/zha-device-handlers/blob/be90f06ff57eac8a62f41a0cdbd753c441fd4d9d/zhaquirks/tuya/ts0001_fingerbot.py

closed 2025-03-26 by TheJulianJES