zigpy/zha-device-handlers #2497: New version of soil sensor - unrecognised

npa-homeassistant, 2023-07-29:
### Bug description

Hello, there seems to be a slightly new version of the TS0601 soil moisture and temperature sensor. Whereas all of the sensors on the original (_TZE200_myd45weu) are discovered, those on the new model (_TZE200_ga1maeof) are not. The device itself is detected and added, but the sensors are missing.

The two are physically identical, but clearly running slightly different software.

### Steps to reproduce

Add device as normal
It appears in HA, but the sensors are missing

### Expected behavior

Should show moisture, temperature and battery

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ga1maeof",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

npa-homeassistant, 2023-07-29:
Sorry, for some reason it didn't correctly paste the signature

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ga1maeof",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

TheJulianJES, 2023-07-30:
@npa-homeassistant Can you download this file (https://raw.githubusercontent.com/zigpy/zha-device-handlers/7ae60526729424db71949ae7c9ce12bba09b6a59/zhaquirks/tuya/ts0601_sensor.py -> SAVE using your browser / CTRL + S or CMD + S in most cases) and then [install it as a custom quirk](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274)?

(For that, you just have to create the `custom_zha_quirks` folder in your `/config` directory and add the zha config section to your `configuration.yaml`, then drop the downloaded file in that folder and restart Home Assistant.
If everything works correctly, the sensor should/might work now.
Let me know if that custom quirk works correctly, so we can get it merged into ZHA (so you can remove that custom quirk and it'll keep on working).

npa-homeassistant, 2023-07-30:
> @npa-homeassistant Can you download this file (https://raw.githubusercontent.com/zigpy/zha-device-handlers/7ae60526729424db71949ae7c9ce12bba09b6a59/zhaquirks/tuya/ts0601_sensor.py -> SAVE using your [...]

Yup! that seems to have done it. All three entities are now listed and updating. 

Thank you!

TheJulianJES, 2023-07-30:
Will be properly fixed with the next release of zha-quirks (likely 0.0.102). That version also needs to be updated in ZHA/HA Core then.
Not sure it'll make the next release of HA Core 2023.8.0, but if it's updated there, then you should delete your custom quirk.

closed 2023-07-30 by TheJulianJES

npa-homeassistant, 2023-07-31:
> Will be properly fixed with the next release of zha-quirks (likely 0.0.102). That version also needs to be updated in ZHA/HA Core then. Not sure it'll make the next release of HA Core 2023.8.0, but if it's updated there, then you should delete your custom quirk.

Ok, thank you.