zigpy/zha-device-handlers #2510: [Device Support Request] - TZE204_qasjif9e - mmWave Sensor

robert1993, 2023-08-04:
### Problem description

Wall mount 5v version of this product: https://nl.aliexpress.com/item/1005005477612297.html?spm=a2g0o.order_list.order_list_main.16.2a5879d214SME3&gatewayAdapt=glo2nld 

### Solution description

Support for the device. Not sure what to provide in order to be able to create quirk. It seems that there's already support for z2m for what it's worth. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_qasjif9e",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.8.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
[... 350 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

drashish87, 2023-08-04:
I have also spent more than 5 hours looking for a quirk to support this one!

jbustdel, 2023-08-05:
Try this, works ok but not regulations and no light sensor. 

""Human Presence Sensor 24GHz."""

from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import (
OccupancySensing,
IlluminanceMeasurement,
)
from zhaquirks.const import (
DEVICE_TYPE,
ENDPOINTS,
INPUT_CLUSTERS,
MODELS_INFO,
OUTPUT_CLUSTERS,
PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
DPToAttributeMapping,
TuyaMCUCluster,
)

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
"""Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
"""Tuya local IlluminanceMeasurement cluster."""

class MmwRadarManufCluster(TuyaMCUCluster):
"""Human Presence Sensor 24GHz."""

attributes = TuyaMCUCluster.attributes.copy()
attributes.update(
    {
        # ramdom attribute IDs
        0xEF65: ("duration_of_attendance", t.uint32_t, True),
        0xEF66: ("duration_of_absence", t.uint32_t, True),
    }
)

dp_to_attribute: Dict[int, DPToAttributeMapping] = {
    1: DPToAttributeMapping(
        TuyaOccupancySensing.ep_attribute,
        "occupancy"
    ),
    101: DPToAttributeMapping(
        TuyaMCUCluster.ep_attribute,
        "duration_of_attendance"
    ),
    102: DPToAttributeMapping(
        TuyaMCUCluster.ep_attribute,
        "duration_of_absence"
    ),
    113: DPToAttributeMapping(
        TuyaNewManufCluster.ep_attribute,
        "dp_113",
    ),
}

data_point_handlers = {
    1: "_dp_2_attr_update",
    101: "_dp_2_attr_update",
    102: "_dp_2_attr_update",
    113: "_dp_2_attr_update",
}
class MmwRadarMotion_var03(CustomDevice):
"""Millimeter wave occupancy sensor."""

signature = {
    #  endpoint=1, profile=260, device_type=81, device_version=1,
    #  input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
    MODELS_INFO: [
        ("_TZE200_0u3bj3rc", "TS0601"),
        ("_TZE200_mx6u6l4y", "TS0601"),
        ("_TZE200_v6ossqfy", "TS0601"),
        ("_TZE204_qasjif9e", "TS0601")
    ],
    ENDPOINTS: {
        1: {
            PROFILE_ID: zha.PROFILE_ID,
            DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
            INPUT_CLUSTERS: [
                Basic.cluster_id,
                Groups.cluster_id,
                Scenes.cluster_id,
                TuyaMCUCluster.cluster_id,
            ],
            OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
        },
        242: {
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            PROFILE_ID: 41440,
            DEVICE_TYPE: 97,
            INPUT_CLUSTERS: [],
            OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
        },
    },
}

replacement = {
    ENDPOINTS: {
        1: {
            PROFILE_ID: zha.PROFILE_ID,
            DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
            INPUT_CLUSTERS: [
                Basic.cluster_id,
                Groups.cluster_id,
                Scenes.cluster_id,
                MmwRadarManufCluster,
                TuyaOccupancySensing,
                TuyaAnalogInput,
                TuyaIlluminanceMeasurement,
            ],
            OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
        },
        242: {
            PROFILE_ID: 41440,
            DEVICE_TYPE: 97,
            INPUT_CLUSTERS: [],
            OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
        },
    }
}_

scotie, 2023-08-05:
There is another issue for a similar device request from a different manufacturer. I tried [This quirk](https://github.com/zigpy/zha-device-handlers/issues/2378#issuecomment-1559219239) and added a line of our device. The occupancy isn’t working, but the light sensor is. Maybe that helps. Didn’t have too much time to play with it

scotie, 2023-08-06:
OK, I had some time to play with this yesterday and made some progress.
Disclaimer: this is my first-ever attempt using quirks. 

@jbustdel, the code you posted doesn't work for me. 
the indentation seems wrong, and some declarations are missing. 

I took the one I mentioned above, with a working light sensor and modified it a bit.
[image]

The light sensor is working.
Occupancy is working, but taking some time to clear.

From the settings - only the `Number Sensitivity` seem to work, but I haven't yet tested how it affects the behavior.

[ts0601_.zip](https://github.com/zigpy/zha-device-handlers/files/12268946/ts0601_.zip)

jbustdel, 2023-08-06:
Great!!!!... works for me. But the option controls seems not to do nothing

scotie, 2023-08-06:
Could you try this one? 

Though I'm a bit skeptical :)
[ts0601_.zip](https://github.com/zigpy/zha-device-handlers/files/12269203/ts0601_.zip)

jbustdel, 2023-08-06:
Its similar. I think options not work yet. Thanks.

peverveer, 2023-08-08:
> OK, I had some time to play with this yesterday and made some progress. Disclaimer: this is my first-ever attempt using quirks.
>
> @jbustdel, the code you posted doesn't work for me. the indentation [...]

I can confirm that the illuminance and occupency sensors are both working fine for me. For the controls however, I am unsure whether these have any effect currently.

vipial1, 2023-08-08:
Same problem here.
With your quirk I can only have occupancy, not illuminance. And even occupancy looks never cleared (waiting for more than 5 minutes in another room).

pgroene, 2023-08-10:
The motion and light detection works, however the setting do not change anything. There is also distance measuring in the device. I'll check if i can add that.

It takes 7 seconds to clear when i leave the room. It would be nice to set the fading time.

jbustdel, 2023-08-10:
I use presence sensor with motion sensor. Its perfect combination. The presence sensor is too sensitive and fire also in other room or when a human pass in front of the door where its instaled. I use to fire the motion sensor and turn off light with presence sensor. Works perfect this way.

pgroene, 2023-08-11:
For me this one seems to work oke, I was able to change the setting (range and sensitivity and fading time)

https://github.com/zigpy/zha-device-handlers/issues/2378#issuecomment-1666843526

https://github.com/zigpy/zha-device-handlers/files/12269600/ts0601_qasjif9e.zip

pgroene, 2023-08-26:
> I use presence sensor with motion sensor. Its perfect combination. The presence sensor is too sensitive and fire also in other room or when a human pass in front of the door where its instaled. I use to fire the motion sensor and turn off light with presence sensor. Works perfect this way.

With the proposed pr you can set the distance and sensitity. I thing you no longer need the motion sensor.

Trikos, 2023-12-02:
Hi, all of yours devices are the 5.8GHz or 24GHz version?
Thank you

> For me this one seems to work oke, I was able to change the setting (range and sensitivity and fading time)
> 
> [#2378 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2378#issuecomment-1666843526)
> 
> https://github.com/zigpy/zha-device-handlers/files/12269600/ts0601_qasjif9e.zip

> Same problem here. With your quirk I can only have occupancy, not illuminance. And even occupancy looks never cleared (waiting for more than 5 minutes in another room).
> 
> EDIT: it is finally cleared...it takes 1-3 minutes.

Trikos, 2023-12-02:
> OK, I had some time to play with this yesterday and made some progress. Disclaimer: this is my first-ever attempt using quirks.
>
> @jbustdel, the code you posted doesn't work for me. the indentation [...]

I tried using this one and other quoted in this post but the sensors always state as CLEAR

mcanteri, 2023-12-14:
Hello,
I have two types of that sensor, one ceiling (240v) and one wall mount (5v) and both have the same model number _TZE204_qasjif9e. The ceiling one has both illuminance and presence that works, the wall mount just discovered,  only the presence. I am okay for now since the presence is the main purpose of that sensor, but if in future this quirk will work, I hope it'll be complete.
Here is my quirks that works:

`"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    DPToAttributeMapping,
    TuyaLocalCluster,
    TuyaManufCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster

ZONE_TYPE = 0x0001

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster."""

class NeoBatteryLevel(t.enum8):
    """NEO battery level enum."""

    BATTERY_FULL = 0x00
[... 346 lines cut ...]
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MmwRadarManufCluster,
                    TuyaOccupancySensing,
                    TuyaAnalogInput,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
`

MichalLytek, 2024-02-14:
I just received a new version of this sensor from Aliexpress. It still reports as `TS0601 by _TZE204_qasjif9e` but has improved position of light sensor and no LED on top:

[image]

[image]

I use the `ts0601_qasjif9e.py` quirk from this topic and the older version works ok, I have full configuration options, presense and lux sensors.

[ts0601_qasjif9e.zip](https://github.com/zigpy/zha-device-handlers/files/14279487/ts0601_qasjif9e.zip)

However, the new version does not report light sensor. 

[image]

I've checked in the "manage zigbee device" and I get the readings that coresponds to physically covering and uncovering the sensor.

[image]

What has to be changed in the quirk in order to make it work with the new model? The hex of cluster or endpoint id has changed? How to debug that? 🤔

igorsantos07, 2024-08-13:
_stale-proof comment_

Nick75wien, 2024-10-02:
Did anybody manage to get working controls? I have the ceiling version and do only get the sensors on presence and light but no controls. Wanted to adjust the sensitivity so that it doesn't detect when someone passes near.

pgroene, 2024-10-02:
Yes, I did have it working a while ago. You can find a working version somewhere in this pr.

thefunkygibbon, 2024-12-08:
anyone got a working quirk for this?  i can't get the last one which was pasted by @mcanteri  (that was formatted in a strange way) to work

naevtamarkus, 2024-12-17:
Hi all,
I've tried the version above (https://github.com/zigpy/zha-device-handlers/files/12269600/ts0601_qasjif9e.zip) 
from pgroene on Aug 11, 2023
And it works like a charm. I've tried the sensitivity, min and max distances and it does the right thing. The luminance sensor works as well.
It would be awesome to see this mainstream and not depend on a quirk. Thanks!

prairiesnpr, 2025-02-17:
This was added in #3612

closed 2025-02-17 by prairiesnpr