zigpy/zha-device-handlers #2514: [Device Support Request] Aqara E1 Double Switch ( With Neutral ) QBKG41LM

scotie, 2023-08-06:
### Problem description

I would like to expose additional functionality for this switch:

- Decoupled mode ( the most important one )
- Power Measurement
- Led Control
- Multiple / Long  / Both presses ( not sure this one is available )

[blakadder](https://zigbee.blakadder.com/Aqara_QBKG41LM.htmll)
[zigbee2mqtt](https://www.zigbee2mqtt.io/devices/QBKG41LM.html)

### Solution description

Custom quirk would work for me ( or help with creating one )
I tried using the [switch_aq2](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_aq2.py) and [switch_h1](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_h1.py) and replacing the signature with `lumi.switch.b2nc01`, but it doesn't seem to be recognized

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheJulianJES, 2023-08-31:
Please provide the device signature and/or device diagnostics for that switch.

scotie, 2023-10-17:
Hey @TheJulianJES ,

Sorry didn't get notified about this...
I added the correct signature

scotie, 2023-10-17:
Mabe relater to  #2251

ruilima, 2024-02-02:
@scotie did you find a solution for the Aqara E1 Double Switch?
I am trying to implement this device in zha with all functionalities but with no luck. i don't want to go to zigbee2mqtt but maybe will be the path.
Thanks

scotie, 2024-02-02:
@ruilima, unfortunately no.
I also don't want to move to `zigbee2mqtt` and it's really weird that this is not supported. I thought those switches are way more popular

ruilima, 2024-02-02:
The european version (https://zigbee.blakadder.com/Aqara_WS-EUK04.html) is supported but the E1 version have the same functionalities (except energy metter) and its cheaper. I Will try to implement it based on the european version before jump to zigbee2mqtt.

scotie, 2024-02-02:
Actually I removed all the quirks that I added when I was trying to play around with it a while ago and I do see some Endpoints defined. Maybe one of them already does what we want 🤔 

[image]

scotie, 2024-02-02:
@ruilima did you try [this](https://github.com/zigpy/zha-device-handlers/issues/2690) quirk ?