zigpy/zha-device-handlers #2516: [Device Support Request] TUYA? magnetic door sensor

njalon, 2023-08-08:
### Problem description

I've gotten a handful of [these](https://www.aliexpress.com/item/1005005445903141.html?spm=a2g0o.order_list.order_list_main.11.52681802Mal3vr) magnetic door sensors.
In HA they shows up as Tuya TS0203 sensors, but as soon as they are connected and paired they stop responding, and shows as offline.

comparing them to other picturs they dont look like the TS0203 sensors that is why i think it is in order with a device support request.

### Solution description

zha device support.

### Screenshots/Video

Image of device

[image]

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_au1rjicn",
  "model": "TS0203",
  "class": "zigpy.device.Device"
}

### Diagnostic information

[config_entry-zha-d1763dabeee5f882571625066c7a949c.json.txt](https://github.com/zigpy/zha-device-handlers/files/12295496/config_entry-zha-d1763dabeee5f882571625066c7a949c.json.txt)

I belive this is the most relevant information, but i m ight be wrong:

2023-08-07 17:57:41.647 DEBUG (MainThread) [zigpy.zcl] [0xDF39:1:0x0000] Received ZCL frame: b'\x18l\x01\x04\x00\x00B\x10_TZ3000_au1rjicn\x05\x00\x00B\x06TS0203'
2023-08-07 17:57:41.648 DEBUG (MainThread) [zigpy.zcl] [0xDF39:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=108, command_id=1, *direction=<Direction.Client_to_Server: 1>)
2023-08-07 17:57:41.650 DEBUG (MainThread) [zigpy.zcl] [0xDF39:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZ3000_au1rjicn')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0203'))])
2023-08-07 17:57:41.652 INFO (MainThread) [zigpy.device] [0xdf39] Read model 'TS0203' and manufacturer '_TZ3000_au1rjicn' from <Endpoint id=1 in=[power:0x0001, identify:0x0003, ias_zone:0x0500, basic:0x0000] out=[identify:0x0003, groups:0x0004, scenes:0x0005, on_off:0x0006, level:0x0008, lightlink:0x1000, ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
2023-08-07 17:57:41.652 INFO (MainThread) [zigpy.device] [0xdf39] Discovered basic device information for <Device model='TS0203' manuf='_TZ3000_au1rjicn' nwk=0xDF39 ieee=a4:c1:38:bc:1d:38:f0:8d is_initialized=True>
2023-08-07 17:57:41.652 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0203' manuf='_TZ3000_au1rjicn' nwk=0xDF39 ieee=a4:c1:38:bc:1d:38:f0:8d is_initialized=True>
2023-08-07 17:57:41.653 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZ3000_au1rjicn TS0203 (a4:c1:38:bc:1d:38:f0:8d)
2023-08-07 17:57:41.653 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2'>
2023-08-07 17:57:41.653 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 242} {1}
2023-08-07 17:57:41.654 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
2023-08-07 17:57:41.654 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1}
2023-08-07 17:57:41.654 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2023-08-07 17:57:41.654 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2023-08-07 17:57:41.655 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2023-08-07 17:57:41.655 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {232, 230} {1}
2023-08-07 17:57:41.655 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts0201.MoesTemperatureHumidtySensorWithScreen'>
[... 40 lines cut ...]
2023-08-07 17:57:41.682 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xDF39:1:0x0008]: Performing cluster binding
2023-08-07 17:57:41.682 DEBUG (MainThread) [zigpy.device] [0xdf39] Extending timeout for 0x74 request
2023-08-07 17:57:41.682 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDF39), dst_ep=0, source_route=None, extended_timeout=True, tsn=116, profile_id=0, cluster_id=<ZDOCmd.Bind_req: 0x0021>, data=Serialized[b't\x8d\xf08\x1d\xbc8\xc1\xa4\x01\x08\x00\x03(j\xab\xfe\xff\x8dy\xe0\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-08-07 17:57:41.682 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xDF39:1:0x0006]: Performing cluster binding
2023-08-07 17:57:41.683 DEBUG (MainThread) [zigpy.device] [0xdf39] Extending timeout for 0x76 request

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2023-08-08:
I think we need deep debug logging from the pairing or one sniffing of it for knowing whats going wrong here then its have the new Telink module ZTU and i have one LIDL and with Silabs and its still working OK then testing it for one minute ago.

PS: My is not doing all Zigbee 3 things like pull control so many things can going wrong we dont knowing like TC-Link key is being requested and not OK transmitted / acked and the device is leaving the network.

One thing is trying paring it with different  routers if you is having more like IKEA is normally very good Zigbee 3 and is supporting all advance futures and other is have  very limited functions implanted like most tuya ones and can making things working or not OK in the end.

elboletaire, 2023-08-11:
Duplicate of #2352 (?)

asbachb, 2024-01-29:
@njalon is this still an issue? I have the same sensor, but no problems with usage/pairing.

MikeEllis-personal, 2024-01-29:
I have a TS0203 (TZ3000_au1rjicn) and I'm having problems with it - pairing is hit-and-miss, often getting stuck at the "configuring" stage, and when it does work, the device drops off line a few minutes later having been working for those few minutes. Restoring connection seems to require a re-pairing attempt. Very happy to provide diagnostics and run tests etc. Downloaded diagnostics attached.
 
[zha-f07bd5d77ed3eb4a63b6b2e79160f890-_TZ3000_au1rjicn TS0203-38df84cd8e3279289be78b589bf1f666.json.txt](https://github.com/zigpy/zha-device-handlers/files/14087883/zha-f07bd5d77ed3eb4a63b6b2e79160f890-_TZ3000_au1rjicn.TS0203-38df84cd8e3279289be78b589bf1f666.json.txt)

njalon, 2024-02-01:
Yes and no, i've forgot about the isshue and moved over to zigbee2mqtt
where it was supported out of the box, but i do not have any issues going
back to zha to help out with hw. support in zha

Med vennlig hilsen / Best regards
Njål Lone

***@***.***
+47 943 17 662

Jumperm, 2024-03-17:
I have brought 3 of them an. I got the issue while the first configuration. It don't finish this process.
Sonoff Dongle E with EmberZnet NCP 7.4.1.0

closed 2024-09-20 by github-actions[bot]

regetskcob, 2024-11-04:
I also have a lot of issues with configuration these sensors, I bought a bunch of them since the tuna sockets work great with my ZHA instance, but these dorr/window sensors do not get configured properly.