zigpy/zha-device-handlers #2517: [Device Support Request] Hue bulbs missing Candle Effect

phbasler, 2023-08-09:
### Problem description

I have two Hue Go here which have a built in canlde effect.
According to https://github.com/Koenkk/zigbee2mqtt/issues/15699 it is possible to start this by sending certain commands to the cluster `0xfc03`.

This cluster supports the Canlde effect, the fireplace effect and also the color loop.
On my device page for the light I can only see the color loop effect.

If I go to manage zigbee device from the device page, then I don't have this cluster, only `0xfc01` in that range.
Is there a possibility to add these effects to zha so that we can call them from Home Assistant?

### Solution description

Some possibility to trigger the candle effect of the hue go through a script.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

nilsreiter, 2024-01-09:
I managed to active the candle effect using the ZCL command documented on [here](https://community.home-assistant.io/t/a-simple-single-automation-candle-effect-for-hue-white-ambience-and-other-bulbs/235023/15), on the following bulbs:

- LCT015, with bulb firmware 1.108.7, but not with 1.101.2
- LCT012, with bulb firmware 1.108.7, but not with 1.88.1
- LTO001, with bulb firmware 1.104.2
- 440400982842 (Playbar), with firmware 1.104.2
- LLC020, with firmware 67.101.2
- 1742830P7 (Lily), with 1.104.2, but not with 1.82.10
- LCL003 (Outdoor Lightstrip), with 1.93.7, but not 1.66.6

The following light does not seem to support the effect (tested with Hue app):
- LST002 (light strip), with firmware: 1.101.2
- LCT024 (Playbar), with firmware 1.108.5

BenJamesAndo, 2024-01-10:
Perhaps these issues would contain code that could be a good start?
https://github.com/zigpy/zha-device-handlers/pull/993
https://github.com/zigpy/zha-device-handlers/issues/769
I've never written a quirk so I'm not really sure.
I did test out the ZCL command on my LCA006 bulb with firmware 1.104.2 and it worked great.
It would be nice to also have fireplace. I believe there's a number of other ones as well.

nilsreiter, 2024-01-10:
Ah cool, thanks, will have a look at it.

Next to the candle effect, there is fireplace and "prism", which is basically color loop. They are documented here as well: https://github.com/Koenkk/zigbee2mqtt/issues/15699

Regarding the ZCL command, @BenJamesAndo, did you use the built-in zha.issue_zigbee_cluster_command or zha_toolkit.zcl_cmd, provided by zha_toolkit? Because the former didn't work for me.

BenJamesAndo, 2024-01-10:
I used zha_toolkit.zcl_cmd.
I tried zha.issue_zigbee_cluster_command just now but I got an Unknown error.

BetaRavener, 2024-01-29:
> I used zha_toolkit.zcl_cmd. I tried zha.issue_zigbee_cluster_command just now but I got an Unknown error.

I also got an "Unknown error" using the `zha.issue_zigbee_cluster_command`, but I was able to issue an identify command for the light using the same, so I wonder if I got some parameters wrong. I've used:
```
IEEE: <Copied from Device page>
Endpoint ID: 11
Cluster ID: 64515 (decimal for 0xfc03)
Cluster Type: in
Command: 0 (can be seen on the capture [here](https://github.com/Koenkk/zigbee2mqtt/issues/10674#issuecomment-1015559646))
Command Type: Server (this is the correct one that worked with identify)
```
And then  I couldn't figure out `Args` or `Params`. I've used couple thing in `Params`, such as `{multicolor: 0x21000101}`, `{multicolor: 21000101}` or `{multicolor: "21000101"}`, trying to emulate what people did with zigbee2mqtt, but none of those worked.

I was wondering if you (@BenJamesAndo) have figured out what to put there, since you say you tested the ZCL command and it worked for you, even though using zha_toolkit.

BenJamesAndo, 2024-01-30:
I unfortunately didn't make any progress beyond what anyone else has done. I only got `zha_toolkit.zcl_cmd` to work as others have, and never got anything out of `zha.issue_zigbee_cluster_command`.

BetaRavener, 2024-01-30:
> I only got zha_toolkit.zcl_cmd to work as others have

Could you point me to where this was described originally then? Because I either can't find it or am overlooking it.

BenJamesAndo, 2024-01-31:
https://community.home-assistant.io/t/a-simple-single-automation-candle-effect-for-hue-white-ambience-and-other-bulbs/235023/16

```
service: zha_toolkit.zcl_cmd
data:
  manf: 4107
  ieee: light.philips_go_light
  endpoint: 11
  cluster: 0xfc03
  dir: 0
  cmd: 0x00
  args:
  - 0x21
  - 0x00
  - 0x01
  - 0x01
```

nilsreiter, 2024-01-31:
It doesn't seem to work with `zha.issue_zigbee_cluster_command`. My wild guess is that it has to do with the `manf` argument. Using `zha_toolkit.zcl_cmd`, it only works with this argument, and not without. If I use `zha.issue_zigbee_cluster_command` in the development page, it always converts the `manf` value to a string, which may not be what's expected.

BetaRavener, 2024-02-01:
I've tracked down why this doesn't work with `zha.issue_zigbee_cluster_command` with help of logs. The thing tries to use the `command` as index to the `commands` hash-table but it doesn't know about any commands (otherwise I'd just use the "Manage Zigbee Device" dialog to issue a command). So it fails with error:
```
File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 824, in issue_cluster_command
    [field.name for field in commands[command].schema.fields],
                             ~~~~~~~~^^^^^^^^^
KeyError: 1
```
Unfortunately unless ZHA is made aware of the possibility to send command to that cluster, it is not possible to use it. The difference with `zha_toolkit` is that it probably just fires away and hopes for the best.

BetaRavener, 2024-02-06:
I've successfully created a quirk that works with Philips Go to set the candlelight and fireplace effects, though I still have to find how to make it more user friendly (preset values, even getting it to the effect dropdown). For now I was testing it just by editing files in my docker deployment under `/usr/local/lib/python3.11/site-packages/zhaquirks/philips`.

In `__init.py__` goes:
```
class PhilipsEffectCluster(CustomCluster):
    """PhilipsEffectCluster."""

    cluster_id = 0xFC03
    ep_attribute = "philips_effect"
    name = "PhilipsEffect"

    server_commands = {
        0x00: foundation.ZCLCommandDef(
            "set_effect", 
            schema={
                "param1": t.uint8_t,
                "param2": t.uint8_t,
                "param3": t.uint8_t,
                "param4": t.uint8_t,
            },
            direction=foundation.Direction.Client_to_Server,
            is_manufacturer_specific=True,
        )
    }
```

and then created new file `7602031P7.py`:
```
"""Philips Go 7602031P7 device."""
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.philips import (
    PHILIPS,
    PhilipsEffectCluster,
)

class Philips7602031P7(CustomDevice):
    """Philips 7602031P7 device."""

    signature = {
        MODELS_INFO: [
            (PHILIPS, "7602031P7"),
        ],
        ENDPOINTS: {
            11: {
                # SizePrefixedSimpleDescriptor(endpoint=11, profile=260, device_type=268, device_version=1, input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096, 64513, 64515], output_clusters=[25])
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                    64513,
                    64515,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            # SizePrefixedSimpleDescriptor(endpoint=242, profile=41440, device_type=97, device_version=0, input_clusters=[], output_clusters=[33])
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            11: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                    64513,
                    PhilipsEffectCluster,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

which gets you something like this, where you can set the 4 numbers from args per parameter (33, 0, 1, 1) and send the command.
[image]

BetaRavener, 2024-02-15:
Think I found the missing link. Just going to write it all out here in case I forget, or somebody wants to do this before I have time :).

For a second, I'll use occupancy sensor as cluster example since it's much easier to search for across repositories than light bulb. In this repo, with zhaquirks, custom cluster [PhilipsOccupancySensing](https://github.com/zigpy/zha-device-handlers/blob/a4c15ef5510564d2fe02794be6cfaebc04f7f44b/zhaquirks/philips/__init__.py#L78) is defined. Important is what's stored in `ep_attribute = "philips_occupancy"`. This is then matched on the HA side with [CLUSTER_HANDLER](https://github.com/home-assistant/core/blob/2024.2.1/homeassistant/components/zha/core/const.py#L81) constant. Finally, this can be then used to implement the [new class ](https://github.com/home-assistant/core/blob/2024.2.1/homeassistant/components/zha/binary_sensor.py#L126) matching the cluster. 

Then, back to light as example, one can get the cluster handler [like this](https://github.com/home-assistant/core/blob/2024.2.1/homeassistant/components/zha/light.py#L652). The cluster handler is another class, specifically for the `CLUSTER_HANDLER_COLOR` it's [this one](https://github.com/home-assistant/core/blob/2024.2.1/homeassistant/components/zha/core/cluster_handlers/lighting.py#L180). I've linked the `color_loop_supported` on purpose, since it is exactly [the condition](https://github.com/home-assistant/core/blob/2024.2.1/homeassistant/components/zha/light.py#L717) I've linked in previous comment where `colorloop` is added, confirming we are looking at the correct thing. But now a big assumption - I think we don't need to write our own cluster handler, it should exist based on the quirk we defined earlier, otherwise it's a strange duplication.

The last step is figuring out what's the interface to issue the `set_effect` command we defined on our cluster and pass the arguments, but I think I'll have to do some live debugging to answer that.

BetaRavener, 2024-02-16:
The assumption regarding cluster being automatically created for us is correct. But the cluster is only passed in `cluster_handlers` when it is explicitly being matched against. Also thanks to the existing `HueLight` class, we need to specify the manufacturer, otherwise the `HueLight` match gets precedence (the logic is [here](https://github.com/home-assistant/core/blob/2024.2.2/homeassistant/components/zha/core/registries.py#L157)).

So far I have hacked this into my HA `zha/light.py`:
```
@STRICT_MATCH(
    cluster_handler_names=CLUSTER_HANDLER_ON_OFF,
    aux_cluster_handlers={CLUSTER_HANDLER_COLOR, CLUSTER_HANDLER_LEVEL, "philips_effect"},
    manufacturers={"Philips", "Signify Netherlands B.V."},
)
class EffectHueLight(Light):
    """Representation of a HUE light with effects."""
    def __init__(
        self, unique_id, zha_device: ZHADevice, cluster_handlers, **kwargs
    ) -> None:
        """Initialize the ZHA light."""
        super().__init__(unique_id, zha_device, cluster_handlers, **kwargs)
        h = self.cluster_handlers.get("philips_effect")
        self.error("Effect Handler: %s, %s", str(h), dir(h))
```

The cluster definition looks like this:
```
2024-02-16 18:17:22.824 ERROR (MainThread) [homeassistant.components.zha.entity] None: Effect Handler: <homeassistant.components.zha.core.cluster_handlers.ClusterHandler object at 0x7f6c01a3e0d0>, ['BIND', 'REPORT_CONFIG', 'ZCL_INIT_ATTRS', '__annotations__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattr__', '__getattribute__', '__getstate__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_cluster', '_configure_reporting_status', '_endpoint', '_generic_id', '_get_attribute_name', '_get_attributes', '_id', '_status', '_unique_id', 'async_configure', 'async_initialize', 'async_send_signal', 'async_update', 'attribute_updated', 'bind', 'cluster', 'cluster_command', 'configure_reporting', 'data_cache', 'debug', 'error', 'generic_id', 'get_attribute_value', 'get_attributes', 'id', 'info', 'log', 'matches', 'name', 'status', 'unique_id', 'warning', 'write_attributes_safe', 'zdo_command', 'zha_send_event']
```

I'm thinking `h.zha_send_event("set_effect", [33, 0, 1, 1])` [should work](https://github.com/home-assistant/core/blob/6f74ea9186fc70d33e628c39251a94bf1a5ca033/homeassistant/components/zha/core/cluster_handlers/__init__.py#L446). Not sure if it's correct though, as the function is decorated as callback.

BetaRavener, 2024-02-17:
Got a working prototype, I'll try to open up PRs both on zigpy and HA sides to integrate this in:
[image]

BetaRavener, 2024-02-18:
Just found out interesting information playing with the new quirk.
* When `param1` is set to `33`, `param3` controls on(1)/off(0) and `param4` is effect ID that seems to go until 8 - funny because Philips app exposes only 3 effect - candle(1), fireplace(2) and prism(3). 8 is another candle-like effect, the 4,5,6,7 are some specific color tranisitions, like warm tones, cold tones,...
* If one sets the `param1` to `34` instead of `33`, then `param3` controls brightness (cannot be 0), `param4` same as before effect ID.

miamilabs, 2024-04-19:
> I managed to activatee the candle effect using the ZCL command documented on [here](https://community.home-assistant.io/t/a-simple-single-automation-candle-effect-for-hue-white-ambience-and-other-bulbs/235023/15), on the following bulbs:
> 
>     * LCT015, with bulb firmware 1.108.7, but not with 1.101.2
> 
>     * LCT012, with bulb firmware 1.108.7, but not with 1.88.1
> 
>     * LTO001, with bulb firmware 1.104.2
> 
>     * 440400982842 (Playbar), with firmware 1.104.2
> 
>     * LLC020, with firmware 67.101.2
> 
>     * 1742830P7 (Lily), with 1.104.2, but not with 1.82.10
> 
>     * LCL003 (Outdoor Lightstrip), with 1.93.7, but not 1.66.6
> 
> 
> The following light does not seem to support the effect (tested with Hue app):
> 
>     * LST002 (light strip), with firmware: 1.101.2
> 
>     * LCT024 (Playbar), with firmware 1.108.5
> 
> 
> I could give it a go to develop a quirk, but it would be helpful if someone could point me to one (for other things) that also add an effect.
> 
> (the firmware versions are actually also documented [here](https://www.philips-hue.com/en-us/support/release-notes/lamps) ... )

Sorry for ping, facing same issue with fireplace after having 20 of the Playbars in my home. 
It is nice to have effect.
I have Playbar with firmware "[1.104.2]" which support Fireplace.. 

Is there any way to downgrade the firmware ?

nilsreiter, 2024-05-15:
Wouldn't that be an upgrade? From 1.104.2 to 1.108.5?

ENWOD, 2024-10-25:
Seen that the pull request didn't get anywhere and any help from that perspective is beyond me but would really like to see this implemented.

BetaRavener, 2025-01-19:
My pull request is finally updated and hopefully ready to be merged, but needs new review and has to be merged by maintainers.

BetaRavener, 2025-01-21:
@miamilabs the new PR now expands the support to couple more lightstrips and tubes on top of original Hue Go request. If you'd be able to test it with the playbars, it could be included in the PR. Or it can be added later separately, just an extra review for maintainer. Adding matching quirk following the pattern for testing should be easy but let me know if you'd need help.

BetaRavener, 2025-01-21:
Just an update on the GUI side of things, the code for matching now moved out of home assistant core to separate zigpy library, for example the light platform is now here https://github.com/zigpy/zha/blob/dev/zha/application/platforms/light/__init__.py . It threw me off reproducing previous attempts, and didn't want other people to trip over it aswell.

BetaRavener, 2025-01-22:
@jtbandes asked in the PR if there's way to [control speed](https://github.com/zigpy/zha-device-handlers/pull/3664#issuecomment-2606472309) as allowed by hue app.
I've been able to reverse engineer the protocol somewhat more. So far I'm assuming the effect command is always 4 bytes long but that could be also wrong. It appears the first byte is command identifier though behaving somewhat like a bitarray (flags). Second byte does something because setting it wrongly gives error, for example low part of the byte has to be even. But I can't see any effect when changing it. That leaves us with 2 more bytes, that change meaning according to the first command byte:

| Byte 1 | Byte 2 | Byte 3 | Byte 4 | Note |
| - | - | - | - | - |
| 0x01 | ? | On/Off | ? | |
| 0x02 | ? | Brightness | ? | |
| 0x03 | ? | On/Off | ? | Starts with minimum brightness |
| 0x04 | ? | XX | YY | Appears to set light to certain color temperature |
| 0x20 | ? | Effect | ? | Applies effect at 27% brightness |
| 0x21 | ? | On/Off | Effect | Applies effect at 27% brightness |
| 0x22 | ? | Brightness | Effect | |
| 0x80 | ? | Speed | ? | Preserves Brightness |
| 0x82 | ? | Brightness | Speed | Preserves Effect | 
| 0xA0 | ? | Effect | Speed | Applies effect at 100% brightness |

Remarks:
* For On/Off, only values 0 and 1 are available and it considers the whole byte
* For Brightness, range 1-254 is allowed and it considers the whole byte
* For Effect values 1-8 are allowed and it considers the whole byte.
* For Speed, values 1-15 are allowed and it uses high portion of the byte (i.e. 0xF0). Lower portion seems ignored.

I failed to find command that could set effect, brightness and speed at the same time, and I'm not sure it can be done since the second byte seems to always be assigned the same unkown meaning. Next best thing would be setting effect and speed **but preserving brightness**. The best option we have currently is to send two commands, for example 0x22 where we can pass existing/requested brightness, followed by 0x80 to set speed.

BetaRavener, 2025-01-22:
I've opened [another PR on the ZHA side](https://github.com/zigpy/zha/pull/355) that would ultimately allow selection of the effects from UI.

jtbandes, 2025-01-22:
@BetaRavener regarding the ZHA PR, I wonder if it is possible to conditionally extend the device's effects list based on the device model number or other device features? For example, the LTB003 only allows adjusting color temperature (not full RGB) and so it only supports the `candle` effect, not the other two.

Actually, I just noticed that some of my bulbs allow me to access these "Glisten" and "Sparkle" effects in the Hue app as well (probably because they have a newer firmware version). Surprisingly the Hue app includes presets for these effects that seem to be designed for RGB bulbs. But if I go to create a custom effect preset, it correctly limits my color choices to the color temperature setting.

[image]

[image]

[image]

BetaRavener, 2025-01-22:
I'll see if I can use some already present information to reasonably limit the choice (as you say no prism on purely white bulbs), I'd like to avoid further manual entries per device on top of adding the quirk itself. I guess good news is that worst case setting the effect on unsupported device would either do nothing, or at most error out when receiving reply from the zigbee device.

Can't test it well though as I don't have non-RGB Philips bulb :|

chrivers, 2025-01-29:
Hey guys - I've spent quite some time *completely* reversing the Hue Zigbee format.

I've written a document that is both an explanation, and a specification of sorts:

https://github.com/chrivers/bifrost/blob/master/doc/hue-zigbee-format.md

I thought it might be helpful for you :)

BetaRavener, 2025-01-30:
That's really cool, especially seeing that the bitarray theory is correct :) And also shows that assuming always 4 bytes was incorrect. It probably won't help much with the ZHA PR, as the UI is too simple., but definitely helpful for people who want to make up custom effect and trigger it directly.

Orange-GT3, 2025-06-20:
Just chiming in with some Philips Hue Downlights, Discover Floodlight and a Solo Lightstrip that I have. I don't have a Hue Bridge so the firmware versions are what is reported by ZHA and are the most recent available via the Z2M repo of device firmware:

- Philips Hue Garnea, LTD021 (White ambiance). Firmware version: 0x01002502;
- Philips Hue Akari, LCD003 (White and colour ambiance). Firmware version: 0x01002502;
- Philips Hue Discover, 1743530P7 (White and colour ambiance). Firmware version: 0x01002402; and
- Philips Hue Solo Lightstrip, LCL008. Firmware version: 0x01002402

I have multiple of both types of the downlights and they have only been bought in the last few months so are BT/Zigbee devices that are capable of the effects.

The LTD021 has the relevant Cluster of 0xfc03. The LCD003 does not, however, it does have clusters 0xfc01 and 02, both of which are Manufacturer Specific. I have tried sending the cluster command to these two clusters but nothing happens so is there some way of exposing cluster 0xfc03 in ZHA?

The Discover floodlight only has cluster 0xfc01, so the same question applies here about exposing cluster 0xfc03.

The Solo Lightstrip exposes clusters 0xfc01, 03 and 04 and so I can send commands to it to start the effects.

Beyond Candle (0x01) and Fireplace (0x02), there are also Prism (0x03), Sunrise (0x09), Sparkle (0x0a), Opal (0x0b), Glisten (0x0c), Sunset (0x0d), Underwater (0x0e), Cosmos (0x0f), Sunbeam (0x10) and Enchant (0x11). I obtained this from [Hue Gradient command generator](https://kjagiello.github.io/hue-gradient-command-wizard/), although the zha.issue_zigbee_cluster_command does not work so I use ZHA Toolkit as advised in 

Prism requires a colour light and I have read this is the same as Colour Loop. Some of other effects work on the LTD021 and LCL008 and some do not. This is possible a firmware issue given that I only have access to the firmware files hosted by the Z2M repo.

What would you need from me to help get the LCD003 Downlight and the Discover Floodlight added?

wbyoung, 2025-06-20:
> I have multiple of both types of the downlights and they have only been bought in the last few months so are BT/Zigbee devices that are capable of the effects.

If you updated them after pairing, you may need to unpaid and repair them for ZHA to have the clusters it associates with the current version.

Orange-GT3, 2025-06-21:
Confirming that, after removing and repairing my downlights, the 0xfc03 cluster is now exposed in ZHA and I can issue the ZHA Toolkit command to set an effect.

@BetaRavener - do you have an update on where your UI enhancement is at?

jtbandes, 2025-11-03:
Hi folks - based on @chrivers's reverse engineering work for bifrost, I created a Python package called [`hue-zigbee-encoding`](https://github.com/jtbandes/hue-zigbee-encoding) that can encode and decode these binary messages. Hoping this can be helpful to the community to further improve integration between HA and Hue lights. (Please let me know if you run into any issues -- I have only tested on some single bulbs, no light strips.)

Here's a demo: https://gist.github.com/jtbandes/39da3dfda0b16521319c63b277ee082a

```py
from hue_zigbee_encoding import HueLightUpdateMessage
HueLightUpdateMessage(is_on=True, brightness=127).to_bytes() # returns b"\x03\x00\x01\x7f"
```

Orange-GT3, 2025-11-05:
> Hi folks - based on [@chrivers](https://github.com/chrivers)'s reverse engineering work for bifrost, I created a Python package called [`hue-zigbee-encoding`](https://github.com/jtbandes/hue-zigbee-encoding) that can encode and decode these binary messages. Hoping this can be helpful to the community to further improve integration between HA and Hue lights. (Please let me know if you run into any issues -- I have only tested on some single bulbs, no light strips.)
> 
> Here's a demo: https://gist.github.com/jtbandes/39da3dfda0b16521319c63b277ee082a
> 
> from hue_zigbee_encoding import HueLightUpdateMessage
> HueLightUpdateMessage(is_on=True, brightness=127).to_bytes() # returns b"\x03\x00\x01\x7f"
> HueLightUpdateMessage.from_bytes(b"\x03\x00\x01\x7f") # returns a HueLightUpdateMessage object

@TheJulianJES