zigpy/zha-device-handlers #2518: [Device Support Request] Bosch Light/shutter control II (BMCT-SLZ)

sopelt, 2023-08-09:
### Problem description

[This device](https://www.bosch-smarthome.com/uk/en/products/devices/shutter-control/) looks like an interesting choice for roller/shutter control. After pairing it via code (as it needs this route to succeed) it is detected as `RBSH-MMS-ZB-EU`.
The issue is that the device has to modes: lights (2 separate relays for switching lights) or shutter (the firmware prevents both relays being switched on at the same time and allows for shutter-specific settings like timings).
Without setting this mode both options (a cover and two lights) are detected but seem non-functional.

### Solution description

Looking at [this code](https://github.com/Koenkk/zigbee-herdsman-converters/commit/54ce782cdc89c6cdc515285766d28cc2bc35f956) I have a basic idea of what value needs to be written to the manufacturer specific cluster to set the mode I am interested in but I have no experience writing quirks. If someone could point me to a quirk with a similar use case (most stuff looks quite okay out of the box but a man. spec. cluster needs to be mapped to two specific settable values) I would love to see what I manage to get together.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4617, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0102",
        "0x0702",
        "0x0b04",
        "0x0b05",
        "0xfca0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xfca0"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xfca0"
      ],
      "output_clusters": []
    }
  },
[... 574 lines cut ...]
                                "attribute_name": "on_off",
                                "value": 0
                            },
                            "0x4003": {
                                "attribute_name": "start_up_on_off",
                                "value": 0
                            }
                        },
                        "unsupported_attributes": {}
                    },
                    "0xfca0": {
                        "endpoint_attribute": "manufacturer_specific",
                        "attributes": {},
                        "unsupported_attributes": {}
                    }
                },
                "out_clusters": {}
            }
        }
    }
}

```

</details>

sopelt, 2023-08-10:
Using [zha-toolkit](https://github.com/mdeweerd/zha-toolkit) I was able to initialize the device to "shutter mode" by sending the relevant value (`0x01`) to the custom cluster (`0xfca0`):

```
service: zha_toolkit.attr_write
data:
  ieee: ...
  manf: 0x1209
  endpoint: 1
  cluster: 0xfca0
  attribute: 0x0000
  attr_type: 0x30
  attr_val: 0x01
```

This enables basic shutter operation (up, down, stop) but I have yet to look at calibrating the times etc.

sopelt, 2023-08-25:
Sticking with poking it with zha-toolkit one can set the `calibration_closing_time` (attribute `0x0002` ... similarly `0x0003` for opening) e.g. to 10s (encoded as 10s * 10 = 100 = `0x64` in hex)

```
service: zha_toolkit.attr_write
data:
  ieee: ...
  manf: 0x1209
  endpoint: 1
  cluster: 0xfca0
  attribute: 0x0002
  attr_type: 0x23
  attr_val: 0x64
```

So this should allow basic operation for many scenarios. I have yet to try to set the different shutter types or tilt angles. As far as I understand settings like the device type, calibration times should be handled directly in homeassistant-core similar to startup behavior. I could not find any traces for configuration for shutters and am thinking of exposing them through a quirk for now to at least be able to set them through HAs UI/services/blueprints for now.
Any pointers towards existing work in that direction would be appreciated.

sopelt, 2023-08-28:
I put together a [prototypical quirk](https://gist.github.com/sopelt/26c6fcddc91953d4fa64655c487b24e5) that allows setting the device type, switch type and shutter calibration times. This should enable basic operation as a shutter without zha-toolkit.

Feedback welcome ;) 

[image]

peterreimer, 2023-10-25:
Hi, thank you for this quirk. I have this switch installed and finally recognized  by ZHA, but it does not respond to any action in home assistant. I managed to install your quirk but i do not understand  how to set the type to light switch. I have two lights connected. Any hints for a home assistant newbie?

sopelt, 2023-10-25:
Hi @peterreimer 

I am assuming the basic pairing via the QR code worked and the device shows up but the roller/light controls do nothing?
This would be because of the "split personality" of being a light/shutter device and you first need to set the mode of the device itself. When navigating to the device in HA it should look something like

[image]

The dots and `manage zigbee device`

[image]

should bring you to the clusters view:

[image]

after choosing `BoschLightShutterControlII` as the cluster and `device_type` as the attribute you can hit `read` to see the initial type which will be something like `undefined` which is why the device refuses to act at all.

Changing the type to `DeviceType.Light` and writing the attribute should give you independent operation of the two relays to control your lights. 

[image]

peterreimer, 2023-10-25:
Hi @sopelt, that worked. Thank you for the detailed explanation. To get the pairing right took me also a while. The qr code on the switch was a long string containing both the IEEE Address and the installation code. Only when i found the little sticker in the box i figured out which value has to go into which field of the zha.permit tool. Now that i works i might get a second unit and use it as shutter control. Thanks again,
Peter

sopelt, 2023-10-25:
You're welcome. Yeah, the "join via code experience" is not great yet but I would assume that will be handled soon through QR scan + vendor code format support in the home assistant app as the devices using it seem to increase in numbers lately... 
How "split personality devices" like this one can be better supported out of the box and in a generic way - no idea ;)

Hedda, 2023-10-26:
FYI, check out the discussion here about Bosch Twinguard (8750001213 smoke detector) which might be related -> https://github.com/zigpy/zha-device-handlers/issues/2561

Also not sure if related this(?) but FYI; noticed @promasu has submitted PR to "**_Add support for Bosch QR-codes for zha.permit_**"

https://github.com/home-assistant/core/pull/102427

https://github.com/home-assistant/home-assistant.io/pull/29459

https://github.com/home-assistant/core/pull/102427/commits/3a753de25a4bd420ad4aa04a86733f3f92d66f59

```
    # Bosch
    r"""
        ^RB01SG
        [0-9a-fA-F]{34}
        ([0-9a-fA-F]{16}) # IEEE address
        DLK
        ([0-9a-fA-F]{36}) # install code
        $
    """,
```

promasu, 2023-10-26:
My change was only for converting the qr-codes from the battery lid etc from the bosch devices to the ieee address and install code used by the zha.allow service. This has nothing to do with device recognition or the handling of the install codes in general.
Just what happens with the user input on the zha.permit page.
I even think my change is not live with the current version (10.5) of Home Assistant, at least not according to the changelog

Hedda, 2023-10-26:
> My change was only for converting the qr-codes from the battery lid etc from the bosch devices to the ieee address and install code used by the zha.allow service. This has nothing to do with device [...]

@promasu Ah, but your QR-code PR/patch probably is related to this other discussion about Bosch Twinguard here? -> https://github.com/zigpy/zha-device-handlers/issues/2561

promasu, 2023-10-26:
> > My change was only for converting the qr-codes from the battery lid etc from the bosch devices to the ieee address and install code used by the zha.allow service. This has nothing to do with device recognition or the handling of the install codes in general.
> > Just what happens with the user input on the zha.permit page.
> > I even think my change is not live with the current version (10.5) of Home Assistant, at least not according to the changelog
> 
> @promasu Ah, but your QR-code PR/patch probably is related to this other discussion about Bosch Twinguard here, or? -> https://github.com/zigpy/zha-device-handlers/issues/2561

Not really I think. It's more to make life easy with putting in the qr instead of typing down two long strings

peterreimer, 2023-10-26:
Apparently i also have to set the `switch_type`. The default seems to be `undefinded`, but than i could only control the lights with home assistant, but not with the physical wall switch. Setting the `switch_type` to  `SwitchType.Rocker_Switch` solved this and now everything works as expected.

ring0r, 2024-02-22:
> Apparently i also have to set the `switch_type`. The default seems to be `undefinded`, but than i could only control the lights with home assistant, but not with the physical wall switch. Setting the `switch_type` to `SwitchType.Rocker_Switch` solved this and now everything works as 

Hi,

 thanks for your message. I have a similar problem with the Bosch Controller II and I want to use it for my roller blinds via Zigbee2mqtt.

 I also have the problem that some attributes are not recognized during pairing, but the biggest problem is that I can no longer use the wall switches.

 Where can I change the settings as you did?

 Thanks in advance!

peterreimer, 2024-02-23:
i did this in the "Manage Zigbee Device" Dialog. The way to get there is described by @sopelt earlier in this threat
[image]
But you say you use zigbee2mqtt, so this is probably not the right place for you to ask since this solution is for ZHA.

Hedda, 2024-02-23:
> I want to use it for my roller blinds via Zigbee2mqtt.

ZHA Device Handlers (also known as ZHA Quirks) have nothing to do with Zigbee2MQTT, instead these are only for the native Zigbee Home Automation that is built into Home Assistant as a core component, see -> https://www.home-assistant.io/integrations/zha/

Darkmelman, 2024-05-22:
@sopelt Ho, thank you for your quirk.
The topic was absolutely new for me and i can't get it to work. I paired the Bosch Licht-/Rolladensteuerung ii with my HA ZHA but it's not using the quirk and i can't figure out why.

I edited the configuration.yaml, checked the config, restarted HA und repaired but it's not using the quirk. Maybe you have an idea.

Thanks in advance!

closed 2024-11-25 by github-actions[bot]

stylefish, 2024-12-23:
@sopelt thanks for the quirk, using it and it works flawless. :)
only thing i recognize is, that when the shutter is opened and it is triggered again to open (while fully opened) it shows as "opening" all the time. to get it to display only "open" again i have to close some percentage and open again. is this a zha / ha thing or can this be integrated somehow into your quirk? 
greetings :)

TomF79, 2025-04-23:
> I put together a [prototypical quirk](https://gist.github.com/sopelt/26c6fcddc91953d4fa64655c487b24e5) that allows setting the device type, switch type and shutter calibration times. This should enable basic operation as a shutter without zha-toolkit.
> 
> Feedback welcome ;)
> 
> [image]

> Apparently i also have to set the `switch_type`. The default seems to be `undefinded`, but than i could only control the lights with home assistant, but not with the physical wall switch. Setting the `switch_type` to `SwitchType.Rocker_Switch` solved this and now everything works as expected.

Thanks for your Quirk and Infos.

Can anyone tell me please which switch-types are possible?

stylefish, 2025-04-24:
you can use `DeviceType.Shutter` or `DeviceType.Light`

stylefish, 2025-04-24:
@sopelt is this quirk still working for you in HA 2025.4.3? i'm having issues when trying to open or close a shutter, HA just states that the action is not available for the entity...

i've also posted to HA community but got no response: https://community.home-assistant.io/t/bosch-light-shutter-control-ii-not-working-after-update-to-2025-4-3/880222

also i think this device request should be reopened :)