zigpy/zha-device-handlers #2519: Add support for Namron stoveguard

bthovda, 2023-08-09:
### Problem description

Please add ZHA support for Namron Stoveguard. https://www.elektroimportoren.no/docs/lib/1402793-Brukerveiledning-5.pdf
Currently it is added ok but not all sensors are working.

1. Switch (Possible to turn off, but not ON) Possibly supposed to be like this. (Only local reset)
    Should it be a toggle switch then? button?
3. Power consumption (Watt) OK, working without problems.
4. Power factor, displaying (Unknown)
5. Temperature showing −327,7 °C (Constant)

### Solution description

Fix Power factor and temperature sensor

### Screenshots/Video

https://www.elektroi
[1402793-Brukerveiledning-5.pdf](https://github.com/zigpy/zha-device-handlers/files/12303280/1402793-Brukerveiledning-5.pdf)
mportoren.no/docs/lib/1402793-Brukerveiledning-5.pdf

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.MainsPowered|RxOnWhenIdle|AllocateAddress: 140>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=10752, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability: 80>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0006",
        "0x0402",
        "0x0702",
        "0x0b04"
      ],
      "output_clusters": [
        "0x0003"
      ]
    },
    "10": {
      "profile_id": "0x0104",
      "device_type": "0xfff0",
      "input_clusters": [
        "0x0000"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "NAMRON AS",
  "model": "1402790",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
[... 650 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Polsemaker, 2023-08-23:
Hi @bthovda , 
I am struggeling to get this into HA.
I have created a converter

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['1402790', '1402791', '1402792', '1402793', '1402794', '1402795', '1402796', '1402797', '1402798'],
    model: '1402790',
    vendor: 'Namron',
    description: 'Namron Stove Guard',
    fromZigbee: [fz.on_off, fz.battery, fz.temperature, fz.metering, fz.electrical_measurement],
    toZigbee: [tz.on_off],
    exposes: [e.temperature(), e.battery(), e.switch(), e.power(), e.energy()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1) || device.getEndpoint(10);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'genOnOff', 'haElectricalMeasurement', 'genPowerCfg', 'msTemperatureMeasurement']);
            await endpoint.read('haElectricalMeasurement', ['acVoltageMultiplier', 'acVoltageDivisor']);
            await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
            await endpoint.read('haElectricalMeasurement', ['acCurrentMultiplier', 'acCurrentDivisor']);
            await reporting.onOff(endpoint);
            await reporting.temperature(endpoint);
            await reporting.activePower(endpoint);
            await reporting.batteryPercentageRemaining(endpoint);
        },
};

module.exports = definition;`

And it seems to be somewhat working.
However, after a short while the device drops of the network.
Did you do anything spesific to get the device not to stop reporting?

[image]

Torgrima, 2023-10-09:
I got this zigbee api description for Namron Stoveguard: 
[FireFence_Zigbee_description_rev1.2 – without OTA.pdf](https://github.com/zigpy/zha-device-handlers/files/12848198/FireFence_Zigbee_description_rev1.2.without.OTA.pdf)

sharknoms, 2024-01-29:
I would also appreciate this one. Able to add but sensors and swich not working. Also disconnects after a few hours.

cbjornerud, 2024-02-08:
I also use the Namron stoveguard in Home Assistant, with ZHA and a Sky connect. The last few days it has reacted strange when Home Assistant is restarted. The Stoveguard shut off and on the power many times until I disconnected it from power (turned off the fuse). I reported this to the supplyer elektroimportoren.no today. They replied that the device do not support Home Assistant and that I should disconnect it from my Zigbee network if the network caused it to behave like that. I tried to use Zigbee2MQTT instead, as I have two Zigbee netwoks to support all devices. That did not work at all. It is added to the network, but as unsupported and without updating any values. As a result of this it will be offline until it is better supported.

ChristianPB, 2024-06-24:
I have used it with zigbee2mqtt for a month. I worked perfekt from 25.may to 22.june. Then suddenly the stovegard was lost... This has happened before. Also when the stovegard was only 3m from the antenna. (Sonoff)

AtzeDeVries, 2024-07-31:
@cbjornerud any updates here?

cbjornerud, 2024-07-31:
> @cbjornerud any updates here?

No, I have not done anything more with this. The stoveguard is stable as a stoveguard, that is the most important function. A smart stoveguard would be nice, but I have not spent any more time with this. Maybye later, in the winter. 🙂

Artur-Fedjukevits, 2024-09-22:
> > @cbjornerud any updates here?
> 
> No, I have not done anything more with this. The stoveguard is stable as a stoveguard, that is the most important function. A smart stoveguard would be nice, but I have not spent any more time with this. Maybye later, in the winter. 🙂

I will wait that.

cbjornerud, 2024-09-26:
Some news, but not good. I thought it was working fine in Home Assistant with ZHA, but after a week I came home to a stove without power. Pushing the button on the sensor of the Stove Guard did not turn the power back on. I then turned off the circuit breaker and back on again after about 10 second. The power came back to the stove, but then the Stove guard startet turing the power off and on again with only a few seconds on at a time. It did not stop until I turned off the circuit breaker and again. I will have to try to reset it and connect with Z2M again, but that will not be today.

olemr, 2025-04-02:
New FW out now, with more rapid power consumption reports and zigbee ON/OFF:
https://www.hjemmeautomasjon.no/forums/topic/12453-namron-komfyrvakt-planlagt-oppgradering-og-st%C3%B8tte-for-home-assistant/#findComment-118096