zigpy/zha-device-handlers #2520: Improve Aqara H1 switch quirks

TheJulianJES, 2023-08-10:
This is for tracking the eventual improvement/cleanup of the Aqara (H1) switch quirks:

1. https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/opple_switch.py is missing `MODELS_INFO`
2. Clean up where H1 quirks are stored (single/double rocker):
    - H1 double switch is in: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/opple_switch.py
    - H1 single switch is in: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/aqara/switch_h1.py
3. Figure out proper naming / should T1 relay quirk be related: https://github.com/zigpy/zha-device-handlers/pull/2513 ?

Related PRs:
- https://github.com/zigpy/zha-device-handlers/pull/1391 (missing `MODELS_INFO` + file name could be improved)
- https://github.com/zigpy/zha-device-handlers/pull/2385
- https://github.com/zigpy/zha-device-handlers/pull/2411

TheJulianJES, 2023-11-29:
Add quirk with energy/power/custom attributes support for: https://github.com/zigpy/zha-device-handlers/issues/2194#issue-1583172238

We already have the two rocker one (`lumi.switch.b2naus01`).

TheJulianJES, 2023-11-29:
Also see:
- https://github.com/zigpy/zha-device-handlers/issues/2270
- https://github.com/zigpy/zha-device-handlers/issues/1693
- https://github.com/zigpy/zha-device-handlers/issues/1134

TheJulianJES, 2023-11-29:
This PR will address some of the issues above:
- https://github.com/zigpy/zha-device-handlers/pull/2784

zanglang, 2024-04-09:
My b2naus01 doesn't seem to be reporting any power usage on Home Assistant 2024.04, and shows 0 for all energy metrics. Should there be yet another signatures added to opple_switch.py?

This is in contrast to the PR in #3101 that I was testing for b1naus01, which seems to at least reports the power usage in watts.

Signature:
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0012",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    },
    "31": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    },
    "41": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "42": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "51": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "61": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.b2naus01",
  "class": "zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitch4"
}
```

MichaelVoelkel, 2024-05-19:
For my Aqara power sensor it also only shows 0 for wattage/voltage etc. It at least shows that a consumer is connected.
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug.maeu01",
  "class": "zhaquirks.xiaomi.aqara.plug_eu.PlugMAEU01"
}
```

Version is 2024.04.03, no updates available.

closed 2024-11-22 by github-actions[bot]