zigpy/zha-device-handlers #2521: [Device Support Request]

LeFlairGoD, 2023-08-10:
### Problem description

The device can be paired, but no entities are added. See screenshot
[image]

https://de.aliexpress.com/item/1005004352027105.html?spm=a2g0o.order_list.order_list_main.29.5b125c5favtCSs&gatewayAdapt=glo2deu#nav-review

### Solution description

Full thermostat support would be good. With temperature control, the switching options and definition of the programs.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_5toc8efa",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.8.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
[... 462 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

javicalle, 2023-08-14:
According to Z2M it seems to be a `Moes BHT series Thermostat` device:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/4d535effcbd6fbae0317ded493e116faf30e00ed/src/devices/moes.ts#L100-L109

Try to enable the local quirk and add your device to the `MoesBHT` signature:
* https://github.com/zigpy/zha-device-handlers/blob/a5d3ef7c370d97a2203adad20903b567dc368350/zhaquirks/tuya/ts0601_electric_heating.py#L136-L147

javicalle, 2023-08-14:
Do you have enabled the custom quirk in the zha config?
Put your local `ts0601_electric_heating.py` file in the configured folder (not in a subfolder).

TheJulianJES, 2023-08-15:
Also, do not put your custom quirks folder in `custom_components`.

You need to create a `custom_zha_quirks` folder in `/config` and then add this to your `configuration.yaml`:
```yaml
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

LeFlairGoD, 2023-08-15:
Hi,
I am relatively new to HomeAssistant, is there a rough guide somewhere on what exactly I need to do?

javicalle, 2023-08-15:
https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

derkorte, 2023-08-15:
[image]
[image]

Like this? Still not working

derkorte, 2023-08-15:
Do i have to add my device to the device list? 
it's _TZE204_5toc8efa 

like this: 
            ("_TZE200_aoclfnxz", "TS0601"),
            ("_TZE200_2ekuz3dz", "TS0601"),
            ("_TZE200_ye5jkfsb", "TS0601"),
            ("_TZE200_u9bfwha0", "TS0601"),
            ("_TZE204_5toc8efa", "TS0601"),

derkorte, 2023-08-15:
this is my entry from logs:
 
Logger: zhaquirks
Source: components/zha/__init__.py:126
First occurred: 18:48:19 (1 occurrences)
Last logged: 18:48:19

Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers

->so my config and folders seems to be correct. 
I also bought a smoke detector from Aliexpress, wich hasnt any entitys but by adding the device id to the quirk it works fine.
So it seems like theres any issue with the converter. I've changed from z2m 2 month ago, there it was possible to detect as Moes BHT-002-GCLZB with Koenkk's converter with all entities. 
But with my change to ZHA all is more stable an seems to be faster.

Theres still no change when i change the lines you wrote above. But here is the full quirk, mabe ive build in some error:

"""Map from manufacturer to standard clusters for electric heating thermostats."""
import logging

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
)

# info from https://github.com/zigpy/zha-device-handlers/pull/538#issuecomment-723334124
# https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L239
# and https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/common.js#L113
MOESBHT_TARGET_TEMP_ATTR = 0x0210  # [0,0,0,21] target room temp (degree)
MOESBHT_TEMPERATURE_ATTR = 0x0218  # [0,0,0,200] current room temp (decidegree)
MOESBHT_SCHEDULE_MODE_ATTR = 0x0403  # [1] false [0] true   /!\ inverted
MOESBHT_MANUAL_MODE_ATTR = 0x0402  # [1] false [0] true /!\ inverted
MOESBHT_ENABLED_ATTR = 0x0101  # [0] off [1] on
MOESBHT_RUNNING_MODE_ATTR = 0x0424  # [1] idle [0] heating /!\ inverted
MOESBHT_CHILD_LOCK_ATTR = 0x0128  # [0] unlocked [1] child-locked

_LOGGER = logging.getLogger(__name__)

class MoesBHTManufCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of some electric heating thermostats."""

    attributes = {
        MOESBHT_TARGET_TEMP_ATTR: ("target_temperature", t.uint32_t, True),
        MOESBHT_TEMPERATURE_ATTR: ("temperature", t.uint32_t, True),
        MOESBHT_SCHEDULE_MODE_ATTR: ("schedule_mode", t.uint8_t, True),
        MOESBHT_MANUAL_MODE_ATTR: ("manual_mode", t.uint8_t, True),
        MOESBHT_ENABLED_ATTR: ("enabled", t.uint8_t, True),
        MOESBHT_RUNNING_MODE_ATTR: ("running_mode", t.uint8_t, True),
        MOESBHT_CHILD_LOCK_ATTR: ("child_lock", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == MOESBHT_TARGET_TEMP_ATTR:
            self.endpoint.device.thermostat_bus.listener_event(
                "temperature_change",
                "occupied_heating_setpoint",
[... 109 lines cut ...]
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesBHTManufCluster,
                    MoesBHTThermostat,
                    MoesBHTUserInterface,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            "242": {
                "profile_id": "0xa1e0",
                "device_type": "0x0061",
                "input_clusters": [],
                "output_clusters": ["0x0021"]
            }
        }
    }

derkorte, 2023-08-19:
Anyone got this work?

bsfaxi, 2023-08-22:
Trying the same here: [https://github.com/zigpy/zha-device-handlers/issues/2433]

bsfaxi, 2023-08-30:
What about your HA log file? Any debug info regarding your ".py" file?

LeFlairGoD, 2023-08-30:
> Was ist mit Ihrer HA-Protokolldatei? Irgendwelche Debug-Informationen zu Ihrer „.py“-Datei?

The only thing in the log after reconnecting are these two entries

```
Logger: frontend.js.latest.202308021
Source: components/system_log/__init__.py:270
First occurred: 21:57:20 (2 occurrences)
Last logged: 21:57:20

http://homeassistant.local:8123/frontend_latest/62575-6mNV-72kqCE.js:4:8121 Uncaught TypeError: Cannot read properties of undefined (reading 'get')
```

```
Logger: zigpy.zcl
Source: runner.py:179
First occurred: 21:57:26 (51 occurrences)
Last logged: 21:57:42

[0xEAF3:1:0xef00] Unknown cluster command 2 b'\x00\xfbl\x00\x00\x11\x03\x06\x00\x00\xc8\x0b\x1e\x00\xc8\r\x1e\x00\xc8\x11\x1e\x00\xc8'
[0xEAF3:1:0xef00] Unknown cluster command 2 b'\x00\xfck\x00\x00\x11\x04\x06\x00\x00\xc8\x0b\x1e\x00\xc8\r\x1e\x00\xc8\x11\x1e\x00\xc8'
[0xEAF3:1:0xef00] Unknown cluster command 2 b'\x00\xfdj\x00\x00\x11\x05\x06\x00\x00\xc8\x0b\x1e\x00\xc8\r\x1e\x00\xc8\x11\x1e\x00\xc8'
[0xEAF3:1:0xef00] Unknown cluster command 2 b'\x00\xfei\x00\x00\x11\x06\x06\x00\x00\xc8\x0b\x1e\x00\xc8\r\x1e\x00\xc8\x11\x1e\x00\xc8'
[0xEAF3:1:0xef00] Unknown cluster command 2 b'\x00\xffe\x00\x00\x11\x07\x06\x00\x00\xc8\x0b\x1e\x00\xc8\r\x1e\x00\xc8\x11\x1e\x00\xc8'
```

LeFlairGoD, 2023-08-30:
> The only thing in the log after reconnecting are these two entries
>
> ```
> Logger: frontend.js.latest.202308021
> Source: components/system_log/__init__.py:270
> First occurred: 21:57:20 (2 occurrences) [...]

and this in HA.log:
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 64, in wrapper
    return await RETRYABLE_REQUEST_DECORATOR(func)(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/util.py", line 132, in retry
    return await func()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/quirks/__init__.py", line 199, in command
    return await self.request(
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 375, in request
    return await self._endpoint.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 253, in request
    return await self.device.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 309, in request
    async with asyncio_timeout(timeout):
  File "/usr/local/lib/python3.11/asyncio/timeouts.py", line 111, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 226, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1974, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2011, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 591, in async_handle_light_off_service
    await light.async_turn_off(**filter_turn_off_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 471, in async_turn_off
    result = await self._on_off_cluster_handler.off()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 66, in wrapper
    raise HomeAssistantError(
```

bsfaxi, 2023-08-31:
Please activate the debug logs to investigate.
Your quirk is not yet taken into account...
```
logger:
  default: info
  logs:
    zigpy: debug
    zigpy.zcl: debug
    custom_zha_quirks: debug
    zhaquirks: debug
    homeassistant.components.zha: debug
```

LeFlairGoD, 2023-08-31:
> Bitte aktivieren Sie die Debug-Protokolle zur Untersuchung. Deine Eigenart ist noch nicht berücksichtigt...
> 
> ```
> logger:
>   default: info
>   logs:
>     zigpy: debug
>     zigpy.zcl: debug
>     custom_zha_quirks: debug
>     zhaquirks: debug
>     homeassistant.components.zha: debug
> ```

I did that, but there is still nothing in the HA.log files.
[image]
[image]

derkorte, 2023-09-05:
So I think there's an issue with the outputs of the devices. I've ordered an other thermostat from AliExpress for about 30€ witch is named moes and this works out of the box. The device from bsfaxi just looks like a moes thermostat. My _TZE204_5toc8efa is manufactured by bseed, so (I believe) there where other clusters. I don't know how to read out this device. Can someone explain how to, or give a link to get this done?

closed 2024-03-03 by LeFlairGoD

achikhv, 2024-03-10:
Got it working with slightly modified [quirk](https://github.com/zigpy/zha-device-handlers/issues/2433#issuecomment-1690198060) by @bsfaxi

I'll file PR in couple of days.

antonGritsenko, 2024-04-23:
@achikhv did you already pushed the PR?

achikhv, 2024-04-25:
Sorry, I didn't manage to get internet working on my RPI, so it's not easy to download quirk right now. I remember about this issue and will do my best to file PR as soon as possible.

Anyway, the quirk is taken [here](https://github.com/zigpy/zha-device-handlers/issues/2433#issuecomment-1690198060) and the only difference is in multipliers of target temperature. As far as I remember target temperature is measured in 1/10th, not 1/100th degrees. So I had to change this multipliers. The rest of the quirk is intact.

antonGritsenko, 2024-04-25:
@achikhv ok, thanks, yes, I'm using it now, but it a bit buggy: button doesn't work (and it shouldn't I think).
Also, why not to move current temperature as readable value for HA?