zigpy/zha-device-handlers #2522: [Device Support Request]

jesushairdo, 2023-08-11:
### Problem description

When paired to Home Assistant the only triggers available are related to battery level and the identify feature. There are no triggers for any button press activities

The logbook in HA does show events for button presses for all buttons

### Solution description

All button presses made available as triggers and / or entities with values

According to [user guide](https://www.innr.com/wp-content/uploads/2021/08/INNR_IM_RC_250_LC_WT10.pdf): there should be a number of events triggered:

- Power Button - maintains an On/Off state
- Dim Up
- Dim Down
- Left - Short-press & long press
- Right - short-press & long press

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4454, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0105",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05",
        "0x1000",
        "0xfd01"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "innr",
  "model": "RC 250",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
[... 421 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheJulianJES, 2023-08-12:
In the "developer tools" in HA, go to the "events" tab and enter `zha_event` into the bottom text field and start listening for events.
Then do all the actions the remote can trigger and paste the events here (and to which action it corresponds).

jesushairdo, 2023-08-14:
Power Button - appears to toggle between On / Off state., Here are 3 button presses to show the toggle

### Power button press #1
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0006
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 6
  command: "off"
  args: []
  params: {}
origin: LOCAL
time_fired: "2023-08-14T16:23:15.157929+00:00"
context:
  id: 01H7TDQNENX9W1FPRVKYZHZATE
  parent_id: null
  user_id: null

### Power Button press #2
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0006
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 6
  command: "off"
  args: []
  params: {}
origin: LOCAL
time_fired: "2023-08-14T16:23:15.157929+00:00"
context:
  id: 01H7TDQNENX9W1FPRVKYZHZATE
  parent_id: null
  user_id: null

### Power Button press #3
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0006
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 6
  command: "off"
  args: []
  params: {}
origin: LOCAL
time_fired: "2023-08-14T16:23:15.157929+00:00"
context:
  id: 01H7TDQNENX9W1FPRVKYZHZATE
  parent_id: null
  user_id: null

jesushairdo, 2023-08-14:
### Brightness Up button
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0008
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 8
  command: step
  args:
    - 0
    - 51
    - 3
    - 0
    - 0
  params:
    step_mode: 0
    step_size: 51
    transition_time: 3
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:27:42.705087+00:00"
context:
  id: 01H7TDZTQH0Z2FCBKCRK4YY14A
  parent_id: null
  user_id: null

### Brightness Down button
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0008
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 8
  command: step
  args:
    - 1
    - 51
    - 3
    - 0
    - 0
  params:
    step_mode: 1
    step_size: 51
    transition_time: 3
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:28:46.872090+00:00"
context:
  id: 01H7TE1SCRA7AS1PVJ53TJYVAQ
  parent_id: null
  user_id: null

jesushairdo, 2023-08-14:
Left button triggers 2 events per button press - seemingly on an increasing scale

### Left Button - press 1, event 1
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0008
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 8
  command: move_to_level_with_on_off
  args:
    - 76
    - 3
  params:
    level: 76
    transition_time: 3
origin: LOCAL
time_fired: "2023-08-14T16:30:14.928254+00:00"
context:
  id: 01H7TE4FCGGRQWT6TM2BBHR18S
  parent_id: null
  user_id: null

### Left button - press 1, event 2
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0300
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 768
  command: move_to_color_temp
  args:
    - 454
    - 3
    - 0
    - 0
  params:
    color_temp_mireds: 454
    transition_time: 3
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:30:14.979672+00:00"
context:
  id: 01H7TE4FE30XXPD8071BBK744D
  parent_id: null
  user_id: null

### Left button - press 2, event 1
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0008
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 8
  command: move_to_level_with_on_off
  args:
    - 178
    - 3
  params:
    level: 178
    transition_time: 3
origin: LOCAL
time_fired: "2023-08-14T16:31:04.320645+00:00"
context:
  id: 01H7TE5ZM0QKVTNC1M76W8GZ8G
  parent_id: null
  user_id: null

### Left button - press 2, event 2
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0300
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 768
  command: move_to_color_temp
  args:
    - 370
    - 3
    - 0
    - 0
  params:
    color_temp_mireds: 370
    transition_time: 3
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:31:04.354108+00:00"
context:
  id: 01H7TE5ZN2V6M0XFEZ3BW48ACK
  parent_id: null
  user_id: null

jesushairdo, 2023-08-14:
Right button triggers 2 events per button press - seemingly on an increasing scale

### Right button - press 1, event 1
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0008
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 8
  command: move_to_level_with_on_off
  args:
    - 76
    - 3
  params:
    level: 76
    transition_time: 3
origin: LOCAL
time_fired: "2023-08-14T16:32:43.697713+00:00"
context:
  id: 01H7TE90NHTPET6T711HS6HMTN
  parent_id: null
  user_id: null

### Right button - press 1, event 2
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0300
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 768
  command: move_to_color_temp
  args:
    - 454
    - 3
    - 0
    - 0
  params:
    color_temp_mireds: 454
    transition_time: 3
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:32:43.741719+00:00"
context:
  id: 01H7TE90PX3TV4E2H0GZKPM061
  parent_id: null
  user_id: null

### Right button - press 2, event 1
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0008
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 8
  command: move_to_level_with_on_off
  args:
    - 2
    - 3
  params:
    level: 2
    transition_time: 3
origin: LOCAL
time_fired: "2023-08-14T16:33:49.797123+00:00"
context:
  id: 01H7TEB175YS02539K12TQB4ZV
  parent_id: null
  user_id: null

### Right button - press 2, event 2
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0300
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 768
  command: move_to_color_temp
  args:
    - 588
    - 3
    - 0
    - 0
  params:
    color_temp_mireds: 588
    transition_time: 3
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:33:49.823201+00:00"
context:
  id: 01H7TEB17ZW23NT20WXM5SNTAA
  parent_id: null
  user_id: null

jesushairdo, 2023-08-14:
### Upbutton Long-hold
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0008
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 8
  command: move
  args:
    - 0
    - 85
    - 0
    - 0
  params:
    move_mode: 0
    rate: 85
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:34:29.407396+00:00"
context:
  id: 01H7TEC7WZH5306M5HHW9DEDHN
  parent_id: null
  user_id: null

### Upbutton long-hold release
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0008
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 8
  command: stop
  args:
    - 0
    - 0
  params:
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:34:31.518507+00:00"
context:
  id: 01H7TEC9YY8S3F047DBDP4C0NP
  parent_id: null
  user_id: null

jesushairdo, 2023-08-14:
### Down button long-hold
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0008
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 8
  command: move
  args:
    - 1
    - 85
    - 0
    - 0
  params:
    move_mode: 1
    rate: 85
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:36:44.632396+00:00"
context:
  id: 01H7TEGBYRGFM3JDEKSWFT60F5
  parent_id: null
  user_id: null

### Down button long-hold release
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0008
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 8
  command: stop
  args:
    - 0
    - 0
  params:
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:36:45.965479+00:00"
context:
  id: 01H7TEGD8DX2XSFH12C7KHW3SN
  parent_id: null
  user_id: null

jesushairdo, 2023-08-14:
### left button long-hold
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0300
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 768
  command: enhanced_move_hue
  args:
    - 3
    - 13108
  params:
    move_mode: 3
    rate: 13108
    options_mask: null
    options_override: null
origin: LOCAL
time_fired: "2023-08-14T16:39:01.839298+00:00"
context:
  id: 01H7TEMHYF9GR31CPWYGK4FF76
  parent_id: null
  user_id: null

### left button long-hold release
event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:99:86:44
  unique_id: 84:ba:20:ff:fe:99:86:44:1:0x0300
  device_id: 99b867de5445caaf3493aeca2d3260cc
  endpoint_id: 1
  cluster_id: 768
  command: stop_move_step
  args:
    - 0
    - 0
  params:
    options_mask: 0
    options_override: 0
origin: LOCAL
time_fired: "2023-08-14T16:39:04.860964+00:00"
context:
  id: 01H7TEMMWWKZWGB5KN9017YZRE
  parent_id: null
  user_id: null

AllardKatan, 2024-01-18:
I can concur with @jesushairdo , there are ZHA events fired but no triggers are available. One correction: if you press the power button it alternately sends the 'on' and 'off' commands, this is not reflected in the post above. 
Something also appears to go wrong with the device configuration, there are errors in the cluster binding and reporting in the GUI. I tried removing the device and adding it anew, and this made the device not bind at all. Restarting HA after removing the device allowed me to re-add it. Here's the diagnostic log:
``` 
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "6.1.71-haos",
    "supervisor": "2023.12.1",
    "host_os": "Home Assistant OS 11.4",
    "docker_version": "24.0.7",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "remeha_home": {
      "version": "0.1.13",
      "requirements": []
    },
    "scene_presets": {
      "version": "0.1.0",
      "requirements": []
    },
    "rpi_gpio": {
      "version": "2022.7.0",
      "requirements": [
        "RPi.GPIO==0.7.1"
      ]
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "zha_toolkit": {
      "version": "v1.1.8",
      "requirements": [
        "pytz"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly",
      "@TheJulianJES"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
    "loggers": [
[... 293 lines cut ...]
          "0x0008": {
            "endpoint_attribute": "level",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0300": {
            "endpoint_attribute": "light_color",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    } 
  }
} ```

Utopiah, 2024-01-22:
Similar problem here. I managed to get once (solely) an event to toggle a light.

Since then while listening to `zha_event` I don't even get anything from it, even while listening to `*` events.

MattWestb, 2024-01-22:
If its configuration is OK is shall sending more commands and `zha_event` shall showing all.
Try do one device reconfigure from the device card and waking the device up so it can receiving the commands.
Z2M device wiki https://www.zigbee2mqtt.io/devices/RC_250.html#action-enum is the commands i can sending.

closed 2024-07-27 by github-actions[bot]