zigpy/zha-device-handlers #2529: [Device Support Request] Zigbee Smart Thermostat BAC-006  _TZE204_mpbki2zm MONEY

douglascrc-git, 2023-08-17:
### Problem description

I have problems with the BAC-006 series zigbee thermostat "TZE204_mpbki2zm". Once added via ZHA, Home Assistant does not recognize any entities and therefore I cannot use the device. 

I have tried various ways but so far I have not been able to fix this compatibility issue. Also, there are a lot of user asking for the same request with the same result: failure.  Any "quirk" barely work. 

Thus, **I am willing to pay if someone  help me with a proper solution.**

Thank you for your time!  I appreciate any help. 

### Solution description

I'd like to have the device working properly as it does from the Tuya integration, reading/setting features as current temperature, set point,fan-coil speed, working mode (heat, cool, iddle), etc. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_mpbki2zm",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Diagnostic information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.7.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Madrid",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
[... 297 lines cut ...]

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

[image]
[image]
[image]
[image]

douglascrc-git, 2023-08-25:
I found a parcial solution, using the next quirk: 

`"""Map from manufacturer to standard clusters for electric heating thermostats."""
"""Tuya MCU based thermostat."""

import logging

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy

from typing import Dict, Optional, Union
from zigpy.zcl import foundation
from zigpy.zcl.clusters.hvac import Thermostat

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
)

from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaMCUCluster,
)

class TuyaTC(t.enum8):
    """Tuya thermostat commands."""
    OFF = 0x00
    ON = 0x01

class ZclTC(t.enum8):
    """ZCL thermostat commands."""
    OFF = 0x00
    ON = 0x01

TUYA2ZB_COMMANDS = {
    ZclTC.OFF: TuyaTC.OFF,
    ZclTC.ON: TuyaTC.ON,
}

MOESBHT6_TARGET_TEMP_ATTR = 0x0210  # [0,0,0,21] target room temp (degree)
MOESBHT6_TEMPERATURE_ATTR = 0x0218  # [0,0,0,200] current room temp (decidegree)
MOESBHT6_MODE_ATTR = 0x0402  # [0] manual [1] scheduled
MOESBHT6_ENABLED_ATTR = 0x0101  # [0] off [1] on
MOESBHT6_RUNNING_MODE_ATTR = 0x0424  # [1] idle [0] heating
MOESBHT6_RUNNING_STATE_ATTR = 0x0424  # [1] idle [0] heating
MOESBHT6_CHILD_LOCK_ATTR = 0x0128  # [0] unlocked [1] child-locked

_LOGGER = logging.getLogger(__name__)

class MoesBHT6ManufCluster(TuyaManufClusterAttributes, NoManufacturerCluster, TuyaLocalCluster):
    """Manufacturer Specific Cluster of some electric heating thermostats."""

    attributes = {
        MOESBHT6_TARGET_TEMP_ATTR: ("target_temperature", t.uint32_t, True),
        MOESBHT6_TEMPERATURE_ATTR: ("temperature", t.uint32_t, True),
[... 179 lines cut ...]
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesBHT6ManufCluster,
                    MoesBHT6Thermostat,
                    MoesBHT6UserInterface,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
			}
        }
    }`

Now, the device shows some entities as hvac_action and mode (heat and off) , but not all of them. 

This is how it looks now: 

[image]

I can't change the set point, fan speed or the mode to cool/fan.

douglascrc-git, 2023-08-29:
I had some issues with the temperature set-point. While the device screen displays for instance 20ºC, Home assistant saved that value as 200ºC. 

I did a little modification to the quirk in order to solve that issue. 

This is the updated quirk: 

```
"""Map from manufacturer to standard clusters for electric heating thermostats."""
"""Tuya MCU based thermostat."""

import logging

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy

from typing import Dict, Optional, Union
from zigpy.zcl import foundation
from zigpy.zcl.clusters.hvac import Thermostat

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
)

from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaMCUCluster,
)

class TuyaTC(t.enum8):
    """Tuya thermostat commands."""
    OFF = 0x00
    ON = 0x01

class ZclTC(t.enum8):
    """ZCL thermostat commands."""
    OFF = 0x00
    ON = 0x01

TUYA2ZB_COMMANDS = {
    ZclTC.OFF: TuyaTC.OFF,
    ZclTC.ON: TuyaTC.ON,
}

MOESBHT6_TARGET_TEMP_ATTR = 0x0210  # [0,0,0,21] target room temp (degree)
MOESBHT6_TEMPERATURE_ATTR = 0x0218  # [0,0,0,200] current room temp (decidegree)
MOESBHT6_MODE_ATTR = 0x0402  # [0] manual [1] scheduled
MOESBHT6_ENABLED_ATTR = 0x0101  # [0] off [1] on
MOESBHT6_RUNNING_MODE_ATTR = 0x0424  # 1[] idle [0] heating
MOESBHT6_RUNNING_STATE_ATTR = 0x0424  # [1] idle [0] heating
MOESBHT6_CHILD_LOCK_ATTR = 0x0128  # [0] unlocked [1] child-locked

_LOGGER = logging.getLogger(__name__)

class MoesBHT6ManufCluster(TuyaManufClusterAttributes, NoManufacturerCluster, TuyaLocalCluster):
[... 180 lines cut ...]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesBHT6ManufCluster,
                    MoesBHT6Thermostat,
                    MoesBHT6UserInterface,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
			}
        }
    }

```

Nevertheless, still does not support de cool mode (only heating) 😢

douglascrc-git, 2023-08-29:
I'v found in Z2MQTT a device that essentialy has the same behaviour: https://www.zigbee2mqtt.io/devices/PCT504.html

douglascrc-git, 2023-08-29:
From the tuya iot platform, I found this information about the device standar instruction set 

<html>
<body>

switch | Boolean | "{true,false}"
-- | -- | --
mode | Enum | {   "range": [     "cold",     "hot",     "wind"   ] }
eco | Boolean | "{true,false}"
temp_set | Integer | {   "unit": "℃",   "min": 50,   "max": 350,   "scale": 1,   "step": 5 }
upper_temp_f | Integer | {   "unit": "℃",   "min": 150,   "max": 350,   "scale": 1,   "step": 10 }
upper_temp | Integer | {   "unit": "℃",   "min": 150,   "max": 350,   "scale": 1,   "step": 10 }
lower_temp_f | Integer | {   "unit": "℃",   "min": 50,   "max": 150,   "scale": 1,   "step": 10 }
lower_temp | Integer | {   "unit": "℃",   "min": 50,   "max": 150,   "scale": 1,   "step": 10 }
temp_correction | Integer | {   "unit": "°C",   "min": -9,   "max": 9,   "scale": 0,   "step": 1 }
level | Enum | {   "range": [     "low",     "middle",     "high",     "auto"   ] }
child_lock | Boolean | "{true,false}"

</body>
</html>

douglascrc-git, 2023-08-29:
I installed the Tuya integration & added the thermostat through it:
[image]

Now, I wonder, is it possible to see the code use by the integration and create a quirk?

prsveloso, 2023-09-01:
I have the same thermostat.
Wishing for a solution too.

douglascrc-git, 2023-09-04:
> I have the same thermostat. Wishing for a solution too.

Hello, I did a quirk based in another quirk. 

It works at 95% but there is a little issue when you turn off the device. For any reason, the screen indeed turn off, but the device continues working in cool/heat ignoring the off instruction.

The quirk:

````
"""Map from manufacturer to standard clusters for electric heating thermostats."""
"""Tuya MCU based thermostat."""

import logging
import asyncio
from typing import Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.hvac import Thermostat, Fan
from zigpy.quirks import CustomDevice
from typing import Dict, Optional, Union
from zigpy.zcl import foundation
from zigpy.zcl.clusters.hvac import Thermostat
from zhaquirks import LocalDataCluster, Bus

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
)

from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaMCUCluster,
)

class TuyaTC(t.enum8):
    """Tuya thermostat commands."""
    OFF = 0x00
    ON = 0x01

class ZclTC(t.enum8):
    """ZCL thermostat commands."""
    OFF = 0x00
    ON = 0x01

TUYA2ZB_COMMANDS = {
    ZclTC.OFF: TuyaTC.OFF,
    ZclTC.ON: TuyaTC.ON,
}

# MOESBHT6_TARGET_TEMP_ATTR = 0x0210  # [0,0,0,21] target room temp (degree)
# MOESBHT6_TEMPERATURE_ATTR = 0x0218  # [0,0,0,200] current room temp (decidegree)
# MOESBHT6_ENABLED_ATTR = 0x0101  # [0] off [1] on
# MOESBHT6_MODE_ATTR = 0x0402  # [0] manual [1] scheduled
# MOESBHT6_CHILD_LOCK_ATTR = 0x0128  # [0] unlocked [1] child-locked
[... 539 lines cut ...]
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaFancoilManufCluster,
                    TuyaFancoilThermostat,
                    TuyaFancoilUserInterface,
                    TuyaFancoilFanCluster
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
´´´

douglascrc-git, 2023-09-04:
In case someone ask, I "created" the quirk posted before in base of those quirks:

1. https://github.com/zigpy/zha-device-handlers/issues/1284#issuecomment-1684896622
2. https://github.com/zigpy/zha-device-handlers/issues/2433#issuecomment-1689829732

prsveloso, 2023-09-04:
Thank you, @douglascrc-git
I'll try your quirk and try to learn something about this.

mmatviichuk, 2024-05-22:
Hello,

It looks like this device still needs to be supported. Are there any plans to add it to the standard?

Everyone is probably using either Quirk or connecting it via the Tuya Smart App.

1sakss0n, 2024-10-19:
I have similar issues with the 2 pipe version, cooling only. Anyone have a quirk for this one?

rmseniv, 2025-01-22:
In case someone is working on a quirk here are Tuya Data Points for this device.
`{"1":"Switch","2":"Mode","4":"ECO mode","16":"Set temperature","18":"Set temperature upper limit -℉","19":"Set temperature ceiling","20":"Set the lower limit -℉","24":"Current temperature","26":"The lower limit of temperature","27":"Temperature correction","28":"Gear","36":"State of the valve","40":"Child lock","101":"Schedule/Manual","102":"Enable_fan_control","103":"Blind zone temperature.","104":"Eco cool temp","105":"Eco heat temp","106":"Schedule","107":"Schedule","108":"Schedule","109":"Schedule","110":"Schedule","111":"Schedule","112":"Schedule"}`

douglascrc-git, 2025-01-24:
> Enable_fan_control

> In case someone is working on a quirk here are Tuya Data Points for this device. Obtained according to instructions [here](https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/advanced/support-new-devices/03_find_tuya_data_points.md).
> 
> `{ "1": "Switch", "2": "Mode", "4": "ECO mode", "16": "Set temperature", "18": "Set temperature upper limit -℉", "19": "Set temperature ceiling", "20": "Set the lower limit -℉", "24": "Current temperature", "26": "The lower limit of temperature", "27": "Temperature correction", "28": "Gear", "36": "State of the valve", "40": "Child lock", "101": "Schedule/Manual", "102": "Enable_fan_control", "103": "Blind zone temperature.", "104": "Eco cool temp", "105": "Eco heat temp", "106": "Schedule", "107": "Schedule", "108": "Schedule", "109": "Schedule", "110": "Schedule", "111": "Schedule", "112": "Schedule" }`

This is the last update I did in the quirk. Some bug were fixed, but there are some details:

Nevertheless I can't control de **fan mode** from home assistant. 

````
"""Map from manufacturer to standard clusters for electric heating thermostats."""
"""Tuya MCU based thermostat."""

import logging
import asyncio
from typing import Optional, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.hvac import Thermostat, Fan
from zigpy.quirks import CustomDevice
from typing import Dict, Optional, Union
from zigpy.zcl import foundation
from zigpy.zcl.clusters.hvac import Thermostat
from zhaquirks import LocalDataCluster, Bus

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
    NoManufacturerCluster,
    TUYA_MCU_COMMAND,
    TuyaLocalCluster,
)

from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaClusterData,
    TuyaMCUCluster,
)

class TuyaTC(t.enum8):
    """Tuya thermostat commands."""
    OFF = 0x00
    ON = 0x01

class ZclTC(t.enum8):
    """ZCL thermostat commands."""
    OFF = 0x00
    ON = 0x01

TUYA2ZB_COMMANDS = {
    ZclTC.OFF: TuyaTC.OFF,
    ZclTC.ON: TuyaTC.ON,
}

# MOESBHT6_TARGET_TEMP_ATTR = 0x0210  # [0,0,0,21] target room temp (degree)
# MOESBHT6_TEMPERATURE_ATTR = 0x0218  # [0,0,0,200] current room temp (decidegree)
# MOESBHT6_ENABLED_ATTR = 0x0101  # [0] off [1] on
[... 333 lines cut ...]

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    MoesBHT6ManufCluster,
                    TuyaFancoilThermostat,
                    TuyaFancoilUserInterface,
                    TuyaFancoilFanCluster
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
			}
        }
    }

rmseniv, 2025-01-27:
Why are you using address 0x041c for fan mode and not 0x001c?

douglascrc-git, 2025-01-31:
I was logging the device in order to pin point how to control & see the fan coil speed. So far I have seen the follow:

When I change the fan speed from the device screen I can see the next logs: 

Vel Low = 
2025-01-31 18:03:57.165 DEBUG (MainThread) [zhaquirks.tuya] [0xca20:1:0xef00] Received value [0] for attribute 0x041c (command 0x0002)

Vel med = 
2025-01-31 18:04:41.515 DEBUG (MainThread) [zhaquirks.tuya] [0xca20:1:0xef00] Received value [1] for attribute 0x041c (command 0x0002)

Vel high =
2025-01-31 18:08:50.342 DEBUG (MainThread) [zhaquirks.tuya] [0xca20:1:0xef00] Received value [2] for attribute 0x041c (command 0x0002)

Vel Auto = 
2025-01-31 18:08:07.903 DEBUG (MainThread) [zhaquirks.tuya] [0xca20:1:0xef00] Received value [3] for attribute 0x041c (command 0x0002)

It is not visualized in Home assistant. Instead of show "low" "mid" "high" "auto, it shows "auto" & "encendido -spanish"

[image]

In the other hand, when I change those values from home assistant UI,  I have these logs:

2025-01-31 18:38:38.387 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xCA20:1:0x0202]: Attribute report 'Fan Control'[fan_mode] = FanMode.On

I would like to know how how the change the way home assistant display the fan control in order to display "low" "mid", etc  & the cluster it addresses and the value it writte/send to device.

closed 2025-08-17 by github-actions[bot]

ProgrammerIamNot, 2025-08-17:
A workaround for controlling fan speed [here.](https://github.com/zigpy/zha-device-handlers/issues/1284#issuecomment-3194221375)