zigpy/zha-device-handlers #2531: _TZE200_hl0ss9oa

jonm802, 2023-08-18:
### Problem description

Please add support for tuya presence detector

### Solution description

Support for siren

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0400",
        "0x0500",
        "0xe000",
        "0xe002",
        "0xee00",
        "0xef00"
      ],
      "output_clusters": []
    },
[... 475 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

CtrlAltSpoods, 2023-08-21:
+1 for this device please :)
I've just had one delivered, hard to know what's supported before they arrive as they don't seem to specify the T****_******** numbers on store pages :/

jonm802, 2023-08-21:
https://github.com/Koenkk/zigbee2mqtt/issues/18497

jonm802, 2023-08-21:
> +1 for this device please :) I've just had one delivered, hard to know what's supported before they arrive as they don't seem to specify the T****_******** numbers on store pages :/

Just motion and illumination, identify doesn't work but sometimes causes the device to enter clear state

gfelisberto, 2023-10-23:
I also just got one. The luminance seems to update but only at 5 minutes intervals. The motion detector when I added the device seemed to work like 2-3 times and then got stuck at Detected while no one is the room for hours. Presence is nowhere to be seen :(

Is there a guide/tutorial that I could follow to add support?

GPSPerth, 2023-11-03:
Here, try this in a quirk... 

"""Device handler for Tuya ZG-205Z-A Mini 24Ghz human presence sensor."""

import math
from typing import Dict, Optional, Tuple, Union
from zigpy.profiles import zgp, zha
from zigpy.profiles.zha import DeviceType
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation

from zigpy.zcl.clusters.general import Basic, Identify, Ota, GreenPowerProxy, AnalogOutput
from zigpy.zcl.clusters.security import IasZone
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
)

from zhaquirks import MotionWithReset
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaManufCluster, TuyaNewManufCluster, TuyaZBE000Cluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class MotionCluster(MotionWithReset):
    """Motion cluster."""

    reset_s: int = 60

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaMmwRadarFadingTime(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for fading time."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self._update_attribute(self.attributes_by_name["description"].id, "Fading time")
        self._update_attribute(self.attributes_by_name["min_present_value"].id, 0)
        self._update_attribute(self.attributes_by_name["max_present_value"].id, 600)
        self._update_attribute(self.attributes_by_name["resolution"].id, 1)
        self._update_attribute(
            self.attributes_by_name["engineering_units"].id, 73
        )

class TuyaMmwRadarLargeMotionDetectionSensitivity(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for Large motion detection sensitivity."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self._update_attribute(self.attributes_by_name["description"].id, "Large motion sensitivity")
        self._update_attribute(self.attributes_by_name["min_present_value"].id, 0)
        self._update_attribute(self.attributes_by_name["max_present_value"].id, 10)
        self._update_attribute(self.attributes_by_name["resolution"].id, 1)

class TuyaMmwRadarLargeMotionDetectionDistance(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for Large motion detection distance."""
[... 308 lines cut ...]
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarStaticMotionDetectionSensitivity,
                ],
                OUTPUT_CLUSTERS: [],
            },
            8: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarStaticMotionDetectionDistance,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

GPSPerth, 2023-11-03:
Here try this in a quirk for the time being.

`"""Device handler for Tuya ZG-205Z-A Mini 24Ghz human presence sensor."""

import math
from typing import Dict, Optional, Tuple, Union
from zigpy.profiles import zgp, zha
from zigpy.profiles.zha import DeviceType
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation

from zigpy.zcl.clusters.general import Basic, Identify, Ota, GreenPowerProxy, AnalogOutput
from zigpy.zcl.clusters.security import IasZone
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
)

from zhaquirks import MotionWithReset
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaManufCluster, TuyaNewManufCluster, TuyaZBE000Cluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class MotionCluster(MotionWithReset):
    """Motion cluster."""

    reset_s: int = 60

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaMmwRadarFadingTime(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for fading time."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self._update_attribute(self.attributes_by_name["description"].id, "Fading time")
        self._update_attribute(self.attributes_by_name["min_present_value"].id, 0)
        self._update_attribute(self.attributes_by_name["max_present_value"].id, 600)
        self._update_attribute(self.attributes_by_name["resolution"].id, 1)
        self._update_attribute(
            self.attributes_by_name["engineering_units"].id, 73
        )

class TuyaMmwRadarLargeMotionDetectionSensitivity(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for Large motion detection sensitivity."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self._update_attribute(self.attributes_by_name["description"].id, "Large motion sensitivity")
        self._update_attribute(self.attributes_by_name["min_present_value"].id, 0)
        self._update_attribute(self.attributes_by_name["max_present_value"].id, 10)
        self._update_attribute(self.attributes_by_name["resolution"].id, 1)

class TuyaMmwRadarLargeMotionDetectionDistance(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for Large motion detection distance."""
[... 309 lines cut ...]
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarStaticMotionDetectionSensitivity,
                ],
                OUTPUT_CLUSTERS: [],
            },
            8: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarStaticMotionDetectionDistance,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
`

igorsantos07, 2024-01-28:
Your report is curious, @gfelisberto. I got two of the same model, which seem to be quite similar to original post - it's even referred on the Hubitat post as well. Mine are `TS0225`, `_TZE200_2aaelwxk`, one is square and the other is the long version. The luminance and motion sensors work great for me on a standard ZHA installation, using a Sonoff Zigbee Dongle Plus.

But then, I'm also here in hopes some extra settings get exposed as well :pray: 

Here are the diagnostics files for the:
- [long version](https://github.com/zigpy/zha-device-handlers/files/14074792/zha-4b673429c0955d0e633571e38e6539dd-_TZE200_2aaelwxk.TS0225-4d21176b04087f9911b9fe1983b4d9c8.json.txt)
- [square version](https://github.com/zigpy/zha-device-handlers/files/14074794/zha-4b673429c0955d0e633571e38e6539dd-_TZE200_2aaelwxk.TS0225-52ba3740c6113cee95d8d4166b2b9ddc.json.txt)
- [a wifi version of the long one (which shows various settings, as expected for such sensors)](https://github.com/zigpy/zha-device-handlers/files/14074796/tuya-02266b4b334e525dfb4a68819f7d43a3-Presenca.cozinha-c1453d701845a73ee97db9157b308ed8.json.txt). _I have no clue if this is of any use or even if they're really similar or just the casing, but in case it's useful..._ :sweat_smile:

MD0005, 2024-07-19:
Im here as well in search of some solution of my TS0225, _TZE200_2aaelwxk 🙏
The device detects motion and luminance, but missing settings.

bobocancan, 2024-10-06:
I can confirm the illuminance value of this device, 2aaelwxk, is missing in HomeAssistant Tuya Integration.