zigpy/zha-device-handlers #2532: [BUG] Lumi MEAU01 plug no power measurement and incorrect device temperature

ruudburger90, 2023-08-18:
### Bug description

All the sensors give 0 values, and the device temperature is 0,2 or 0,3. I've read all the topics about this problem, and I'm aware the firmware of my Zigbee coordinator and smart plug are outdated (see additional information). Does this cause the problem, or is there another way to fix it? I'm not familair yet with updating firmware of coordinators or devices.

### Steps to reproduce

The bug is in the sensors.

### Expected behavior

I expect to read sensor values.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
```

</details>

[... 881 lines cut ...]
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Additional information

Home Assistant OS 2023.8.2
Zigbee coordinator CC2531, Z-Stack Home 1.2 (build 20190608)
Lumi MEAU01 plug, app version 22

TheJulianJES, 2023-08-21:
"CC2531, Z-Stack Home 1.2 (build 20190608)" is your issue.
Instead of updating the firmware, you likely want to buy a newer coordinator that's CC2652 based for example.

ruudburger90, 2023-08-28:
Thanks Julian! I will directly order the one you mentioned. I also already have 32 zigbee devices connected and still counting. So I guess the network will get to its limit. Keep you updated about the device!

closed 2024-06-28 by github-actions[bot]