zigpy/zha-device-handlers #2533: [Device Support Request] TS0601 by _TZE204_nklqjk62

Axel-72, 2023-08-18:
### Problem description

I'm new with ZHA, and my first difficulties were this product, it seems to be unsupported at this moment, and i don"t know how to make hard code python for that zigbee device.
I don't really know how to do this.

### Solution description

I need a python hard code probably

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[image]

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here] "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, omplex_descriptor_available=0,  user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>,  mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>,  manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752,  maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True,  *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True,  *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
 
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

evlas, 2023-09-15:
these quirks seem to work

"""Tuya based cover and blinds."""
from typing import Dict

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import NoManufacturerCluster, TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
    TuyaOnOff,
)
from zhaquirks.tuya.ts0601_dimmer import TuyaOnOffNM

ZONE_TYPE = 0x0001

class ContactSwitchCluster(TuyaLocalCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class TuyaGarageManufCluster(NoManufacturerCluster, TuyaMCUCluster):
    """Tuya garage door opener."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # ramdom attribute IDs
            0xEF02: ("dp_2", t.uint32_t, True),
            0xEF04: ("dp_4", t.uint32_t, True),
            0xEF05: ("dp_5", t.uint32_t, True),
            0xEF0B: ("dp_11", t.Bool, True),
            0xEF0C: ("dp_12", t.enum8, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        # garage door trigger ¿on movement, on open, on closed?
        1: DPToAttributeMapping(
            TuyaOnOffNM.ep_attribute,
            "on_off",
        ),
        2: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_2",
        ),
        3: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            lambda x: IasZone.ZoneStatus.Alarm_1 if x else 0,
            endpoint_id=2,
        ),
        4: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_4",
        ),
        5: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_5",
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_11",
        ),
        # garage door status (open, closed, ...)
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_12",
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class TuyaGarageSwitchTO(CustomDevice):
    """Tuya Garage switch."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_nklqjk62", "TS0601"),
            ("_TZE200_wfxuhoea", "TS0601"),
            ("_TZE204_nklqjk62", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    ContactSwitchCluster
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

evlas, 2023-09-15:
```python
"""Tuya based cover and blinds."""
from typing import Dict

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import NoManufacturerCluster, TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
    TuyaOnOff,
)
from zhaquirks.tuya.ts0601_dimmer import TuyaOnOffNM

ZONE_TYPE = 0x0001

class ContactSwitchCluster(TuyaLocalCluster, IasZone):
    """Tuya ContactSwitch Sensor."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

    def _update_attribute(self, attrid, value):
        self.debug("_update_attribute '%s': %s", attrid, value)
        super()._update_attribute(attrid, value)

class TuyaGarageManufCluster(NoManufacturerCluster, TuyaMCUCluster):
    """Tuya garage door opener."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            # ramdom attribute IDs
            0xEF02: ("dp_2", t.uint32_t, True),
            0xEF04: ("dp_4", t.uint32_t, True),
            0xEF05: ("dp_5", t.uint32_t, True),
            0xEF0B: ("dp_11", t.Bool, True),
            0xEF0C: ("dp_12", t.enum8, True),
        }
    )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        # garage door trigger ¿on movement, on open, on closed?
        1: DPToAttributeMapping(
            TuyaOnOffNM.ep_attribute,
            "on_off",
        ),
        2: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_2",
        ),
        3: DPToAttributeMapping(
            ContactSwitchCluster.ep_attribute,
            "zone_status",
            lambda x: IasZone.ZoneStatus.Alarm_1 if x else 0,
            endpoint_id=2,
        ),
        4: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_4",
        ),
        5: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_5",
        ),
        11: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_11",
        ),
        # garage door status (open, closed, ...)
        12: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "dp_12",
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
        3: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        5: "_dp_2_attr_update",
        11: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class TuyaGarageSwitchTO(CustomDevice):
    """Tuya Garage switch."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_nklqjk62", "TS0601"),
            ("_TZE200_wfxuhoea", "TS0601"),
            ("_TZE204_nklqjk62", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0x0051
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    ContactSwitchCluster
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

```

landaisbenj, 2023-09-16:
hello @evlas did you know where can we have some documentation to test this ? (i'm new too with quirks)

dualmacops, 2023-12-21:
This works, but it's detected as a light instead of a garage door
[image]

tszdr, 2023-12-28:
Paired, quirk seems to be loaded but no enitities shown:
[image]
Works fine when setting/reading attributes:
[image]
 I'm not sure what I did wrong... Any idea I should try/check?

intra-aud, 2024-01-05:
I have the exact same issue, I woke up one morning to find the entity unavailable. Removed the working device to do a re-pair in ZHA and now the light entity is missing and the quirk appears to be loaded

Manually writing to the cluster works. 

Has anyone else this or have any suggestions?

aussiebish, 2024-01-10:
Same issue here, although I've never been able to get the Entities showing. Please help! (also, I'm assuming I don't need local tuya or similiar installed, just using ZHA directly)

I've followed the steps, added the custom ZHA py file, adn config entries, but still no entities.
[image]

[image]
 I have the following file entries (but with carriage returns, not sure why they don't display):
Configuration.yaml
`zha:
  database_path: /config/zigbee.db
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks`

Folders created:
[image]

and ts0601_garage.py is as #[above](https://github.com/zigpy/zha-device-handlers/issues/2533#issuecomment-1721308395)

intra-aud, 2024-01-12:
So i didn't get the entity part working but i did figure out how to call the service directly from HA. 

Just as a workaround until someone can input why the entity is missing.

Use the call service command - select "Zigbee Home Automation: Set zigbee cluster attribute"

set the conditions as follows (i'll include the yaml) 

service: zha.set_zigbee_cluster_attribute
data:
  ieee: <input IEEE> 
  endpoint_id: 1
  cluster_id: 61184
  attribute: 61185
  value: 0

That should allow you to atleast write an automation assuming you can call the attribute directly.

aussiebish, 2024-01-12:
Thanks @intra-aud , this works perfectly.  Obviously not the ideal solution, but this achieves everything I needed it to on my dashboard.  I'm calling the set attribute command from a Script, so that I can just add a 'Run' command to my Dashboard under the existing 'Garage' entity card.  I guess we just wait and see if the original tuya entity starts getting fully detected in the future. :)

mfdlr, 2024-02-07:
I had the same problem of no entities being detected.

As I was continously updating the file I missed deleting the cache, basically
- (1) deleting the `__pycache__` folder in your `custom_quirks_path` folder is important (see [comment here](https://github.com/zigpy/zha-device-handlers/issues/1898#issuecomment-1304922525))
- (2) adding the `database_path` to my zha config. Before everything was ignored
```yaml
zha:
  database_path: /config/zigbee.db
  enable_quirks: true
  custom_quirks_path: /config/zhaquirks/
```

Clement-B, 2024-02-11:
I used the quirks that create a light and binary sensor, it works for me but even if light was a swtich, I was disapointed to not have a real garage door or gate entity.

Downsides: 
* I reproduce that with a switch or light, trigger of open/close command was only fired off when entity is turn on.
* During opening / closing door, we dont have state return so it can be difficult to see if action has been fired

So, to compensates those inconvenience, I have set up multiple configurations.

In my case, it was for a gate.

# Open/close only when turning on

I use a simple automation because I dit not spend much time on this, seems a dirty solution but was a quick fix.
I just turn off light/switch just after it has been turned during 1 second.

```yaml
alias: "Reset state of Garage door / Gate"
description: ""
trigger:
  - platform: state
    entity_id:
      - light.gate # Or switch.garage_door depending of quirks used
    to: "on"
    for:
      hours: 0
      minutes: 0
      seconds: 1
condition: []
action:
  - service: light.turn_off
    target:
      entity_id: light.gate
    data: {}
mode: single
```

# State of opening/closing 

I created some helpers in HA interface:
* `input_boolean.gate_opening`: Switch that when turning on, indicate that gate is in a opening state.
* `input_boolean.gate_closing`: Switch that when turning on, indicate that gate is in a closing state.
* `input_number.duration_opening_gate`: Number of seconds the gate make to open. 
* `input_number.duration_closing_gate`: Number of seconds the gate make to close. It can be different of the opening duration.

Then a simple automation to switch on/off the boolean input depending of the state and duration:

```yaml
alias: "Update gate status"
description: ""
trigger:
  - platform: state
    entity_id:
      - light.gate
    to: "on"
condition: []
action:
  - if:
      - condition: state
        entity_id: binary_sensor.gate_sensor
        state: "off" # Mean that gate is open in my configuration
    then:
      - service: input_boolean.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: input_boolean.gate_closing
      - delay:
          hours: 0
          minutes: 0
          seconds: "{{ states('input_number.duration_closing_gate') | int }}"
          milliseconds: 0
      - service: input_boolean.turn_off
        target:
          entity_id: input_boolean.gate_closing
        data: {}
    else:
      - service: input_boolean.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: input_boolean.gate_opening
      - delay:
          hours: 0
          minutes: 0
          seconds: "{{ states('input_number.duration_opening_gate') | int }}"
          milliseconds: 0
      - service: input_boolean.turn_off
        metadata: {}
        data: {}
        target:
          entity_id: input_boolean.gate_opening
mode: single
```

With this automation, I have 2 switchs, one will be on when gate is opening the other one when gate is closing. 

_Obviously, it's not the more reliable you can have. Multiple calls could mess up those switch states, wind will speed up or slow down closing/opening action, gate security could occurs etc._

# Gate/Garage door entity

Finally, I can create a custom entity that I could add to my dashboard.
I simply add to my `configuration.yaml` file a new custom cover entity following [HA documentation](https://www.home-assistant.io/integrations/cover.template/): 

```yaml
cover:
  - platform: template
    covers:
    gate:
      device_class: gate # can be 'garage' as well
      friendly_name: "Gate (custom)"
      icon_template: >-
          {% if is_state('binary_sensor.gate_sensor', 'on') %}
            mdi:gate
          {% else %}
            mdi:gate-open
          {% endif %}
      value_template: >-
          {% if is_state('input_boolean.gate_opening', 'on') %}
            opening
          {% elif is_state('input_boolean.gate_closing', 'on') %}
            closing
          {% elif is_state('binary_sensor.gate_sensor', 'on') %}
            closed
          {% elif is_state('binary_sensor.gate_sensor', 'off') %}
            open
          {% endif %}
      availability_template: "{{ is_state('binary_sensor.gate_sensor', 'on') or is_state('binary_sensor.gate_sensor', 'off') }}"
      open_cover:
        service: light.turn_on
        data: {}
        target:
          entity_id: light.gate
      close_cover:
        service: light.turn_on
        data: {}
        target:
          entity_id: light.gate
```

Now I can have a well working card of my gate on my dashboard without any specific card configuration or script calls.

# Go further ?

When using this entity in a dashboard, when gate is opening, we will only see a message saying "opening", not really how much time it remain to be fully opened. Like a classic cover, we miss the percentage progression of opening.

I haven't tested it really but it should be pretty easy with the use of attribut `position_template` instead of `value_template` in custom entity.

It should looks like something like this:

```yaml
      position_template: >-
          {% if is_state('input_boolean.gate_opening', 'on') %}
            {{ ((as_timestamp(now()) - as_timestamp(states.input_boolean.gate_opening.last_changed))/states('input_number.duration_opening_gate')|int)|int }} 
          {% elif is_state('input_boolean.gate_closing', 'on') %}
            {{ ((as_timestamp(now()) - as_timestamp(states.input_boolean.gate_closing.last_changed))/states('input_number.duration_closing_gate')|int)|int }} 
          {% elif is_state('binary_sensor.gate_sensor', 'on') %}
            0
          {% elif is_state('binary_sensor.gate_sensor', 'off') %}
            100
          {% endif %}
```

Hope it can help some people while we wait a fully integrated module in ZHA 🙂

liouma, 2024-04-02:
Thanks for the quirk, it's working fine.
Turning on the switch activate the physical connection for a second, then it is automaticaly deactivated. This is the expected behaviour for a garage door but not for the device I use. I need a normal on/off switch.

Is this behaviour built in the hardware or is it programmed in the quirk? Do you know if there is a software way to modify it??
Thanks a lot!

noseshimself, 2024-06-27:
the built-in quirk is definitely not working and I can't override it:

[image]

Siress, 2024-06-28:
> the built-in quirk is definitely not working and I can't override it: [...]

I'm having the exact same issue.

viktor-sc, 2024-09-04:
Hi,
unfortunately I don't have any experiance in quirks.
I've the same problem and with the here quirk posted here it works. But - as already mentioned - as a light switch.
I wonder if this can be changed as the quirk says:

`ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaGarageManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },`

So it's build in in the quirk. Is there any chance this can be changed to a garage door directly?

Thanks

freiadam, 2024-11-14:
any updates on this?

mvanimpe, 2024-11-15:
@freiadam -> check my post over here: https://community.home-assistant.io/t/how-to-add-a-garage-door-opener-to-zha/454783/6

I got it working with ZHA

noseshimself, 2024-11-20:
> @freiadam -> check my post over here: https://community.home-assistant.io/t/how-to-add-a-garage-door-opener-to-zha/454783/6
> 
> I got it working with ZHA

Too late for me... I bought a dual-relay-board with an ESP8266-01s as conrtroller for 7€, connected it (it has a 9V to 25V power supply on board) to the internal power connector of my Hörmann motor, used two door sensors as "door is open" and "door is closed" sensors (that way I can even tell which direction the door is going if someone is pressing the emergency button inside or using the original wireless remote) and don't have to invoke magic incantations for ZHA.

After adding a second coordinator for Z2M the TS0601 is working without any problem (but also without any garage), too.

rdehuyss, 2025-01-06:
Thanks, @mvanimpe ! Your solution works for me. Only request: the entity is shown as a switch in HomeAssistant where a button would be more applicable.

In HomeAssistant, the relay closes for 500ms and open agains after I open the switch and close it again immediately.

Bruners, 2025-05-07:
Working quirk based on https://github.com/zigpy/zha-device-handlers/pull/3891 since it does not work with the current quirk. Had to change the converter to 1 on the door_contact sensor for my application, but this should probably be changeable in the UI depending on how the sensor is mounted?

`"""Tuya garage relays."""

from zigpy.quirks.v2 import BinarySensorDeviceClass, EntityPlatform, EntityType
from zigpy.quirks.v2.homeassistant import UnitOfTime

### From init
from collections.abc import Callable
from typing import Any, Optional
from zigpy.quirks.v2 import CustomDeviceV2, QuirkBuilder, QuirksV2RegistryEntry
from zigpy.quirks.v2.homeassistant import EntityPlatform, EntityType
from zigpy.quirks.v2.homeassistant.binary_sensor import BinarySensorDeviceClass
from zigpy.quirks.v2.homeassistant.number import NumberDeviceClass
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.measurement import (
    PM25,
    CarbonDioxideConcentration,
    FormaldehydeConcentration,
    IlluminanceMeasurement,
    RelativeHumidity,
    SoilMoisture,
    TemperatureMeasurement,
)

###

from zhaquirks.tuya import TUYA_CLUSTER_ID
from zhaquirks.tuya.builder import TuyaQuirkBuilder

class ExtendedTuyaQuirkBuilder(TuyaQuirkBuilder):
    def tuya_binary_sensor_converter(
        self,
        dp_id: int,
        attribute_name: str,
        converter: Optional[Callable[[Any], Any]] = None,
        endpoint_id: int = 1,
        entity_type: EntityType = EntityType.DIAGNOSTIC,
        device_class: BinarySensorDeviceClass | None = None,
        initially_disabled: bool = False,
        attribute_initialized_from_cache: bool = True,
        translation_key: str | None = None,
        fallback_name: str | None = None,
    ) -> QuirkBuilder:
        """Add an EntityMetadata containing BinarySensorMetadata and return self.

        This method allows exposing a binary sensor entity in Home Assistant.
        """
        self.tuya_dp_attribute(
            dp_id=dp_id,
            attribute_name=attribute_name,
            converter=converter,
            type=t.Bool,
            access=foundation.ZCLAttributeAccess.Read
            | foundation.ZCLAttributeAccess.Report,
        )
        self.binary_sensor(
            attribute_name=attribute_name,
            cluster_id=TUYA_CLUSTER_ID,
            endpoint_id=endpoint_id,
            entity_type=entity_type,
            device_class=device_class,
            initially_disabled=initially_disabled,
            attribute_initialized_from_cache=attribute_initialized_from_cache,
            translation_key=translation_key,
            fallback_name=fallback_name,
        )

        return self

class TuyaDoorStatus(t.enum8):
    """Tuya garage door status enum."""

    Open_time_alarm = 0x00
    Run_time_alarm = 0x01
    Normal = 0x02

(
    ExtendedTuyaQuirkBuilder("_TZE204_nklqjk62", "TS0601")
    .tuya_dp_attribute(
        dp_id=1,
        attribute_name="garage_door_trigger",
        type=t.Bool,
    )
    .write_attr_button(
        attribute_name="garage_door_trigger",
        cluster_id=TUYA_CLUSTER_ID,
        attribute_value=True,
        translation_key="open_door",
        fallback_name="Open door",
    )
    .write_attr_button(
        attribute_name="garage_door_trigger",
        cluster_id=TUYA_CLUSTER_ID,
        attribute_value=False,
        translation_key="close_door",
        fallback_name="Close door",
    )
    .tuya_number(
        dp_id=2,
        attribute_name="relay_countdown",
        type=t.uint32_t,
        min_value=0,
        max_value=43200,
        step=1,
        unit=UnitOfTime.SECONDS,
        translation_key="relay_countdown",
        fallback_name="Relay countdown",
    )
    .tuya_binary_sensor_converter(
        dp_id=3,
        attribute_name="door_contact",
        converter=lambda x: x == 1,
        entity_type=EntityType.STANDARD,
        device_class=BinarySensorDeviceClass.GARAGE_DOOR,
        translation_key="door_contact",
        fallback_name="Door state",
    )
    .tuya_number(
        dp_id=4,
        attribute_name="run_time",
        type=t.uint32_t,
        min_value=0,
        max_value=120,
        step=1,
        unit=UnitOfTime.SECONDS,
        translation_key="run_time",
        fallback_name="Run time",
    )
    .tuya_number(
        dp_id=5,
        attribute_name="open_alarm_time",
        type=t.uint32_t,
        min_value=0,
        max_value=86400,
        step=1,
        unit=UnitOfTime.SECONDS,
        translation_key="open_alarm_time",
        fallback_name="Open alarm time",
    )
    .tuya_enum(
        dp_id=12,
        attribute_name="door_status",
        enum_class=TuyaDoorStatus,
        entity_platform=EntityPlatform.SENSOR,
        entity_type=EntityType.DIAGNOSTIC,
        translation_key="door_status",
        fallback_name="Self test result",
    )
    .tuya_enchantment()
    .skip_configuration()
    .add_to_registry()
)`