zigpy/zha-device-handlers #2534: [Device Support Request] TZE204_81yrt3lo

fernac03, 2023-08-18:
### Problem description

Problem description
Only device exist, with anything else.

TS0601
por _TZE204_81yrt3lo
Conectado a través de [Zigbee Coordinator]IEEE: a4:c1:38:54:5f:2b:61:16
Nwk: 0xe4c3
Device Type: Router
LQI: 83
RSSI: Desconhecido
Visto pela Última Vez: 2023-08-18T15:22:22
Fonte de Energia: Mains

Device https://pt.aliexpress.com/item/1005004128359494.html?spm=a2g0o.productlist.main.1.70b83902CvC3hg&algo_pvid=22eddbfe-c69f-442b-b1b9-a316e33500d1&aem_p4p_detail=202308181131447013297338416930004743781&algo_exp_id=22eddbfe-c69f-442b-b1b9-a316e33500d1-0&pdp_npi=4%40dis%21BRL%21148.76%2162.49%21%21%2128.47%21%21%402101f49816923835044257999e4d09%2112000028133309983%21sea%21BR%21140573505%21&curPageLogUid=TvsjIzGKCbpp&search_p4p_id=202308181131447013297338416930004743781_1

I've see an external converter js to run it on z2m.

### Solution description

I post it here if anyone can adapt it as a custom quirk or include it on ZHA library

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.8.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Sao_Paulo",
    "os_name": "Linux",
    "os_version": "5.10.0-21-amd64",
    "supervisor": "2023.08.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "24.0.5",
    "chassis": "desktop",
    "run_as_root": true
  },
  "custom_components": {
    "correios": {
      "version": "1.0",
      "requirements": []
    },
[... 396 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

fernac03, 2023-10-13:
is  a quirk on other post

[ https://github.com/zigpy/zha-device-handlers/issues/2553](url)

Lobo2022, 2023-10-20:
I have this device too. 
Thank you in advance for a quirk.

closed 2024-04-24 by github-actions[bot]