zigpy/zha-device-handlers #2536: [Device Support Request] TS0601 by _TZE204_vmcgja59

matthewcoates, 2023-08-19:
### Problem description

Hey I coulde use some help with this quirk. I just got this 16ch tuya relay switch board but no exsisting quirks work with it and my custom quirk only controlled 6 out of the 16 relays. 

Thanks!

https://vi.aliexpress.com/item/1005005802197374.html?spm=a2g0o.order_detail.order_detail_item.28.17c1f19cMdSDOJ&gatewayAdapt=glo2vnm

and here is the custom quirk i built that only worked for 6 relays when it should have controlled 8 of the 16 i used this other quirk to make it

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_switch.py

"""Tuya DP based switches."""
from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    MoesSwitchManufCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)

class TuyaQuadrupleSwitch_GP(TuyaSwitch):
    """Tuya quadruple channel switch with GreenPowerProxy cluster device."""

    signature = {
        MODELS_INFO: [
            ("_TZE200_1n2kyphz", "TS0601"),
            ("_TZE200_mexisfik", "TS0601"),  # reported #1634
            ("_TZE204_vmcgja59", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81 device_version=1
            # input_clusters=[0, 4, 5, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
[... 447 lines cut ...]
    }
  }
}``

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

matthewcoates, 2023-10-02:
got it working due to these fine people
https://github.com/zigpy/zha-device-handlers/issues/2492
[ts0601_switch_8.txt](https://github.com/zigpy/zha-device-handlers/files/12784904/ts0601_switch_8.txt)

javicalle, 2023-10-05:
Probably not in HA 2023.10 but would be included in the 2023.11

PR #2616 (2023-10-04): Add new Tuya multi switch devices (x8, x16)
## Proposed change

Adds new implementations for the Tuya x8 and x16 switch devices

## Additional information

Fixes #2492
Fixes #2536

## Checklist

- [ ] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
commit 10a17d3b (2023-10-04): Add new Tuya multi switch devices (x8, x16) (#2616)

Fixes #2492
Fixes #2536