zigpy/zha-device-handlers #2537: [Device Support Request] - Zemismart Zigbee Switch

DM-AU, 2023-08-21:
### Problem description

The switch integrates into Home Assistant, and when using the physical buttons the entities update in Home Assistant as expected. However, the issue is when using Home Assistant to toggle the switches on or off will cause ALL the switch entities (of the device) to toggle on or off.

I have the 3 switch unit, though they sell these in 1 and 2 switch units as well.

Examples:
1) Using HA to toggle switch 1 ON will turn on switch 1 + 2 + 3. HA updates each entity to ON. 
2) Using HA to toggle switch 1 OFF will turn off switch 1 + 2 + 3. HA updates each entity to OFF. 
3) Using the device to toggle switch 1 ON will turn switch 1 on. HA updates switch 1 entity to ON.
4) Using the device to toggle switch 1 OFF will turn switch 1 off. HA updates switch 1 entity to OFF.

I use switch 1 as the example above, but the issue occurs on switch 2 and 3 also. See attached video.

I would love for this to work as expected as its quite a nice physical switch and would like to order more.

### Solution description

Have Home Assistant only trigger the relevant switch entity.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

https://github.com/zigpy/zha-device-handlers/assets/66342521/24ce59c6-baf4-4a1c-b62d-de1eddb05e71

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0003",
        "0x0004",
[... 467 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Thankyou!!!

DM-AU, 2023-09-22:
Hi,

Just wondering what the general turn around time on implementing fixes? Is there anything more I can provide to assist? 

Thanks

AthosXtreme, 2023-11-04:
Add variation 06 to [zhaquirks/tuya/ts001x.py](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts001x.py) and add it as a custom quirk.

```
class TuyaTripleGang_var06(EnchantedDevice, TuyaSwitch):
    """Tuya 3 gang light switch (variation 06)."""

    signature = {
        # "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0,
        # user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>,
        # mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66,
        # maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66,
        # descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False,
        # *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False,
        # *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
        MODEL: "TS0013",
        ENDPOINTS: {
            # "profile_id": 260, "device_type": "0x0100",
            # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006"],
            # "out_clusters": ["0x000a","0x0019"]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
            # "profile_id": 260, "device_type": "0x0100",
            # "in_clusters": ["0x0003","0x0004","0x0005","0x0006"],
            # "out_clusters": []
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # "profile_id": 260, "device_type": "0x0100",
            # "in_clusters": ["0x0003","0x0004","0x0005","0x0006"],
            # "out_clusters": []
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

closed 2024-06-06 by github-actions[bot]

oopswow, 2025-04-06:
I have the same problem with the same switch.