zigpy/zha-device-handlers #2538: [Device Support Request] TZ3000_wkr3jqmr

Woll3, 2023-08-21:
### Problem description

**Product:**
- Smart Switch based on TS0004
- Model: TG-003-RF
- Input: AC/DC 7-32V / USB 5V
- Output: 10A/Gang

**Zigbee logs:**
- Manufacturer: _TZ3000_wkr3jqmr
- Model: TS0004
- Name: _TZ3000_wkr3jqmr TS0004
- Quirk Class: zhaquirks.tuya.ts000x.Switch_4G_GPP

**Problem:**
- Controll entities are unavailable (4 ligth entities)
- Configuration entities are unavailable (Backlight mode entity, Power on state entity)

**Setup:**
- automatically added via ZHA
- quirk: zhaquirks.tuya.ts000x.Switch_4G_GPP.py was automatically selected

### Solution description

ZHA support for the smart switch ZG-003-RF (_TZ3000_wkr3jqmr)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": []
    },
[... 736 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheJulianJES, 2023-08-21:
> Configuration entities are unavailable

Then they're not implemented [in ZHA](https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/switch.py).
If a quirk is applied and the custom attributes are set, you should be able to at least change them using "Manage Zigbee device" -> "Clusters".

javicalle, 2023-08-21:
Can it be a pairing problem?
It seems to be a battery powered device. Are you keeping awake the device during all the pairing process?

Woll3, 2023-08-22:
> Can it be a pairing problem? It seems to be a battery powered device. Are you keeping awake the device during all the pairing process?

It is not battery powered. The unit was powered the complete pairing process. I tried it with 24 VAC and with the Mirco USB port.

Woll3, 2023-08-22:
> Can it be a pairing problem? It seems to be a battery powered device. Are you keeping awake the device during all the pairing process?

It is not battery powered. The unit was powered the complete pairing process. I tried it with 24 VAC and with the Mirco USB port.

> Its one DIN-rail mounted 4 relay module with RF433 functionality. It cant being one true CE marking (or its the Chine Export marking) then its not written on the device the maximum operation voltage on the relays only the power supply input.

You are right.
[image]

Woll3, 2023-08-25:
> Shall being implanted on the custom cluster on all endpoints:
> 
> https://github.com/zigpy/zha-device-handlers/blob/e9099ed26eb129e960934965cf5f0b1273b38e4f/zhaquirks/tuya/__init__.py#L989-L996
> 
> And the switch part is only sending events and not having some "switch" in the GUI ad HA standard.

In which cluster can I find the current status of the relay. How can I change the relay status? How can I use the events in automations?
Thanks in advance!

Woll3, 2023-08-28:
The switch is now working for me with the standard ZHA quirk: zhaquirks.tuya.ts000x.Switch_4G_GPP
I had to enable the entities. I don't know why!
[image]
[image]

Thephaseshifter, 2024-01-18:
Can someone help me out on this one?

Installed the same Chinese 4 channel switch.
Had a small automation made for 2 channels, but the channel 1 kept switching on and off irregularly (on its own?) So deleted the switch in HA and re-configured it and tested it both with and without automation, but remains the same irregular behaviour.

[image]

[image]

closed 2024-07-23 by github-actions[bot]