zigpy/zha-device-handlers #2542: [BUG]

ialexyi, 2023-08-23:
### Bug description

I have got two doors sensors TS0203
by _TZ3000_decxrtwa. I have added it to the HA.
The configuration process has failed. Even after restarting the Ha the sensor doesn't work at all.

Home Assistant 2023.8.3
Supervisor 2023.08.1
Operating System 10.5
Frontend 20230802.1 - latest

[image]

### Steps to reproduce

Bind the device TS0203
by _TZ3000_decxrtwa.

[image]

[image]

### Expected behavior

The door sensor needs to be properly configured. Needs to receive door state. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

nocusham, 2023-08-29:
can confirm the problem described by ialexyi

vladkozlov69, 2023-09-14:
I can confirm on HA 9.2 it shows "Configuring" popup but never finish pairing.

closed 2023-12-07 by ialexyi