zigpy/zha-device-handlers #2544: [Device Support Request] TS0601 by _TZE200_pw7mji0l - Add Power Source

DLV111, 2023-08-26:
### Problem description

The device added here https://github.com/zigpy/zha-device-handlers/pull/1616 doesn't expose the battery level.

I'd love to get this code extended to also provide the battery level

I see its managed by - zhaquirks.tuya.ts0601_cover.TuyaZemismartSmartCover0601_3

I'm not really sure how to query the device to get all this information out of it, but i'm happy to work on this PR myself if i can get some guidance.

### Solution description

I'd like to be able to see the battery level on these devices.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102",
        "0xef00"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_pw7mji0l",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_cover.TuyaZemismartSmartCover0601_3"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.8.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Australia/Adelaide",
    "os_name": "Linux",
[... 319 lines cut ...]
    }
  }
}```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

DLV111, 2023-08-26:
I've actually just noticed this PR is open do this may be a duplicate - https://github.com/zigpy/zha-device-handlers/pull/2404

jezzaaa, 2023-09-05:
Same here. I'm an amateur programmer, but I have no understanding about how the quirks work, and not much experience with git/github either. But I'll see what I can do.

I also don't see entities for LQI or RSSI. And the Zigbee info shows the power source as "Battery or Unknown" so that's probably not being populated either.

closed 2024-03-10 by github-actions[bot]

jezzaaa, 2024-03-11:
> There hasn't been any activity on this issue recently. Due to the high number of incoming GitHub notifications, we have to clean some of the old issues, as many of them have already been resolved [...]

Upgrading to 2024.3.0 didn't solve the issue.