zigpy/zha-device-handlers #2546: [Device Support Request] _TZE204_sxm7l9xa

Manx-Nostromo, 2023-08-26:
### Problem description

Presence sensor always shows up as a router with no parameters, even when adding to a custom quirk.

[image]

### Solution description

I have had these working easily in Hubitat, They are very good, would like to be able to easily add them into HA.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_sxm7l9xa",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.8.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
[... 832 lines cut ...]
                    TuyaMmwRadarMaxRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

</details>

javicalle, 2023-08-26:
Your quirk signature doesn't match with yours device's one. It lacks the GPP endpoint. 

Maybe the `MmwRadarMotionGPP` would match yours device signature:
https://github.com/zigpy/zha-device-handlers/blob/e9099ed26eb129e960934965cf5f0b1273b38e4f/zhaquirks/tuya/ts0601_motion.py#L391C7-L391C24

Or maybe you just could add the GPP endpoint in the signature and in the replacement part. It would depend on the device capabilities.

Manx-Nostromo, 2023-08-26:
@javicalle I have also tried the option you have suggested, but still doesn't see the quirk.
[image]
[image]
[image]

I am very new to this and have no python experience, could you please help?
Screen shots show my current setup.
Do I need to be adding more than just the product ID to the quirk file?

javicalle, 2023-08-27:
Replace your `ts0601_motion.py` with the current version from:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_motion.py

(Make a backup if you have any other local quirk from this file!!!!)

Search for the `MmwRadarMotionGPP` class and add there your device. It would be something like:
```python
class MmwRadarMotionGPP(CustomDevice):
    """Millimeter wave occupancy sensor."""

    signature = {
        #  endpoint=1, profile=260, device_type=81, device_version=1,
        #  input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
        MODELS_INFO: [
            ("_TZE200_ar0slwnd", "TS0601"),
            ("_TZE200_sfiy5tfs", "TS0601"),
            ("_TZE200_mrf6vtua", "TS0601"),
            ("_TZE204_sxm7l9xa", "TS0601"),
        ],
        .../...
```

Save changes, restart HA and pair the device again.

Manx-Nostromo, 2023-08-27:
Thanks, I'll give this a try.

Manx-Nostromo, 2023-08-29:
The Presence senors are now recognised, two elements are also available, Presence and luminance.
Which is a big step forward.
Only problem is they aren't sensing any movement, oh well back to a PIR sensor.

smithbill17, 2023-09-02:
> they aren't sensing any movement

This is ts0601_motion.py I used for my  _TZE204_sxm7l9xa  and it's sensing 'presence' just fine:

https://github.com/zigpy/zha-device-handlers/blob/b2c5ead8ca75e6f406c941fec0cd6665f59724bb/zhaquirks/tuya/ts0601_motion.py

I'm not sure of any differences to the one that @javicalle linked above - there seems to be a LOT of variations on ts0601_motion.py, but I've posted the link for works & even exposes some config sliders (although I'm not sure they actually work as they just 'reset' to defaults if you try changing them).

Manx-Nostromo, 2023-09-02:
@smithbill17 Thanks, will give that quirk a try, althouth it looks to be the same as the previous link, yours has around another 200 lines so hoping this will work.

Manx-Nostromo, 2023-09-03:
@smithbill17 Thank you that has worked, had to do a reboot of HA for Lux to appear. But all good. 
Not quite sure how both links appear to go to the same repo but are different?

rahulpdev, 2023-10-23:
@Manx-Nostromo none of the quirks seem to work, could you please list the exact steps taken that worked, did you remove/repair the device, restart or reboot HA, etc?

smithbill17, 2023-10-24:
I think you'll find the link I posted above works, although there are a few issues with some of the additional features.

rahulpdev, 2023-10-30:
I can't seem to get the custom quirk picked up by ZHA, it keeps reporting the quirk used by the device is `zhaquirks.tuya.ts0601_motion.MmwRadarMotion`.

I have the _TZE204_ztc6ggyl device and these are the steps I followed:

- Upload new quirk
- Remove device from HA
- Restart HA
- Add device to HA

Any pointers would be appreciated!

rahulpdev, 2023-10-30:
My mistake, I have the device `_TZE204_ztc6ggyl` for which the custom quirk appears to be here: https://github.com/zigpy/zha-device-handlers/issues/1645

smithbill17, 2023-10-30:
I don't think you need a custom quirk for _TZE204_ztc6ggyl it works 'right out of the box'.

It's _TZE204_sxm7l9xa that needs the custom quirk.

AkazaRenn, 2024-05-04:
If anyone got all functions working (occupation, illuminance, configs), could you please create a PR so we can get it merged to main? That could also help track what needs to be added to the main even if something have changed.

lanrat, 2024-11-03:
Bump.
The quirk seems to be working well enough.

PR #3736 (2025-01-20): Add Tuya motion `_TZE204_sxm7l9xa`, `_TZE204_e5m9c5hl`
## Proposed change

Adds:
- _TZE204_sxm7l9xa
- _TZE204_e5m9c5hl

Closes: #2485
Closes: #2378
Closes: #2546

## Additional information

Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/f42aa5080ee933e8cde9fc08c67a88bf886d71ff/src/devices/tuya.ts#L7959C71-L7959C87

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2025-01-20 by TheJulianJES