zigpy/zha-device-handlers #2549: [Device Support Request] _TZE204_81yrt3lo TS0601 Zigbee power meter with clamp

Srjosep, 2023-08-28:
### Problem description

The device is discovered, but with no sensors. I have all connected fine, I can see total and instantaneous power in Tuya app. Where to start?

### Solution description

Any

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Coordinator: 0>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice|AlternatePanCoordinator: 143>, manufacturer_code=43981, maximum_buffer_size=82, maximum_incoming_transfer_size=128, server_mask=11329, maximum_outgoing_transfer_size=128, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=True, *is_coordinator=True, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0400",
      "input_clusters": [
        "0x0000",
        "0x0006",
        "0x000a",
        "0x0019",
        "0x0501"
      ],
      "output_clusters": [
        "0x0001",
        "0x0020",
        "0x0500",
        "0x0502"
      ]
    }
  },
  "manufacturer": "Silicon Labs",
  "model": "EZSP",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Madrid",
    "os_name": "Linux",
    "os_version": "5.10.103-v7l+",
    "supervisor": "2023.08.1",
    "host_os": "Raspbian GNU/Linux 10 (buster)",
[... 406 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Srjosep, 2023-08-28:
I saw there is quirk data for z2m, so I created a /config/zha_quirks/ folder, create the file Tuya_TZE204_81yrt3lo.js with this data, and added  custom_quirks_path: /config/zha_quirks/ to configuration.yaml. But after reboot and repair, still no sensors. Must I delete pycache?Have no idea how to do it... 

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const valueConverter = {
divideBy10: tuya.valueConverterBasic.divideBy(10),
divideBy100: tuya.valueConverterBasic.divideBy(100),
divideBy1000: tuya.valueConverterBasic.divideBy(1000),
};

const definition = {
fingerprint: [
{
modelID: 'TS0601',
manufacturerName: '_TZE204_81yrt3lo',
},
],
model: 'TS0601_Bidirectional_Energy_Meter_with_80A_Current_Clamp',
vendor: 'TuYa',
description: 'Bidirectional Energy Meter with 80A Current Clamp',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
whiteLabel: [ tuya.whitelabel('Tuya', 'PJ-1203A', 'Bidirectional Energy Meter with 80A Current Clamp', ['_TZE204_81yrt3lo']), ],
// whiteLabel: [{vendor: 'MatSeePlus', model: 'PJ-1203A'}],
exposes: [ //only report data
e.ac_frequency(),
exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'),
exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'),
exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'),
exposes.numeric('power_direction A', ea.STATE).withDescription('Power direction A 0/1 for forward/reverse'),
exposes.numeric('power_direction B', ea.STATE).withDescription('Power direction B 0/1 for forward/reverse'),
exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),
exposes.numeric('energy_reverse_A', ea.STATE).withUnit('kWh').withDescription('Total energy A reverse'),
exposes.numeric('energy_reverse_B', ea.STATE).withUnit('kWh').withDescription('Total energy B reverse'),
exposes.numeric('update_frequency', ea.STATE).withUnit('sec').withDescription('Update frequency'),
],

meta: {
    tuyaDatapoints: [//only report data
      [111, 'ac_frequency', tuya.valueConverter.divideBy100],
      [101, 'total_power_A', tuya.valueConverter.divideBy10],
      [105, 'total_power_B', tuya.valueConverter.divideBy10],
      [115, 'total_power_AB', tuya.valueConverter.divideBy10],
      [112, 'voltage', tuya.valueConverter.divideBy10],
      [113, 'current_A', tuya.valueConverter.divideBy1000],
      [114, 'current_B', tuya.valueConverter.divideBy1000],
      [110, 'power_factor_A', tuya.valueConverter.divideBy100],
      [121, 'power_factor_B', tuya.valueConverter.divideBy100],
      [102, 'power_direction A', tuya.valueConverter.raw],
      [104, 'power_direction B', tuya.valueConverter.raw],
      [106, 'energy_forward_A', tuya.valueConverter.divideBy100],
      [108, 'energy_forward_B', tuya.valueConverter.divideBy100],
      [107, 'energy_reverse_A', tuya.valueConverter.divideBy100],
      [109, 'energy_reverse_B', tuya.valueConverter.divideBy100],
      [129, 'update_frequency' , tuya.valueConverter.raw],
    ],
},
};

module.exports = definition;

Srjosep, 2023-08-28:
I think the quirk it's not applied, but I can figure why... I followed all the steps... this is the log:

  "manufacturer": "_TZE204_81yrt3lo",
    "model": "TS0601",
    "name": "_TZE204_81yrt3lo TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",

MattWestb, 2023-08-28:
In the opening post you have signature and diagnostic from the coordinator and not the device you have described. `"device_type": "Coordinator"`.
Pleas updating them with information for the right device (device card).

Srjosep, 2023-08-28:
Sorry, wrong device..... Fixed now. ( I Hope)  :-)

Srjosep, 2023-08-29:
Tried to add custom quirk to ZHA following https://github.com/home-assistant/home-assistant.io/pull/23884/files/e03462d75c7ad37c611ab3cf4d47210f7114bc94#diff-f6bd4d42fd2565238968333a466474a25384fb4eecddc0235902e5ff5e798c8d  but still no sensors.
This means quirk isn't been applied, right? (Diagnose dump)
"manufacturer": "_TZE204_81yrt3lo",
"model": "TS0601",
"name": "_TZE204_81yrt3lo TS0601",
"quirk_applied": false,
"quirk_class": "zigpy.device.Device",

I am sure I followed all steps, creating the quirks folder and adding to yaml... 
 where could I start to look for the problem?

MattWestb, 2023-08-29:
`"quirk_class": "zigpy.device.Device"` = the device signature is ont matching your device, so need being fixed by adding it i the right device class in the quirk of if no working one making one new for your device.

Srjosep, 2023-08-29:
I created a file called TS0601_Router.py inside "customs_zha_quirks" folder. And `zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks/  `  to configuration.yaml
The data inside the file is `const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const valueConverter = {
divideBy10: tuya.valueConverterBasic.divideBy(10),
divideBy100: tuya.valueConverterBasic.divideBy(100),
};

const definition = {
fingerprint: [
{
modelID: 'TS0601',
manufacturerName: '_TZE204_81yrt3lo',
},
],
model: 'TS0601_Bidirectional_Energy_Meter_with_80A_Current_Clamp',
vendor: 'TuYa',
description: 'Bidirectional Energy Meter with 80A Current Clamp',
fromZigbee: [tuya.fz.datapoints], //, fz.ignore_tuya_set_time not working for No converter available for friendly_name with cluster manuSpecificTuya and type commandMcuSyncTime and data {payloadSize:XXX}
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
configure: tuya.configureMagicPacket,
whiteLabel: [ tuya.whitelabel('Tuya', 'PJ-1203A', 'Bidirectional Energy Meter with 80A Current Clamp', ['_TZE204_81yrt3lo']), ],
// whiteLabel: [{vendor: 'MatSeePlus', model: 'PJ-1203A'}],
exposes: [ //only report data
e.ac_frequency(),
exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'),
exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'),
exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'),
exposes.numeric('power_direction A', ea.STATE).withDescription('Power direction A 0/1 for forward/reverse'),
exposes.numeric('power_direction B', ea.STATE).withDescription('Power direction B 0/1 for forward/reverse'),
exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),
exposes.numeric('energy_reverse_A', ea.STATE).withUnit('kWh').withDescription('Total energy A reverse'),
exposes.numeric('energy_reverse_B', ea.STATE).withUnit('kWh').withDescription('Total energy B reverse'),
exposes.numeric('update_frequency', ea.STATE).withUnit('sec').withDescription('Update frequency'),
],

meta: {
    tuyaDatapoints: [//only report data
	    [111, 'ac_frequency', tuya.valueConverter.divideBy100],
        [101, 'total_power_A', tuya.valueConverter.divideBy10],
        [105, 'total_power_B', tuya.valueConverter.divideBy10],
        [115, 'total_power_AB', tuya.valueConverter.divideBy10],
		[112, 'voltage', tuya.valueConverter.divideBy10],
		[113, 'current_A', tuya.valueConverter.divideBy100],
		[114, 'current_B', tuya.valueConverter.divideBy100],
		[110, 'power_factor_A', tuya.valueConverter.divideBy100],
		[121, 'power_factor_B', tuya.valueConverter.divideBy100],
        [102, 'power_direction A', tuya.valueConverter.raw],
        [104, 'power_direction B', tuya.valueConverter.raw],
        [106, 'energy_forward_A', tuya.valueConverter.divideBy100],
        [108, 'energy_forward_B', tuya.valueConverter.divideBy100],
        [107, 'energy_reverse_A', tuya.valueConverter.divideBy100],
        [109, 'energy_reverse_B', tuya.valueConverter.divideBy100],
		[129, 'update_frequency' , tuya.valueConverter.raw],
    ],
},
};

module.exports = definition;`

This is wrong?
THANKS for your help

MattWestb, 2023-08-30:
You have posting Z2M code in the ZHA quirk and its not working.
You must doing one quirk from scratch if not can using one as temples and changing it so its  working for your device.

MattWestb, 2023-08-30:
I  think one good start  i using this quirk and caning it so its working for your device if that  is using the same DPs (tuya MCU commands).
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_din_power.py

lhanneus, 2023-08-31:
I bought the same device and had the same problem.

I found this information :  https://github.com/Koenkk/zigbee2mqtt/issues/18419#issuecomment-1657066719

Hope it can help to fix the problem.

Srjosep, 2023-09-03:
Quirk applied finally, and getting some data. Only Total power delivered and Power factor. Log: 
`"data": {
    "ieee": "**REDACTED**",
    "nwk": 56699,
    "manufacturer": "_TZE204_81yrt3lo",
    "model": "TS0601",
    "name": "_TZE204_81yrt3lo TS0601",
    "quirk_applied": true,
    "quirk_class": "TS0601_Router.TuyaPowerMeter_GPP",
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": 76,
    "rssi": -81,
    "last_seen": "2023-09-03T12:31:15",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0x0702",
            "0x0b04",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
      "manufacturer": "_TZE204_81yrt3lo",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "sensor.tze204_81yrt3lo_ts0601_power_factor",
        "name": "_TZE204_81yrt3lo TS0601"
      },
      {
        "entity_id": "sensor.tze204_81yrt3lo_ts0601_summation_delivered",
        "name": "_TZE204_81yrt3lo TS0601"
      }
    ],
    "neighbors": [
      {
        "device_type": "Coordinator",
        "rx_on_when_idle": "On",
        "relationship": "Parent",
        "extended_pan_id": "**REDACTED**",
        "ieee": "**REDACTED**",
        "nwk": "0x0000",
        "permit_joining": "Unknown",
        "depth": "0",
        "lqi": "33"
      },
[... 203 lines cut ...]
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}`

Srjosep, 2023-09-03:
This sentences make the difference. With this, quirk isn't applied, and device have no sensor. Doesn't matter if the signature is right, quirk isn't applied.
`PROFILE_ID: zgp.PROFILE_ID,
 DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,` 
With this, quirk is applied, but only sensors for total power and power factor are show.
`PROFILE_ID: 41440,
 DEVICE_TYPE: 97,` 

I have no idea what means the numbers or where find how it work,,,not much info about quirks development out there, anyway, I'll keep trying....

BiduleMac, 2023-09-03:
Hello, I've just read your post, even following your parameters I don't get the power entities. Can you post your file ts0601_din_power.py please? 

Do you confirm that once the file is placed in the custom_zha_quirks folder, I have to clear the ts0601_din_power.cpython-311.pyc cache and reboot HA? and rebuild or do I have to delete the device and reinstall it?

Srjosep, 2023-09-03:
Yes, every time I do any change I delete the device, delete _pycache_ file inside zha_quirks folder, and rebbot HA.
This is what I get after repair the device (only Power factor sensor):
[image]

And this after reboot again HA (one more sensor is discovered, "Summation delivered"). More HA reboots don't add more sensors. 
[image]
And this is the file ts0601_din_power.py for _TZE204_81yrt3lo I use.
`"""Tuya Din Power Meter."""
from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaManufClusterAttributes,
    TuyaSwitch,
)

import logging
_LOGGER = logging.getLogger(__name__)

TUYA_TOTAL_ENERGY_ATTR = 0x0201 #total energy /100  0x0211
TUYA_CURRENT_ATTR = 0x0011      #0x0212
TUYA_POWER_ATTR = 0x0006        #0x0213
TUYA_VOLTAGE_ATTR = 0x0004      #0x0214

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint16_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),      #t.int16s
        TUYA_POWER_ATTR: ("power", t.uint64_t, True),         #t.uint16_t
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
    }

    def _update_attribute(self, attrid, value):
        #super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_reported(value)
            super()._update_attribute(attrid, value)
        elif attrid == TUYA_CURRENT_ATTR:
            _LOGGER.warning("TUYA_CURRENT_ATTR (0x%04x) value: %d (0x%x)" % (attrid, value, value))
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.electrical_measurement.voltage_reported((value >> 48) & 0xffff)
            self.endpoint.electrical_measurement.current_reported((value >> 24) & 0xffffff)
            self.endpoint.electrical_measurement.power_reported(value & 0xffffff)
            super()._update_attribute(TUYA_VOLTAGE_ATTR, (value >> 48) & 0xffff)
            super()._update_attribute(TUYA_CURRENT_ATTR, (value >> 24) & 0xffffff)
            super()._update_attribute(TUYA_POWER_ATTR, value & 0xffffff)
        elif attrid == TUYA_VOLTAGE_ATTR:
            _LOGGER.warning("TUYA_VOLTAGE_ATTR (0x%04x) value: %d (0x%x)" % (attrid, value, value))
        else:
            super()._update_attribute(attrid, value)
            _LOGGER.warning("attrid: 0x%04x value: %d (0x%x)" % (attrid, value, value))

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

[... 136 lines cut ...]
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=61
                # input_clusters=[]
                # output_clusters=[21]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }`

BiduleMac, 2023-09-03:
Do you think you could send me the whole file at once? I've gone step by step, but it doesn't work.

Srjosep, 2023-09-04:
The folder must be pointed in configuration.raml, something  like 
`zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks/ `
Pointing to your quirks folder.
Can you see the quirk applied in the device info?
[image]
I read the clamp must be connected, in right position, and there must be power in the wire to get the sensors... I didn't tried with the clamp off, so, I am not sure if this is needed. I tried with Tuya app before, and only after getting all data working fine, deleted the device from tuya app and starting with ZHA.  
[ts0601_powermeter.zip](https://github.com/zigpy/zha-device-handlers/files/12509984/ts0601_powermeter.zip)

BiduleMac, 2023-09-04:
Yes !!
thanks for the file, it worked immediately. 
the first "Power factor" sensor actually appeared on the first boot, and the second "Summation delivered" after the second reboot.

for your information, the second "Summation delivered" sensor appeared with the clamp closed (currently at 0.0kWh), my blank assembly. 

I'm going back to my temporary set-up to measure a significant current.  
Many thanks for this great step forward .

But I'm going to ask you again, and maybe you won't be able to answer me, the model I have has 2 sensors, and I can only see one :-(

Do you have any idea how to activate the second sensor to help me in my search?  I didn't go through TUYA, I don't have an independent gateway, and I didn't manage to associate it in Zigbee on the application.  

thanks again, I'm desperate after 150 restarts.

Srjosep, 2023-09-04:
Mine have two sensors plug too, for use with solar or wind generators. I use only one because I have no generators i+right now. This device have much more sensors, like voltage, actual consumption, etc... I am trying to get all working, but there is info (or I can't find it) about quirks creation, so I am blindfolding, just trying things....you can see all the sensors available and the entrypoints here [https://github.com/Koenkk/zigbee2mqtt/issues/18419#issuecomment-1657066719](url)  but I don't know how translate from z2mqtt to ZHA. Any progress I will post here, if any... any suggestion, progress, you make, please let me know...

BiduleMac, 2023-09-04:
OK I'm looking 
yes, of course I'll publish here if I make any progress

maccmaic, 2023-09-05:
[ts0601_din_power_MatSee.zip](https://github.com/zigpy/zha-device-handlers/files/12526691/ts0601_din_power_MatSee.zip)
Hi, i made this file using TuyaPowerMeter_GPP. Not work 100% but is near.
[image]

BiduleMac, 2023-09-05:
👍 Great

Srjosep, 2023-09-05:
You make my day, it works like charm... just have a doubt about the active power... a current of 2.4Ah at 235volts = 566W
[image]

Srjosep, 2023-09-05:
This is the screen capture of Tuya app. It seems to need some calibration too, (is possible to modify calibration factor in the app). Last number in active power must be decimal... but you did a GREAT job. Thanks.
[image]

fernac03, 2023-09-06:
Great ,
is work for  me.
 I have two sensors, how do I show the second one?
Is the reverse for the network already working?

BiduleMac, 2023-09-07:
Concerning the two sensors, I noticed with the latest version of @maccmaic that both sensors are present (in Manage Zigbee device) and the values updated, but I don't understand how to make them appear as entities.  In fact, I'm trying to understand the structure of the file, but the display part escapes me ;-(.

fernac03, 2023-09-07:
from what I saw in the code only the attributes energy_deliver_reported and  energy_deliver_reported
are being reported channel B is commented

maccmaic, 2023-09-07:
I have tried to change delivered/recived but the integration not working. 
Sorry, i dont know the architecture and is the first time programming in python. 
[image]

If someone know why?

For channel B, I tried to report the CURRENT_TIER1_DELIVERED_ID attribute of the TuyaElectricalMeasurement class but it didn't work either.

The bad thing is that when I modify things that should not affect the rest, it stops showing all the entities.

BiduleMac, 2023-09-07:
dans mes recherches sur Github,  je ne vois aucun TS601 qui affiche 2 capteurs  .... 

ce qui est surprenant c'est  que les données des 2 capteurs remontent bien dans ZHA, on les voit varier dans le menu " gerer l'appareil Zigbee" puis "MatSeeManufClusterDinPower" et current & currentB et avec les valeurs 0x000 et 0X001 et c'est vrai également pour la puissance et le courant 
[image]
[image]

Srjosep, 2023-09-29:
Still my summation delivered is 0. Tried a lot of things but no result... do you get the right value for it? 
[image]

freesty82, 2023-09-29:
Hi everyone,
I managed, with a bit of fiddling, to get the second clamp to work.
I installed via HACS "zha_toolkit" and created an automation that every second calls the zha_toolkit.execute service which reads the second power and creates/updates a sensor.

Automation:
```
alias: aggiorno sensore
description: ""
trigger:
  - platform: time_pattern
    seconds: /1
condition: []
action:
  - service: zha_toolkit.execute
    data:
      command: attr_read
      ieee: xx:xx:xx:xx:xx:xx:xx:xx
     cluster: 0xef00
     attribute: 0x0269
      state_id: sensor.smartmeter_active_power2
      allow_create: true
mode: single
```

 However, this sensor does not have "power" device_class and it is not possible to make a customization as the sensor is created after loading the HA customizations.
So I created a template that creates a new sensor taking as an attribute the value of "sensor.smartmeter_active_power2" created earlier.

Template
```
template:
  - sensor:
      - name: "Active power Isola"
         unit_of_measurement: "W"
         device_class: power
         state_class: measurement
         state: "{{ states('sensor.smartmeter_active_power2')|float(0) / 10}}"
         availability: "{{ states('sensor.smartmeter_active_power2')|is_number }}"
         icon: mdi:transmission-tower-export
```

In this way I have an entity for the instantaneous consumption. For the total consumption I created a "Utility Meter" helper and a template and a customization for this sensor to be able to modify the unit of measurement.

Customization:

```
sensor.consumo_isola3:
    friendly_name: Consumo energetico Isola
    last_reset: '1970-01-01T00:00:00+00:00'
    device_class: energy
    state_class: measurement
    unit_of_measurement: Wh
```

Template:
```
template:
  - sensor:
      - name: "Isola totale in kWh"
        unit_of_measurement: "kWh"
        device_class: energy
        state_class: total_increasing
        state: "{{ states('sensor.consumo_isola3')|float(0) / 1000 }}"
        availability: "{{ states('sensor.consumo_isola3')|is_number }}"
        icon: mdi:transmission-tower-export
```

Thi is the result:

[image]

[image]

Srjosep, 2023-09-29:
I followed the wiring diagram included  withe device, that is, "A" connections for load clamp. I use only one clamp because I have no solar generator, I only want to know the energy delivered. The position of the clamp is checked by the Tuya official app, to get "Forward" indication for load circuit, or "energy delivered". All works fine this way with tuya official app, readings are right. 
[image]

But with the clamp at this way, in home assistant, with this quirk, summation delivered value is always 0. So, I had to reverse the position of the clamp to get the summation delivered value working. BUT on the tuya official app this way show the clamp as reverse, and the active power is too high.. I think the "energy delicered" and "energy received" reverse in the quirk... is possible?are  is this working fine for u? How are you connections? 
[image]

freesty82, 2023-09-29:
Hi Srjosep,
i had the same problem, i just made some change in the quirk.
This is the file i'm using
[ts0601_din_power_MatSee_mod.zip](https://github.com/zigpy/zha-device-handlers/files/12772309/ts0601_din_power_MatSee_mod.zip)

Anyway, I also don't have the solar generator but I use the second clamp to monitor another load

Srjosep, 2023-09-30:
> Hi Srjosep, i had the same problem, i just made some change in the quirk. This is the file i'm using [...]

Working fine now, THANKS.

[image]

freesty82, 2023-10-02:
> > Hi Srjosep, i had the same problem, i just made some change in the quirk. This is the file i'm using [ts0601_din_power_MatSee_mod.zip](https://github.com/zigpy/zha-device-handlers/files/12772309/ts0601_din_power_MatSee_mod.zip)
> > Anyway, I also don't have the solar generator but I use the second clamp to monitor another load
> 
> Hi, [freesty82](https://github.com/freesty82) Is your mod file running with the second clamp? You have generated a second endpoint, does that work for you? Thx!

Hi,
if you look at a couple of posts above I put a guide on how I solved it

here: https://github.com/zigpy/zha-device-handlers/issues/2549#issuecomment-1741305110

pepepere, 2023-10-02:
> 
> Hi, if you look at a couple of posts above I put a guide on how I solved it
> 
> here: [#2549 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2549#issuecomment-1741142070)

Thanks, I didn't see that post was yours too. So there is no way to expose the entities, right?

freesty82, 2023-10-02:
> > Hi, if you look at a couple of posts above I put a guide on how I solved it
> > here: [#2549 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2549#issuecomment-1741142070)
> 
> Thanks, I didn't see that post was yours too. So there is no way to expose the entities, right?

Not from tuya device itself , but with this solution you get a full usable energy  entity (sensor.active_power_isola in my setup)

mickaelr3003, 2023-10-18:
hi everyone,
freesty82, I am a new home assistant user, where do you put the files automation, template and customization ?
thanks

KRLH2, 2023-11-13:
I have the same issue. Device info TS0601 by _TZE204_81yrt3lo I installed the Quirk: (ts0601_powermeter.TuyaPowerMeter_GPP) but the sensor only shows the power delivered.
Please help me.
[image]
[image]

freesty82, 2023-11-13:
> I have the same issue. Device info TS0601 by _TZE204_81yrt3lo I installed the Quirk: (ts0601_powermeter.TuyaPowerMeter_GPP) but the sensor only shows the power delivered. Please help me. [image] [image]

Hi,
you can try my modified quirk posted here https://github.com/zigpy/zha-device-handlers/issues/2549#issuecomment-1741540474

freesty82, 2023-11-13:
> hi everyone, freesty82, I am a new home assistant user, where do you put the files automation, template and customization ? thanks

Hi,
Configuration:
you can Include the customize.yaml file (with the line below) in your configuration.yaml (if it isn't already there) and you can put the customization there

 `customize: !include customize.yaml`

Template:
You can put the template directly in the configuration.yaml 

Automation:
you can create the automation From the UI: choose Settings which is located in the sidebar, then click on Automations & Scenes

fdrevet, 2023-12-14:
Hi,

Thanks for your work, I managed to fetch both A and B power 👍 

I also need to fetch the power direction for A and B to know when solar panels are producing more power than consumed power (ie. power B can be "negative")

Look like we need to use the IDs in "MatSee Power Meter Attributes" quicks's section :

MATSEE_DPID_POWER_DIRECTION_ID_A_ATTR = 0x0266
MATSEE_DPID_POWER_DIRECTION_ID_B_ATTR = 0x0268

Not working when using service 🧰 ZHA Toolkit: Read Attribute with 0x0266 and 0x0268 ids (it works with 0x0265 and 0x0269 to fetch power)

Any chance to get these values through the quirks or with ZHA Toolkit/automation ?

Regards

agnus777, 2024-01-02:
Hi,

any one discovered how to show info from the second clamp?
Using the ts0601_din_power_MatSee_mod.zip I can only see the "A" clamp, data from "B" clampare not exposed.

[image]

jeverley, 2024-01-02:
Hi all,

I've created my own quirk for the device which is able to expose sensors within Home Assistant for both clamps:

[image]

Please do test and let me know of any issues - I plan to upstream this.

https://github.com/jeverley/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_smart_meter.py

markwaters, 2024-01-04:
> Please do test and let me know of any issues - I plan to upstream this once ready.

Hey @jeverley , thanks for the work you are doing with this.

My Mains sensor value is negative.
I see one of your power values is negative too.
Is that normal ?
Thanks.

[image]

jeverley, 2024-01-04:
> My Mains sensor value is negative. I see one of your power values is negative too. Is that normal ? (Sorry , I am new to all this.) Thanks.

A couple of things you'll want to check and confirm there:
- first ensure that you have both clamps wired consistently (red should go into S1, black S2)
- since the clamps are directional, you should be able to flip the clamp direction around to reverse the reading it gives - there's an arrow inside the clamp that's visible when it is open that should help with helping you put it on the correct way round.

Lastly please can you confirm if you're using the latest version of the current linked file:
https://github.com/jeverley/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_energy_meter.py

I'd have expected to see voltage and frequency reported (unless you've disabled them).

agnus777, 2024-01-04:
Hi @jeverley,

thanks for the wonderful work done, now I can see both clamps, now I'm doing a test remotely, but everything seems to work as it should!

some questions:

-  in comparison with a shelly EM (already calibrated with on the same cables I notice a significant difference, is there a way to do a fine tuning?
- as in your screenshot, there are no differentiated names of the 2 calipers, is there a way to have them diversified?
- where do I find the raw command to reverse the direction without turning the clamps?

thanks!!

Alessio

markwaters, 2024-01-04:
> A couple of things you'll want to check and confirm there:
>
>     * first ensure that you have both clamps wired consistently (red should go into S1, black S2)
>
>     * since the clamps are directional, [...]

Thanks @jeverley ,
I have checked and repositioned the errant clamp and it is now showing a positive value , I had no idea there was a direction to the clamps.
Also updated to the latest version of your file and am getting the extra entities now too.
Thank you again.
Mark.

jeverley, 2024-01-04:
> * in comparison with a shelly EM (already calibrated with on the same cables I notice a significant difference, is there a way to do a fine tuning?
> * as in your screenshot, there are no [...]

Thanks for testing @agnus777, answers below:

- You can adjust the reported values by writing to the coefficient attributes in the TuyaMatSeeManufCluster. Reading the product page it looks like values between 800 and 1200 are supported (80-120%):

[image]

It'd be good to know what values you find to be most accurate as I'm seeing similar (though can't vouch for the accuracy of the device I'm comparing with!)

- Default naming of sensors is determined by ZHA rather than by the device/quirk, you can rename them through the UI however to suit your setup.
- A command isn't being used, the value reported by the device is being applied to the power attribute (i.e. 0 =  forward, 1 means reverse i.e. negative)

jeverley, 2024-01-04:
I've made a further update this evening which does the following:

- Removes redundant voltage and frequency sensors (it's only reading and reporting these on clamp A.
  Previously I was duplicating these onto endpoint 2 so that apparent power could be calculated, this is no longer needed as I've a cleaner method of getting the value required for the calculation.
- Introduce the total power attribute being reported by the device (i.e. the sum of clamp A and B power).
  This can be used to report power consumption in a setup where you put one clamp on your inbound grid power and the second on your solar generation output.

See the results below:

[image]

Note I've renamed the power sensors in my setup through the UI, also Solar production currently 0 as it's night).

I'm also mapping through the current_summ_received values (i.e. power returned to grid) from the device however the ZHA integration doesn't currently have code in place to create sensors for these for zigbee devices, I'll pursue a separate PR to address that.
in the meantime you can read them into a sensor using the ZHA Toolkit attribute read service as people have mentioned earlier  in this thread when trying to read clamp B values.

Latest code still found here:
https://github.com/jeverley/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_energy_meter.py

enricco1, 2024-01-12:
> I've made a further update this evening which does the following:
> 
> * Removes redundant voltage and frequency sensors (it's only reading and reporting these on clamp A.
>   Previously I was duplicating these onto endpoint 2 so that apparent power could be calculated, this is no longer needed as I've a cleaner method of getting the value required for the calculation.
> * Introduce the total power attribute being reported by the device (i.e. the sum of clamp A and B power).
>   This can be used to report power consumption in a setup where you put one clamp on your inbound grid power and the second on your solar generation output.
> 
> See the results below:
> 
> <img alt="image" width="255" src="https://private-user-images.githubusercontent.com/46714706/294451164-ae63bf27-415f-4071-9387-bde510a6f2cb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwMzY3MzUsIm5iZiI6MTcwNTAzNjQzNSwicGF0aCI6Ii80NjcxNDcwNi8yOTQ0NTExNjQtYWU2M2JmMjctNDE1Zi00MDcxLTkzODctYmRlNTEwYTZmMmNiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEyVDA1MTM1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk1Y2VmNTJhZDU1YjI1YTNmOThhNDU3NmM1ZjgzZDE2ODUxZDgxNTQzMTcwNjRhYmQ4YjJmMjk5MmNkMDRiNTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.gHKcn3-Sdw8JQ7Hd_Y-WBiTnaQVsHeSe44LpN6fRdP0">
> Note I've renamed the power sensors in my setup through the UI.
> 
> I'm also mapping through the current_summ_received values (i.e. power returned to grid) from the device however the ZHA integration doesn't currently have code in place to create sensors for these for zigbee devices, I'll pursue a separate PR to address that. in the meantime you can read them into a sensor using the ZHA Toolkit attribute read service as people have mentioned earlier in this thread when trying to read clamp B values.
> 
> Latest code still found here: https://github.com/jeverley/zha-device-handlers/blob/ts0601_energy_meter/zhaquirks/tuya/ts0601_energy_meter.py

> I've made a further update this evening which does the following:
> 
> * Removes redundant voltage and frequency sensors (it's only reading and reporting these on clamp A.
>   Previously I was duplicating these onto endpoint 2 so that apparent power could be calculated, this is no longer needed as I've a cleaner method of getting the value required for the calculation.
> * Introduce the total power attribute being reported by the device (i.e. the sum of clamp A and B power).
>   This can be used to report power consumption in a setup where you put one clamp on your inbound grid power and the second on your solar generation output.
> 
> See the results below:
> 
> <img alt="image" width="255" src="https://private-user-images.githubusercontent.com/46714706/294451164-ae63bf27-415f-4071-9387-bde510a6f2cb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwMzY3MzUsIm5iZiI6MTcwNTAzNjQzNSwicGF0aCI6Ii80NjcxNDcwNi8yOTQ0NTExNjQtYWU2M2JmMjctNDE1Zi00MDcxLTkzODctYmRlNTEwYTZmMmNiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEyVDA1MTM1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk1Y2VmNTJhZDU1YjI1YTNmOThhNDU3NmM1ZjgzZDE2ODUxZDgxNTQzMTcwNjRhYmQ4YjJmMjk5MmNkMDRiNTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.gHKcn3-Sdw8JQ7Hd_Y-WBiTnaQVsHeSe44LpN6fRdP0">
> Note I've renamed the power sensors in my setup through the UI.
> 
> I'm also mapping through the current_summ_received values (i.e. power returned to grid) from the device however the ZHA integration doesn't currently have code in place to create sensors for these for zigbee devices, I'll pursue a separate PR to address that. in the meantime you can read them into a sensor using the ZHA Toolkit attribute read service as people have mentioned earlier in this thread when trying to read clamp B values.
> 
> Latest code still found here: https://github.com/jeverley/zha-device-handlers/blob/ts0601_energy_meter/zhaquirks/tuya/ts0601_energy_meter.py

How do I apply it to HA?

enricco1, 2024-01-12:
> I've made a further update this evening which does the following:
>
> * Removes redundant voltage and frequency sensors (it's only reading and reporting these on clamp A.
>   Previously I was duplicating [...]

How do I apply it to HA?

marceloajunqueira, 2024-01-16:
I have the exactly same device. 

When I add the device to ZHA, only the Power Factor sensors are added. 

[image]

I know that the Quirck is working because I can see the "Active_Power" on the 3 endpoints in the Device Info Popup.

[image]

Do I need to force anything to add the other sensors?

jeverley, 2024-01-25:
> I have the exactly same device.
> 
> When I add the device to ZHA, only the Power Factor sensors are added.
> 
> I know that the Quirck is working because I can see the "Active_Power" on the 3 endpoints in the Device Info Popup.
> 
> 
> Do I need to force anything to add the other sensors?

Apologies for the delay in responding, I've been working on the PRs to expose the summation received sensors in Home Assiustant core - that functionality is now merged https://github.com/home-assistant/core/pull/107576, so I can focus my efforts on working out remaining issues with the device quirk.

Did the issue of only seeing Power factor sensors resolve after a restart of ZHA?

FRivasFlores, 2024-01-31:
> > I have the exactly same device.
> > When I add the device to ZHA, only the Power Factor sensors are added.
> > I know that the Quirck is working because I can see the "Active_Power" on the 3 endpoints in the Device Info Popup.
> > Do I need to force anything to add the other sensors?
> 
> Apologies for the delay in responding, I've been working on the PRs to expose the summation received sensors in Home Assiustant core - that functionality is now merged [home-assistant/core#107576](https://github.com/home-assistant/core/pull/107576), so I can focus my efforts on working out remaining issues with the device quirk.
> 
> Did the issue of only seeing Power factor sensors resolve after a restart of ZHA?

jeverley thanks for this,

I have the same issue; I can read the "active_power" value from Endpoint 1, but In my case no sensors are displayed at all. I just have 1 clamp connected.

[image]

TheJulianJES, 2024-02-01:
@jeverley RE: https://github.com/home-assistant/core/pull/107576
It's not an issue in your PR, but due to how ZHA currently handles initializing attributes and creating entities, there are some "useless entities" created until ZHA has fully initialized once and ZHA restarted. Then, they stay as "ghost entities" that can be deleted.

We're currently trying to think of a proper solution for this, but could you provide the diagnostics/device signature(s) for the device(s) you need those sensors to be available on (just in case)?
If it's multiple Tuya devices that require a quirk anyway, we could assign `quirk_id` in quirks and match that in ZHA, so the `current_summ_received` only gets created for matching entities for now. (Seems like it's supported by very few devices anyway)
Just so we can "work around" the issue of creating useless entities for all power-measurement devices before the 2024.2.0 release.

TheJulianJES, 2024-02-01:
Also, is there a merged quirk in the repo that could use this entity already?

jeverley, 2024-02-01:
Hi 

> @jeverley RE: [home-assistant/core#107576](https://github.com/home-assistant/core/pull/107576) It's not an issue in your PR, but due to how ZHA currently handles initializing attributes and creating entities, there are some "useless entities" created until ZHA has fully initialized once and ZHA restarted. Then, they stay as "ghost entities" that can be deleted.
> 
> We're currently trying to think of a proper solution for this, but could you provide the diagnostics/device signature(s) for the device(s) you need those sensors to be available on (just in case)? If it's multiple Tuya devices that require a quirk anyway, we could assign `quirk_id` in quirks and match that in ZHA, so the `current_summ_received` only gets created for matching entities for now. (Seems like it's supported by very few devices anyway) Just so we can "work around" the issue of creating useless entities for all power-measurement devices before the 2024.2.0 release.

Hi @TheJulianJES here's my current draft PR https://github.com/zigpy/zha-device-handlers/pull/2961, it aims to add functionality for the different Tuya clamp Energy Meter devices I've tracked down.

We could potentially include a quirk_id in there, but it also includes the full list of device fingerprints that I'm mapping the value for.

marcogeni, 2024-02-26:
Thank you all for the work, I've just integrated this Power Meter with 2 clamps through the custom quirk.

Best regards, Marco

marceloajunqueira, 2024-02-26:
> Apologies for the delay in responding, I've been working on the PRs to expose the summation received sensors in Home Assiustant core - that functionality is now merged [...]

I did a HA restart after the installation, but In my case, all Readings appeared suddenly after 5 days. I don't know why, but it's working like a charm now.

eyalsac, 2024-03-02:
> Thank you all for the work, I've just integrated this Power Meter with 2 clamps through the custom quirk.
> 
> Best regards, Marco

Dear Marco
which quirk exsactly you used?

[36 more replies]