zigpy/zha-device-handlers #2550: [Device Support Request] ts0601_trv_me167 TRV

sebbelese, 2023-08-28:
### Problem description

I bought these thermostatic valves:
https://www.tomtop.com/p-c-e0289-1.html

The device pairs well, but without quirks, I see no entity, just LQI and RSSI.

I tried the following quirks:
[ts0601_trv_me167.py](https://github.com/jacekk015/zha_quirks/blob/main/ts0601_trv_me167.py)
https://github.com/zigpy/zha-device-handlers/files/6633888/thermostat_88teujp.zip

Some entities may appear in this case, but they are not all present and not functional

### Solution description

Is there a working quirk, or is it possible to adapt one?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Logs</summary>

```python
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_6rdj8dzm",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.8.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
[... 271 lines cut ...]
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

xxLeoxx93, 2023-08-30:
[ts0601_trv_me167.py](https://github.com/jacekk015/zha_quirks/blob/main/ts0601_trv_me167.py)
This file had an issue and is corrected now. Works fine for me!

sebbelese, 2023-08-30:
Just tried again with the new quirk, it is is not functioning, I get this:

[image]

jacekk015, 2023-09-06:
@dgrabla Every PR for quirk need to have Python test completed. I don't have time to learn this [11 quirks available] and no one else wanna to do it. There was one big PR for few quirks and died long time ago.
I will not try again.

sebbelese, 2023-09-07:
I believe that the device id is common to different clones working differently, which is why it works for some and not others.

jacekk015, 2023-10-15:
@sebbelese If still interested you need to enable debug logging
https://github.com/jacekk015/zha_quirks#in-case-of-errors-or-to-support-new-function

Put that quirk ts0601_trv_me167.py in the custom quirks folder
https://github.com/jacekk015/zha_quirks#usage

Remove TRV from HA.
Restart HA
Pair TRV like a brand new.
Wait still 10 minutes
Post here FULL log file from HA. Download logs from HA and drag/drop them into Github post window - it will be added as a file.

dgrabla, 2023-11-17:
@jacekk015 I would like to have an action to 'set target temperature'. At the moment I can only turn it on and off with the HVAC simulation. When it turns on, it remembers the last temperature set. But I want to be able to change that target temperature as well. The homeassistant UI can do it, and I can do it as well setting the state on the entity by hand. But I don't know where to add actions.

At the moment these are the actions available with this quirk file:
[image]

Is this something we can do on the quirk file?

sebbelese, 2023-11-17:
> @sebbelese If still interested you need to enable debug logging https://github.com/jacekk015/zha_quirks#in-case-of-errors-or-to-support-new-function
>
> Put that quirk ts0601_trv_me167.py in the custom [...]

When trying that on the latest quirk, my device is correctly recognized now and the thermostat works

[image]

closed 2023-11-17 by sebbelese

jacekk015, 2023-11-19:
@dgrabla It's the only way to do this.