zigpy/zha-device-handlers #2553: [Device Support Request]  TS0601 by _TZE204_rhblgy0z

lesissou, 2023-08-29:
### Problem description

Is it possible to add this tuya device to ZHA devices : https://www.amazon.fr/dp/B0C7C71DRS?psc=1&ref=ppx_yo2ov_dt_b_product_details

It is a DIN power.
It seems similar to the tuya din_power quirk

### Solution description

I would love if someone could add this device to the official quirk

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_rhblgy0z",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.8.4",
    "dev": false,
[... 293 lines cut ...]
      }
    }
  }
}
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

lesissou, 2023-08-29:
it seems the same device (same signature but different manufacturer) of issue #2549)

Srjosep, 2023-08-29:
Not sure... Is looks different [https://es.aliexpress.com/item/1005005881727281.html?spm=a2g0o.order_list.order_list_main.18.29de194dZYWYK6&gatewayAdapt=glo2esp](url)

Srjosep, 2023-08-30:
> @Srjosep Yes sure, but the signature is the same. I would like to create (try) the quirk but I don't know how can i find the replacement code. Is there a guide to do that ?

I have no idea... searching for the same. I found some code for z2m, and the manufacturer data to make it work. But not sure where to start learning to create the quirk

lesissou, 2023-08-31:
Hi,
I have created a quirk (copy of another similar quirk) but no entities found.
The quirck is well use (I can see it in the info device) but no entities and no command the switch
See below the quirk code :

"""Tuya Din Power Meter."""
from zigpy.profiles import zha, zgp
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch

TUYA_TOTAL_ENERGY_ATTR = 0x0211
TUYA_CURRENT_ATTR = 0x0212
TUYA_POWER_ATTR = 0x0213
TUYA_VOLTAGE_ATTR = 0x0214
TUYA_DIN_SWITCH_ATTR = 0x0101

SWITCH_EVENT = "switch_event"

"""Hiking Power Meter Attributes"""
HIKING_DIN_SWITCH_ATTR = 0x0110
HIKING_TOTAL_ENERGY_DELIVERED_ATTR = 0x0201
HIKING_TOTAL_ENERGY_RECEIVED_ATTR = 0x0266
HIKING_VOLTAGE_CURRENT_ATTR = 0x0006
HIKING_POWER_ATTR = 0x0267
HIKING_FREQUENCY_ATTR = 0x0269
HIKING_POWER_FACTOR_ATTR = 0x026F
HIKING_TOTAL_REACTIVE_ATTR = 0x026D
HIKING_REACTIVE_POWER_ATTR = 0x026E

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    cluster_id = ElectricalMeasurement.cluster_id

    POWER_ID = 0x050B
    VOLTAGE_ID = 0x0505
    CURRENT_ID = 0x0508
    REACTIVE_POWER_ID = 0x050E
    AC_FREQUENCY_ID = 0x0300
    TOTAL_REACTIVE_POWER_ID = 0x0305
    POWER_FACTOR_ID = 0x0510

    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603
    AC_FREQUENCY_MULTIPLIER = 0x0400
    AC_FREQUENCY_DIVISOR = 0x0401

    _CONSTANT_ATTRIBUTES = {
        AC_CURRENT_MULTIPLIER: 1,
        AC_CURRENT_DIVISOR: 1000,
        AC_FREQUENCY_MULTIPLIER: 1,
        AC_FREQUENCY_DIVISOR: 100,
    }

    def voltage_reported(self, value):
        """Voltage reported."""
        self._update_attribute(self.VOLTAGE_ID, value)
[... 126 lines cut ...]
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    HikingManufClusterDinPower,
                    TuyaElectricalMeasurement,
                    TuyaPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            16: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

Srjosep, 2023-08-31:
In the link I posted, they wrote a converter for z2m , and they posted the Dpid data, etc. I am still searching for any manual or tutorial to understand how quirks must be created,named and used... there is not much info out there....meanwhile I think I will change to z2m to test if the device works locally or not. I can send it back to the seller yet , so I must be sure it does before the return period ends....

Srjosep, 2023-08-31:
Yes sorry... I mean Tuya TS0601 TZE204_81yrt3lo, my device. It seems finally they are different. You can try asking the manufacturer...Did you find any manual or tutorial about quirk creation?

lesissou, 2023-08-31:
I find this issue, I will try this quirk : https://github.com/zigpy/zha-device-handlers/issues/1768

lesissou, 2023-08-31:
It works with this quirk

Srjosep, 2023-09-02:
Sill not working for me... no sensors and quirk isn't applied... I am missing something about how to add quirks

lesissou, 2023-09-02:
Hello @Srjosep 
You need to create the config/zha_quirks et copy your quirk into
The quirky signature must exactly the same with the original signature  of your device
To sse if the quirky is applied you can see it in the details info of your device when is added with ZHA
Hope that help you

Srjosep, 2023-09-02:
> Hello @Srjosep You need to create the config/zha_quirks et copy your quirk into The quirky signature must exactly the same with the original signature of your device To sse if the quirky is applied you can see it in the details info of your device when is added with ZHA Hope that help you

That's what I did... 
But in the details info I can read quirk isn't applied. I must have any problem with the signature, but I can't find it...

lesissou, 2023-09-02:
Could you post your device signature ?

Srjosep, 2023-09-02:
Sure, this is what I get on log:

 "manufacturer": "_TZE204_81yrt3lo",
    "model": "TS0601",
    "name": "_TZE204_81yrt3lo TS0601",
    "quirk_applied": false,    
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": 108,
    "rssi": -73,
    "last_seen": "2023-09-02T18:04:45",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {

lesissou, 2023-09-02:
And the endpoint ?

Srjosep, 2023-09-02:
This...

 "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
      "manufacturer": "_TZE204_81yrt3lo",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      },
      {
        "name": "PROXY_BASIC"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "c7f139fc3dc0c3829df8cef20d87199b",
    "area_id": null,
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "SMART_PLUG",
          "id": 81
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0001": {
                "attribute_name": "app_version",
                "value": 74
              },
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZE204_81yrt3lo"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

lesissou, 2023-09-02:
OK
I think you can used my quirck because the list of data are different (see https://github.com/Koenkk/zigbee2mqtt/issues/18419)
Do you add _TZE204_81yrt3lo TS0601 in your quirk ? See below
        MODELS_INFO: [
            ("_TZE200_lsanae15", "TS0601"),
            ("_TZE204_lsanae15", "TS0601"),
            ("_TZE204_rhblgy0z", "TS0601"),
            ("_TZE204_81yrt3lo ", "TS0601"),

Srjosep, 2023-09-02:
Is ts0601_din_power.py ? Mine looks like:

MODELS_INFO: [
            ("_TZE200_byzdayie", "TS0601"),
            ("_TZE200_ewxhg6o9", "TS0601"),
            ("_TZE204_81yrt3lo ", "TS0601"),

Maybe I've got the wrong file

lesissou, 2023-09-02:
this file is used by ZHA so you cannot used it into the custom quirk
You need to rename this file (as you want), modify and copy it into custom quirk directory
You need to modify the "HikingPowerMeter" class not the other (TuyaPowerMeter)
add this in the signature 

        MODELS_INFO: [
              ("_TZE200_bkkmqmyo", "TS0601"),
              ("_TZE204_81yrt3lo ", "TS0601"),
        ],

and

            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=61
                # input_clusters=[]
                # output_clusters=[21]
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
replace the first 5 lines by
from zigpy.profiles import zha, zgp
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

Srjosep, 2023-09-03:
Working now, but only summation delivered. I will cotinue in issue https://github.com/zigpy/zha-device-handlers/issues/2549

lesissou, 2023-09-03:
Don't use my quirk file because it isn't for your device.

Copy the code below into your file (replace the code)
I hope this will work ....

"""Tuya Din Power Meter."""
from zigpy.profiles import zha, zgp
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaManufClusterAttributes, TuyaOnOff, TuyaSwitch

TUYA_TOTAL_ENERGY_ATTR = 0x0211
TUYA_CURRENT_ATTR = 0x0212
TUYA_POWER_ATTR = 0x0213
TUYA_VOLTAGE_ATTR = 0x0214
TUYA_DIN_SWITCH_ATTR = 0x0101

SWITCH_EVENT = "switch_event"

"""Hiking Power Meter Attributes"""
HIKING_DIN_SWITCH_ATTR = 0x0110
HIKING_TOTAL_ENERGY_DELIVERED_ATTR = 0x0201
HIKING_TOTAL_ENERGY_RECEIVED_ATTR = 0x0266
HIKING_VOLTAGE_CURRENT_ATTR = 0x0006
HIKING_POWER_ATTR = 0x0267
HIKING_FREQUENCY_ATTR = 0x0269
HIKING_POWER_FACTOR_ATTR = 0x026F
HIKING_TOTAL_REACTIVE_ATTR = 0x026D
HIKING_REACTIVE_POWER_ATTR = 0x026E

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint32_t, True),
        TUYA_CURRENT_ATTR: ("current", t.int16s, True),
        TUYA_POWER_ATTR: ("power", t.uint16_t, True),
        TUYA_VOLTAGE_ATTR: ("voltage", t.uint16_t, True),
        TUYA_DIN_SWITCH_ATTR: ("switch", t.uint8_t, True),
    }

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == TUYA_TOTAL_ENERGY_ATTR:
            self.endpoint.smartenergy_metering.energy_deliver_reported(value / 100)
        elif attrid == TUYA_CURRENT_ATTR:
            self.endpoint.electrical_measurement.current_reported(value)
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.electrical_measurement.power_reported(value / 10)
        elif attrid == TUYA_VOLTAGE_ATTR:
            self.endpoint.electrical_measurement.voltage_reported(value / 10)
        elif attrid == TUYA_DIN_SWITCH_ATTR:
            self.endpoint.device.switch_bus.listener_event(
                SWITCH_EVENT, self.endpoint.endpoint_id, value
            )

class TuyaPowerMeasurement(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""
[... 208 lines cut ...]
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    HikingManufClusterDinPower,
                    TuyaElectricalMeasurement,
                    TuyaPowerMeasurement,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            16: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    TuyaOnOff,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

Srjosep, 2023-09-03:
Don't work... the quirk isn't applied.  
[image]

TheJulianJES, 2023-09-17:
Is this solved now? We should ideally keep issues open until the issue is fixed in zha-quirks/HA (without using custom quirks).

lesissou, 2023-09-17:
No it isn't solve : I reopen this issue.

Roos-AID, 2024-01-05:
Hi, I also acquired this device. Tried all above suggestions to modify existing TS0601 quirks but cannot get it to work.
No values are shown with ZHA and switch cannot be used.
The only values shown are RSSI and LQI (when enabled). 

"manufacturer": "_TZE204_rhblgy0z",
    "name": "_TZE204_rhblgy0z TS0601",

So, I did setup Rpi with ZIgbee2MQTT and voila, it works there.  Not the ideal long term solution.  This is the Zigbee2MQTT page https://www.zigbee2mqtt.io/devices/TS0601_din_3.html that works with the device.

Will try to debug a little bit more.

Roos-AID, 2024-01-16:
The unknown cluster commands are gone, but now I am stuck. 

I have tried several different quirks, but I keep getting these kind of errors in the log.

I dont know how to debug this further .  The value differs . 

> Logger: bellows.ezsp
Source: /usr/local/lib/python3.11/site-packages/bellows/ezsp/__init__.py:493
First occurred: 16:06:58 (2 occurrences)
Last logged: 16:08:23

Exception running handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/bellows/ezsp/__init__.py", line 491, in handle_callback
    handler(*args)
  File "/usr/local/lib/python3.11/site-packages/bellows/zigbee/application.py", line 616, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/usr/local/lib/python3.11/site-packages/bellows/zigbee/application.py", line 661, in _handle_frame
    self.packet_received(
  File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 1024, in packet_received
    return device.packet_received(packet)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 472, in packet_received
    endpoint.handle_message(
  File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 235, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 426, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/tuya/__init__.py", line 493, in handle_cluster_request
    zvalue = ztype(tuya_data)
             ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/types/basic.py", line 94, in __new__
    raise ValueError(
ValueError: 658088501102772301 is not an unsigned 32 bit integer

BluePenguin0, 2024-02-07:
Does anyone have a working quirk file for 'TS0601 _TZE204_81yrt3lo'?

lesissou, 2024-02-08:
See beelow, I put a working quirk file for 'TS0601 _TZE204_81yrt3lo' (for ZHA).
You need to put the file in the quircks directory and add the files into configuration.yaml :
_zha:
  custom_quirks_path: /config/<quirks_directory>_

When you add your device, sometimes it isn't working for the first time.
So you need to reboot HA and after is working well (normally).

I hope that's help you

BluePenguin0, 2024-02-08:
> See beelow, I put a working quirk file for 'TS0601 _TZE204_81yrt3lo' (for ZHA). 
Hi,
Sorry - I'm not being very bright today. There was no file attached?
I have checked the 'code' section and cannot see a file either. 
The 'ts0601_din_power.py' - does not reference the device I have (_TZE204_81yrt3lo) which is a 2 channel CT clamp.
I have tried adding my device to this file. Cleared cache, rebooted, reconfigured, removed and readded....still nothing.
So sorry for my newbieness! (only had HA for about a month - sooo much to learn).

noseshimself, 2024-06-21:
I don't want to complain but things were easier if people formatted their postings a bit better; this is python and damaging the indentation is making it quite useless.

VinzzB, 2024-12-04:
I bought an energy meter on Amazon having the following ManufacturerName: "_TZE204_rhblgy0z ". I installed one of the ZHA quirks above and initially got readings for Voltage, Current and Power.

The device also transmits the consumed and received power (kWh) in a packet with attributeId 0x201. However, the conversion was incorrect. The first 4 bytes (0.148Kwh) contains the consumed energy, the next 4 bytes are unknown values (perhaps PF and freq?). The last 4 bytes (presumably) contains the generated energy (0 Kwh in my case).
```
Received value [0, 0, 0, 148, 2, 2, 0, 4, 0, 0, 0, 0] for attribute 0x0201 (command 0x0002)
TUYA_DEBUG_ATTR (0x0201) value: 0 (0x0)
```
I managed to fix this by overriding the 'handle_cluster_request' method which will now return the whole payload for these messages. Then I was able to parse the data in the '_update_attribute' method. This is probably not the right way. I did try to understand the 'ztypes' used in the 'attributes' dictionary. But i was unable to build a custom 'ztype' method or class that was able to parse the correct bytes from these payloads. 

The device also has a built-in switch. HA receives a message when I manually control the switch on the device itself. But the same problem occurs (reading wrong bytes). Below are some payload examples when controlling the relay:

Switch OFF
```
Received value [0,18, 3, 0, 8, 50, 52, 50, 51, 48, 54, 54, 56] for attribute 0x0110 (command 0x0002)
```
Switch ON:
```
Received value [1, 18, 3, 0, 8, 50, 52, 50, 51, 48, 54, 54, 56] for attribute 0x0110 (command 0x0002)
```
The first byte contains the state of the switch . I've no idea what the other bytes represent in this payload.

The quirk below works for the majority. Readouts for PF and frequency are still missing and toggling the relay switch in HA does not work (yet).

```python
"""Tuya Din Power Meter."""
from zigpy.profiles import zha, zgp
import zigpy.types as t
from typing import Any, Optional, Union
from zigpy.zcl import BaseAttributeDefs, foundation
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, GreenPowerProxy
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaOnOff,
    TuyaSwitch,
)

import logging
_LOGGER = logging.getLogger(__name__)

TUYA_TOTAL_ENERGY_ATTR = 0x0201 #total energy /100  
TUYA_TOTAL_ENERGY_RCV_ATTR = 0x0211 #total energy /100 
TUYA_CURRENT_ATTR = 0x0011      #0x0212
TUYA_POWER_ATTR = 0x0006        #0x0213
TUYA_VOLTAGE_ATTR = 0x0004      #0x0214
TUYA_DIN_SWITCH_ATTR = 0x0110 # First byte contains switch state on _TZE204_rhblgy0z.
SWITCH_EVENT = "switch_event"

class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""
    
    #attributes = attrId: { id, type, is_manufacturer_specific } ?
    # How to extract data payload using these types?
    attributes = {
        TUYA_DIN_SWITCH_ATTR: ("switch", t.Bool, True),
        TUYA_TOTAL_ENERGY_ATTR: ("energy", t.uint64_t, True),
        TUYA_POWER_ATTR: ("power", t.uint64_t, True),         #t.uint16_t
    }
    
[... 223 lines cut ...]
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterDinPower,
                    TuyaPowerMeasurement,
                    TuyaElectricalMeasurement,
                    TuyaOnOff
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },

            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=61
                # input_clusters=[]
                # output_clusters=[21]
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
       
```

FluffyPotatoz, 2025-03-28:
> I bought an energy meter on Amazon having the following ManufacturerName: "_TZE204_rhblgy0z ". I installed one of the ZHA quirks above and initially got readings for Voltage, Current and Power.
>
> The [...]

I just wanted to let you know, i was able to load your qurik and get the features you mentioned. Hopefully an update to the quirk adds more features. At the moment im getting current, power, summation delivered and voltage. Once i have some actual power draw thru the unit, ill see if the values actually register.

One thing to note for others searching the net and stumbling on this thread, what makes this Din (the TZE204_rhblgy0z )power meter different then others (ie the _TZE200_lsanae15 ) is that the rhblgy0z is a 2 way power meter, its designed to read power consumed and power generated, thus has additional bits of data. Typically used in a solar or power generation application.
Im planning to use mine on some grid tie solar panels with micro inverters, and just track consumption/usage via home assistant.

VinzzB, 2026-09-25:
I managed to get the relay working. You can find an updated quirk in this repository: https://github.com/VinzzB/ZHA-Quirk-for-TS0601-DIN-Power/blob/main/ts0601_din_power.py