zigpy/zha-device-handlers #2558: [BUG] TRV _TZE200_cpmgn2cf not all entities available

yannik1015, 2023-08-31:
### Bug description

I have integrated the thermostat _TZE200_cpmgn2cf but it does seem like not all entities are showing up.
Reading the current temperature and setting the target temperature seem to work but the other entities I can see when looking at the source code of the quirk are not present.
Only some switch entity is showing up but I have no idea what it's functionality is and where it comes from.
Furthermore, the action can only be set to heating and not to off (Also mentioned in #2480 )
Missing entities are also mentioned here https://github.com/zigpy/zha-device-handlers/issues/1439#issuecomment-1596615813

### Steps to reproduce

1. Pair _TZE200_cpmgn2cf
2. Look at its device page

### Expected behavior

The missing entities should show up as they are set in the quirk, e.g. `MoesManufCluster`...

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

tommyguuuun, 2023-09-04:
As there was no reaction to my issue yet, i doubt there will be happening anything soon about this..

yannik1015, 2023-09-04:
> As there was no reaction to my issue yet, i doubt there will be happening anything soon about this..

I wouldn't mind me fixing it myself, but I lack some understanding as in my opinion it should at least display the entities as they are present in the code.

pepelillo, 2023-10-05:
The same here, same device, same results, I can see the parameters in the tab configuration of the device but no sensor populated into HA.
In my case the valve_status is mandatory, so please can someone fix it?
Thanks

jacekk015, 2023-10-15:
As for issue:
- no logs
- no signature
- no screenshot
- nothing.

Use quirk file:
ts0601_trv_moes.py
from my repo:
https://github.com/jacekk015/zha_quirks

tommyguuuun, 2023-10-15:
> As for issue:
> 
> * no logs
> * no signature
> * no screenshot
> * nothing.
> 
> Use quirk file: ts0601_trv_moes.py from my repo: https://github.com/jacekk015/zha_quirks

Should this add the „off“ state? I use your custom quirk and all i got is still „heating“ but not „off“. 

I also meantioned it here, with screenshots etc. : #2480

jacekk015, 2023-10-16:
That TRV doesn't have an OFF state. It just heats.
There's sensor which says that it is Idle/Heating
How do you turn off that TRV?

pr0l4m3, 2023-10-20:
> That TRV doesn't have an OFF state. It just heats. There's sensor which says that it is Idle/Heating How do you turn off that TRV?

on zigbee2mqtt are off not only heat.

jacekk015, 2023-10-20:
And how TRV behaves when you turn off in Z2M?

pr0l4m3, 2023-10-20:
> And how TRV behaves when you turn off in Z2M?

status was "off" and was fully closed. Today i migrate to ZHA and was suprise when "off" doesnt exist.

pr0l4m3, 2023-10-20:
[https://i.ibb.co/hmWQVjy/380208566-214637538308330-6483871896189501202-n.png](url)

pr0l4m3, 2023-10-20:
[image]

pr0l4m3, 2023-10-21:
[image]
screen from z2m

jacekk015, 2023-10-22:
Those TRV doesn't have build-in Off mode.
Z2M faked it by using/forcing "system mode" to "Force valve"
heat -> open
auto -> normal
off -> close
So it's all a fake using those code lines below:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L2542C29-L2542C58
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/legacy.ts#L7507

From the quirk code I see that it was just exposed on the Device Attributes level.
We can fake it also... need to do some changes and volunteer for tests.

jacekk015, 2023-10-22:
Quirk was modified.
Now you have fake system mode like Z2M: Auto, Off, Heat
Off -> Off
Heat -> Force to heat
Auto -> Schedule heating from TRV

https://github.com/jacekk015/zha_quirks
ts0601_trv_moes.py

yannik1015, 2023-10-23:
@jacekk015 Just checked and with your quirks it seems to work like a charm as it shows the entities. Why don't you open a merge request here such that your quirks and changes get added to the Zigpy project. Doesn't seem useful to develop them and write about them here without adding them to the project.

jacekk015, 2023-10-23:
@yannik1015 long story short. Tried one time.

wormvortex, 2023-11-15:
> Quirk was modified. Now you have fake system mode like Z2M: Auto, Off, Heat Off -> Off Heat -> Force to heat Auto -> Schedule heating from TRV
> 
> https://github.com/jacekk015/zha_quirks ts0601_trv_moes.py
> 
> In case of errors etc. HA full logs on DEBUG level needed. According to: https://github.com/jacekk015/zha_quirks#in-case-of-errors-or-to-support-new-function

I've tried using this quirk but I don't have any of the fake system modes. Never used a quirk before and not sure how to diagnose what is wrong. I've placed the quirk in the correct folder and referenced it also in my config. Is there anything else I need to do or check?

jacekk015, 2023-11-16:
> not sure how to diagnose what is wrong

How do you know that it doesn't work then???
You didn't provide any info that I could help you.
No picture, no error, nothing.
HA has logs under Settings -> System -> Logs

wormvortex, 2023-11-16:
Because I don’t have any of the fake system modes showing. Was only showing the heat option. Will attach logs when I get home and chance to 👌

jacekk015, 2023-11-16:
Don't understand the meaning of sentence: " fake system modes"
What do you expect to see??
Those TRV's doesn't have any other option than Heat. It won't Cool for sure.

MokryPatrik, 2024-01-02:
Hello, I tried to install your quirk and I have "heat" only. Based on your comment https://github.com/zigpy/zha-device-handlers/issues/2558#issuecomment-1774188855 I assume that it should also contain "off" and "auto" modes.

[image]

tommyguuuun, 2024-01-02:
> Don't understand the meaning of sentence: " fake system modes" What do you expect to see?? Those TRV's doesn't have any other option than Heat. It won't Cool for sure.

Fact is, that my TRVs show „OFF“ on their screen, when i select it in z2mqtt. So i highly doubt there is no such option, when the led screen has a own status message for it.

robbionline, 2024-03-07:
Any news? How can I help? Do you need logs?

jacekk015, 2024-03-09:
@robbionline Any news about what??
People here probably expect that I should buy every TRV and make it work.
Since November 15 no one posted DEBUG level logs here, where they operate the TRV device and make it OFF.

closed 2024-09-12 by github-actions[bot]