zigpy/zha-device-handlers #2561: [Device Support Request] Bosch Twinguard

Smiley098, 2023-09-02:
### Problem description

Hello,

Unfortunately, the message Installation code too short appears when I want to add the Twinguard (Zha-permit).
The twinguard cannot be added without a code.

https://github.com/home-assistant/core/issues/44805
https://www.zigbee2mqtt.io/devices/8750001213.html

### Solution description

Allow all installation code lengths?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Hedda, 2023-09-04:
Which exact Zigbee Coordinator and firmware do you use? Not all radio adapters/dongles support “Install Code” or “QR Code”:

https://www.home-assistant.io/integrations/zha#limitations

"Support for commissioning Zigbee 3.0 devices via “Install Code” or “QR Code” via the ‘zha.permit’ service has so far only been implemented for ‘ezsp’ (Silicon Labs EmberZNet) or ‘znp’ (Texas Instruments) radio type in ZHA. Other radio types are missing support in their respective [radio libraries for zigpy](https://github.com/zigpy/) or manufacturer’s firmware commands/APIs."

deCONZ serial protocol for ConBee/RaspBee Zigbee Coordinator adapters does not yet support joining/pairing via install code or qr code, and that is due to the manufacturer (dresden-elektronik) not fully documenting their serial protocol for it. See:

https://github.com/zigpy/zigpy-deconz/issues/214

and

https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20

So if you want to use commissioning via “Install Code” today then you at least need a new Silicon Labs or Texas Instruments) radio.

PS: This is the same for Zigbee2MQTT / zigbee-herdsman when it comes to and deCONZ serial protocol for ConBee/RaspBee. See:

https://github.com/Koenkk/zigbee2mqtt/issues/17492

Hedda, 2023-09-04:
> I can not find any Zigbee info about it. Wondering if Bosch has posted specifications for it and which Zigbee profile(s) it supports?

Hmm, according to Bosch's Twinguard help it uses "**_ZigBee HA 1.2 protocol and proprietary, manufacturer-specific protocol extensions_**" which I would assume means that it uses their own non-standard custom install code for it?

https://www.bosch-smarthome.com/uk/en/service/help/product-help/twinguard-help/

Check out this related Z2M discussion about Bosch Twinguard and Bosch Outdoor Siren using similar proprietary install code / key:

https://github.com/Koenkk/zigbee2mqtt/issues/13520

https://github.com/Koenkk/zigbee2mqtt/issues/14355

Smiley098, 2023-09-04:
Hello,

thanks for the fast research.

I added logs and a photo to my first post.

Seems as if the twinguards use a shorter installation code (16 bytes instead of 18).

Looks like the Twingurad isn't going to ZHA?

Hedda, 2023-09-05:
> **Seems as if the twinguards use a shorter installation code (16 bytes instead of 18).**
> 
> Looks like the Twingurad isn't going to ZHA?

Reading the comments by [Adminiuga](https://github.com/Adminiuga) and @atmurray in https://github.com/home-assistant/core/issues/44805 it sounds as if the radio libraries could support shorter install codes if both ZHA and zigpy library (and its radio libraries) was patched to also accept 16 bytes size when using the `install_code` command?

@atmurray even mentioned in that https://github.com/home-assistant/core/issues/44805 that he could "_prepare a pull request to patch ZHA's `helpers.py` so that it aligns to the code length check in zigpy will it get accepted_".

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/core/helpers.py

Suggest that you at the very least submit a new issue as a feature request about "support for shorter install codes" to the zigpy library/project since that is what Home Assistant's ZHA integration depends on:

https://github.com/zigpy/zigpy/issues

It sounds like ZHA Device Handlers (quirks) for such type of devices should be possible if the zigpy library/project already had support for shorter install codes in `convert_install_code` or? As in also accepting install_code lengths of 16 bytes?

https://github.com/search?q=repo%3Azigpy%2Fzigpy%20zigpy%20convert_install_code&type=code

At least now you have a clear use-case for such a request if now more than one device in Bosch Smart Home series requires this. That is, while the Zigbee 3,0 standard only uses 18 bytes install code there clearly are some other commonly available devices that use Zigbee Home Automation (ZHA) profile with shorter install codes as link key (e.g. 16 bytes for Bosch Smart Home devices and Jetlun RD77715 -> https://fccid.io/X5QRD77715/User-Manual/Users-Manual-1592194 )

PS: Again for reference, please note that install codes are currently only supported in zigpy radio libraries for newer Zigbee Coordinator adapters with new radio SoC chips from Silicon Labs and Texas Instruments, (so at this point there is no point in even testing on other Zigbee Coordinator adapters like ConBee/RaspBee, XBee, and ZiGate).

puddly, 2023-09-11:
Can you post the full QR code? Unfortunately, the install code doesn't appear valid even when you remove the length restriction.

Smiley098, 2023-09-11:
Hello,

QrCode:
RB01SG0D836591B3CC0010000000000000000000000D6F0017E0870CDLK999F98A7DFBCA6DD3955823AD9089631

puddly, 2023-09-11:
Is this the entire QR code, nothing is missing?

Compared to the one in the Z2M issue, yours is shorter by two octets:

```python
# Z2M
RB01SG0D83101826480040000000000000000000003C84FFFEEDA00BDLKCE5D5C1E3D59D1440ABFE85251205C36ADE4
# Yours
RB01SG0D836591B3CC0010000000000000000000000D6F0017E0870CDLK999F98A7DFBCA6DD3955823AD9089631
```

puddly, 2023-09-11:
Just to be 100% sure it's not an app problem, could you post a photo of the QR code as well? The first example I posted validates properly with existing but the second one from your QR code does not. There's no link key in it even if you try every combination, which is unusual.

Smiley098, 2023-09-11:
Maybe mine is missing the crc?

key - a byte array representing DSLK with length 16 or IC with length 18

https://docs.bosch-iot-suite.com/gateway-software/API/modules/com.prosyst.mbs.zigbee.da.items.api/index.html?com/prosyst/mbs/services/zigbee/da/items/ZigBeeDiscovery.html

Smiley098, 2023-09-11:
[image]

puddly, 2023-09-11:
Interesting. Your device's model number is `8750001215`, while the linked device's model number is `8750001213`. Perhaps yours is a newer revision?

The "Device Local Key" (`4SLYCAGQXWMUFLKNC27BCHJ5JJ`) doesn't correspond to any data within the QR code. The value itself looks to be base32-encoded (without padding), and decodes to the following key:

```
e4:97:81:00:d0:bd:99:42:ad:4d:16:be:11:1d:3d:4a
```

The supposed key within the QR code (without CRC) is:

```
99:9f:98:a7:df:bc:a6:dd:39:55:82:3a:d9:08:96:31
```

There's currently no way to directly provide install codes but you can easily patch them into your respective radio library (e.g. hardcode the key to be `t.KeyData.convert("99:9f:...")` in https://github.com/zigpy/zigpy-znp/blob/e60c556197f640b07e57aa3f655cdcc220f2e809/zigpy_znp/zigbee/application.py#L351 and supply any valid install code).

Hedda, 2023-09-12:
Guessing that a developer would have to buy the older "Bosch Smart Home Controller" (BSHC-1, which is more commonly available and costs a lot more than their newer gateway/bridge/hub), and use Zigbee sniffer with Wireshark to sniff and dump Zigbee traffic when joining/paring this device to the official Bosch Smart Home Controller"?

https://www.bosch-smarthome.com/uk/en/service/help/product-help/smart-home-controller-help/

https://www.amazon.com/Bosch-Smart-Home-Controller-Zentraleinheit/dp/B01N7AE32J

The newer "Bosch Smart Home Controller II" (BSHC-2) should not be needed for the Bosch Twinguard device?

https://www.bosch-smarthome.com/uk/en/products/devices/smart-home-controller

According to their compatibility page it looks like both old BSHC-1 and new BSHC-2 should be compatible with Bosch Twinguard:

https://www.bosch-smarthome.com/uk/en/service/help/compatibility-controller/

https://www.bosch-smarthome.com/uk/media/en/content/03_produkte/01_geraete/27_kompatibilitaet_controller_ii/kompatibilitaets_listen/pr22202_bosch_smart_home_controller_ii_kompatibilitat_tabelle_rz1-6.pdf

It is the newer "Bosch Smart Home Controller II" (BSHC-2) that is not compatible with old devices unless add their radio stick:

https://www.bosch-smarthome.com/uk/en/products/control-elements-and-accessories/radio-stick/

[image]

[image]

Smiley098, 2023-09-13:
I was able to recreate the crc.
The bits are reversed

999F-98A7-DFBC-A6DD-3955-823A-D908-9631-9F51

https://crccalc.com/?crc=999F-98A7-DFBC-A6DD-3955-823A-D908-9631&method=CRC-16/X-25&datatype=hex&outtype=0

The installation code is accepted. The twinguard is also found, but cannot be added.

I'll try again later and look at the logs.

Smiley098, 2023-09-14:
Is the installation code still the problem?

<details><summary>Log</summary>
<p>

Device 0xe240 (00:0d:6f:00:17:e0:87:0c) joined the network
[0xe240] Requesting 'Node Descriptor'
Device 0xe240 (00:0d:6f:00:17:e0:87:0c) joined the network
[0xe240] Requesting 'Node Descriptor'
Device 0x485c (00:0d:6f:00:17:e0:87:0c) joined the network
[0x485c] Requesting 'Node Descriptor'
Device 0xe240 (00:0d:6f:00:17:e0:87:0c) joined the network
[0xe240] Requesting 'Node Descriptor'
Device 0x049b (00:0d:6f:00:17:e0:87:0c) joined the network
[0x049b] Requesting 'Node Descriptor'
Device 0x049b (00:0d:6f:00:17:e0:87:0c) joined the network
[0x049b] Requesting 'Node Descriptor'
Device 0xe240 (00:0d:6f:00:17:e0:87:0c) joined the network
[0xe240] Requesting 'Node Descriptor'
Device 0x049b (00:0d:6f:00:17:e0:87:0c) joined the network
[0x049b] Requesting 'Node Descriptor'
Device 0x049b (00:0d:6f:00:17:e0:87:0c) joined the network
[0x049b] Requesting 'Node Descriptor'
Device 0x647a (00:0d:6f:00:17:e0:87:0c) joined the network
[0x647a] Requesting 'Node Descriptor'
Device 0x049b (00:0d:6f:00:17:e0:87:0c) joined the network
[0x049b] Requesting 'Node Descriptor'
Device 0xb420 (00:0d:6f:00:17:e0:87:0c) joined the network
[0xb420] Requesting 'Node Descriptor'
Device 0xb420 (00:0d:6f:00:17:e0:87:0c) joined the network
[0xb420] Requesting 'Node Descriptor'
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0xb420 (00:0d:6f:00:17:e0:87:0c) joined the network
[0xb420] Requesting 'Node Descriptor'
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0x647a (00:0d:6f:00:17:e0:87:0c) joined the network
[0x647a] Requesting 'Node Descriptor'
Device 0xe240 (00:0d:6f:00:17:e0:87:0c) joined the network
[0xe240] Requesting 'Node Descriptor'
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0xe240 (00:0d:6f:00:17:e0:87:0c) joined the network
[0xe240] Requesting 'Node Descriptor'
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
Device 0xba57 (34:25:b4:ff:fe:3f:e6:7f) joined the network
[0xe240] Requesting 'Node Descriptor'
[0xe240] Requesting 'Node Descriptor'
[0xe240] Requesting 'Node Descriptor'

</p>
</details> 

<details><summary>Log</summary>
<p>

Logger: zigpy.application
Source: runner.py:186 
First occurred: 10:02:30 (1 occurrences) 
Last logged: 10:02:30

Received relays from an unknown device: 0xe240

</p>
</details>

puddly, 2023-09-14:
@Smiley098 Yes, I think so. From your log, the device joins the network multiple times (indicative of it leaving and trying again) and is unresponsive.

From the perspective of a Zigbee device, there are only link keys. They don't know or care about install codes. For that reason, manually appending a CRC to the `999F...` code probably won't work for two reasons:

1. Install codes are in the form: `[code][crc]`, where `code` is then *hashed* with AES-MMO to derive a link key. You can turn anything into an install code if you append a valid CRC, but that doesn't mean that `aes_mmo_hash(code)` is going to give you the link key.
2. Bosch, on the other hand, seems to directly specify a *link key* for your device, not an install code. There's no way to convert this into an install code.

To try these out, you will need to try both keys directly by patching the radio library as I mentioned above. Using them as install codes probably won't work.

Hedda, 2023-09-14:
> Interesting. Your device's model number is `8750001215`, while the linked device's model number is `8750001213`. Perhaps yours is a newer revision?

FYI, screenshot from [safakaltun](https://github.com/safakaltun) of his Bosch Twingard also has device model number  `8750001215` and he have also posted several dumps when sniffing trying to join it as well his analysis of them -> https://github.com/Koenkk/zigbee2mqtt/issues/14355

[datenzar](https://github.com/datenzar) also posted to https://github.com/Koenkk/zigbee2mqtt/issues/13520 that he figured out hw to get the link keys from sniffed packages following stm32mcu docs (but that is way over my head)

_Originally posted by @datenzar in https://github.com/Koenkk/zigbee2mqtt/issues/13520#issuecomment-1501220624_
> How did I get above keys? It's mentioned in the docs that the key exchange will take place based on a derived link key. The derivation is actually just the MMO (Matyas-Meyer-Oseas) hash of the install code + crc.
> See here for a nice illustration: https://wiki.st.com/stm32mcu/nsfr_img_auth.php/thumb/2/24/Conectivity_InstallCode.png/700px-Conectivity_InstallCode.png    

So, have Zigbee2MQTT been able to fully work around these Bosch quirk(s) in zigbee-herdsman and zigbee-herdsman-converters? 

https://github.com/Koenkk/zigbee2mqtt/issues/14355

https://github.com/Koenkk/zigbee2mqtt/issues/13520

https://github.com/Koenkk/zigbee-herdsman/pull/623

https://github.com/Koenkk/zigbee-herdsman/pull/630

https://github.com/Koenkk/zigbee-herdsman-converters/pull/5176

Btw, are these same quirks in more Bosch Smart Home series products than the Bosch Twinguard and the Bosch Outdoor Siren?

Smiley098, 2023-09-23:
I'm too stupid to find the file. :(

I'll wait and see if someone else can do it. 

I was able to add the Bosch adapter plugs using the installation code.
https://www.bosch-smarthome.com/de/de/produkte/geraete/zwischenstecker-kompakt/

Hedda, 2023-10-26:
Not sure if related this issue(?) but FYI; noticed @promasu has submitted PR to "**_Add support for Bosch QR-codes for zha.permit_**"

https://github.com/home-assistant/core/pull/102427

https://github.com/home-assistant/home-assistant.io/pull/29459

blatthew, 2023-11-17:
@Smiley098 Have you been able to add the Twinguard through ZHA in the meantime?

I have exactly the same problem as you describe here..

Smiley098, 2023-11-20:
@blatthew 
Unfortunately I still haven't managed it

promasu, 2023-12-20:
I'm currently implementing a quirk for the Bosch Radiator Thermostat 2. If I have more time I can take a look at Twinguard too

4s1, 2023-12-20:
Can anyone provide a step by step documentation how to zniff bosch' smart home controller zigbee traffic? I've got an ember stick and am able to capture the encrypted traffic of the controller and a few brand new gen II smart home bosch devices (I've just bought). But I've no cue how to retrieve the network key from the captured traffic.

puddly, 2023-12-22:
Can you upload the traffic PCAP and a photo of the QR code on your devices?

If the network key is encrypted with the default link key `ZigBeeAlliance09`, it will be automatically decrypted. If not, make sure you have the key loaded:

[image]
[image]

tsaG1337, 2023-12-23:
I also just bought the Twinguards and realized they use a different scheme to the ordinary Bosch smoke detector 2 (missing the installation code)...

I do have HA on Debian with a Skyconnect stick on a Debian VM. Any way I could help?

QR code would be the following: RB01SG0D836591B3CC0010000000000000000000000D6F0019105047DLKE4F7A8C6A8C54F2281640ABC518B8A1B where 000D6F0019105047 is the Source IEEE. The local key is mentioned on the device but not in the QR code (so I assume its not needed (?)).

idstein, 2024-01-07:
One quick update on the Bosch Twinguard. Bosch QR Code seems to contains straight a link key that is commonly derived from the installation code.

I have successfully paired my Twinguard via ZHA using some small changes on the QR code handling. Zigpy changes are not required. Yet, not sure if the reported values (except maybe humidity and temperature) are correct.

[image]

tsaG1337, 2024-01-07:
Can you elaborate what these changes were and how you implemented it? Or did you alter the Sourcecode on ZHA?

idstein, 2024-01-07:
See https://github.com/home-assistant/core/pull/107460 for further details

Hedda, 2024-01-07:
> See [home-assistant/core#107460](https://github.com/home-assistant/core/pull/107460) for further details

Nice! Is Bosch Twinguard their only device that needs such a unique solution or could it also be applies to other Bosch devices?

https://github.com/zigpy/zha-device-handlers/issues?q=is%3Aissue+Bosch

idstein, 2024-01-07:
> Nice! Is Bosch Twinguard their only device that needs such a unique solution or could it also be applies to other Bosch devices?
So far yes it seems that it is their only devices. For Bosch Outdoor Siren I could not verify yet, as I do not have it physically to test. If one could count the characters after `DLK` in the QR code, we could validate it. 36 hex characters translates to 18 bytes is the installation code (16 bytes) + crc (2 bytes). For 32 hex characters (like Bosch Twinguard) it represents the 16 bytes link key which is the AES MMO hash.

Zigpy internal does not directly passes along the installation code to the coordinator, but computes the AES MMO hash from the 18 bytes installation code + crc.

Mar1usW3, 2024-02-03:
does anyone get this running with Skyconnect?

4s1, 2024-02-07:
> Can anyone provide a step by step documentation how to zniff bosch' smart home controller zigbee traffic? I've got an ember stick and am able to capture the encrypted traffic of the controller and a [...]

Okay, figured it out by myself

1.  used an install code from a new device
2. calculated link key from it (though `zigpy.util.convert_install_code(...)`)
3. captured zigbee traffic as described [here](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#_3-sniffing-traffic)
4. inserted link key in wireshark zigbee protocol preferences
5. loaded a capture file (taken while joined the device with the given install code)
6. filtered for `_ws.col.info == "Transport Key"` and voila, there it is: Key Type: Standard Network Key (0x01).

Added that key to wiresharks zigbee key list too, now I can see all traffic decrypted.

4s1, 2024-02-08:
> With the update of today 2024.02 I could use the QR-Code. After 8 Hours the device did not show any status changes nor sensor output, right now it's unavailable.

What kind of coordinator do you use? It is known to work with Texas Instrument based adapters (e.g. CC2652), but not with Silicon Image (e.g. Skyconnect, HA yellow - EFR32 based)

idstein, 2024-02-08:
Pairing itself works (manual cluster querying works), but it requires further quirks (manual heartbeat, pairing complete, periodic reporting)

I’ll try to push my changes soon to zha-quirks

firstusing, 2024-02-08:
> Pairing itself works (manual cluster querying works), but it requires further quirks (manual heartbeat, pairing complete, periodic reporting)
> 
> 
> 
> I’ll try to push my changes soon to zha-quirks

Thank you for your work!

KorbinianPei, 2024-03-03:
Hey everybody,
is there still an opportunity to connect the Twinguard via Zigbee in Home Asisstant? Which components are required for this setup than?
I would like to buy a smoke detektor which can also measure some other nice things like humidity and air purity. I found the twinguard and it seems like this device would be perfect for this. The problem is the integration into home assistant when the smart home controller should not bei used and the twinguard should run as stand alone device.

Thanks for helping

Mar1usW3, 2024-03-24:
i just wonder if this a firmware issue or a quirk issue

firstusing, 2024-03-29:
@idstein have you a custom quirk to use? Or do you think it can be enabled for ZHA?
I do not want to sell the devices but my wife now comes after me :)

KorbinianPei, 2024-03-29:
> @id
[image]
stein have you a custom quirk to use? Or do you think it can be enabled for ZHA? I do not want to sell the devices but my wife now comes after me :)

Hey @firstusing,

I tried the Twinguard with the Zigbee2MQTT Integration an this works fine. So if you would like to use it you could also try this Integration. But therefore you have to switch all your Devices to zigbee2mqtt. This could also be a lot of work.

idstein, 2024-03-29:
I fixed the readings for the sensors, but struggling to get a continuous reporting. It looks like the values are not continuously send.

KorbinianPei, 2024-03-29:
> I fixed the readings for the sensors, but struggling to get a continuous reporting. It looks like the values are not continuously send.
> 
> Carbon Monoxide is actually not what the BME sensor provides. There is only a Indexed Air Quality available to indicate roughly VOC.

The Screenshot is what the zigbee2mqtt Integration provides.

https://www.zigbee2mqtt.io/devices/8750001213.html

I also thought that I red, that the twinguard "only" measures voc instead of co2. But at the end of the day it dose not matter what the Integration shows as long as you know what it means.
Are you sure that the ppm value is voc?
But except this one failure the values come continious into home Assistant. Until now I tested the twinguard for 3 days.

Also the siren can be switched in home assistant 

[image]
[image]

Mar1usW3, 2024-03-31:
for me z2m with a skyconnect is also not working

firstusing, 2024-04-23:
> > @id
> > ![Screenshot_20240329_065640_Home Assistant](https://private-user-images.githubusercontent.com/114978654/317966835-2d082964-908f-4644-b07f-355cef4b0c04.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5MDM3MTksIm5iZiI6MTcxMzkwMzQxOSwicGF0aCI6Ii8xMTQ5Nzg2NTQvMzE3OTY2ODM1LTJkMDgyOTY0LTkwOGYtNDY0NC1iMDdmLTM1NWNlZjRiMGMwNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyM1QyMDE2NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MGJiYjAyY2MxOTdlZmUzYzdkOTM0NjI4NDAwZWIxMDUxOTM3OGNlMTM2NWUzNDFjOTY2YmJmMTgwNDljN2RjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.R3fQ3XQ8JKzCinSc4sqR98DPqiakQtbcBpf2dn3N05U)
> > stein have you a custom quirk to use? Or do you think it can be enabled for ZHA? I do not want to sell the devices but my wife now comes after me :)
> 
> Hey @firstusing,
> 
> I tried the Twinguard with the Zigbee2MQTT Integration an this works fine. So if you would like to use it you could also try this Integration. But therefore you have to switch all your Devices to zigbee2mqtt. This could also be a lot of work.

I have to say i do not see how it works, i changed from ZHA to Z2M completly, the paring does work great but after that.... no measurments at all.

KorbinianPei, 2024-04-23:
> I have to say i do not see how it works, i changed from ZHA to Z2M completly, the paring does work great but after that.... no measurments at all.

It depends on the z2m version. There are some issues with the new (1.36.1-1) version. You have to go back to the 1.36.0-1 than it should work.

paundl, 2024-07-01:
hi @idstein and all others here  
i'm trying to get the twinguard working in my network
* ZHA
* Skyconnect
* HA on latest version
* brand new Bosch Twinguard

what i understood in this thread is that with @idstein 's merged pull request binding should work via the QR code  
but i'm not able to get my twinguard running  
  
what i did:
* read the qr code
* got to my homassistant instance and call the service zha.permit
* enter the qr code
* set duration to max
  
then twinguard gets conected to my zigbee network but...
* the name is BOSCH ST Champion
* i see no values and no updates
  
id there somewhere a step by step guide how to add the twinguard via ZHA to homeassistant?  
  
thank you for your input!  
  
[image]

MoFa85, 2024-07-03:
Same for me 😕

paundl, 2024-07-12:
@idstein  @KorbinianPei @Mar1usW3 @firstusing 
any idea?
also with the lates HA update, so change in pairing behavior

thank you so much!

paundl, 2024-07-18:
i know, nobody gets paid here and we all have to respond in our free time but still i'm a little bit sad to get no response, i'll send it back then and use another solution.

KorbinianPei, 2024-07-19:
> i know, nobody gets paid here and we all have to respond in our free time but still i'm a little bit sad to get no response, i'll send it back then and use another solution.

Hello,
to be true I do not know what's the problem know.
It is known, that the integration of the Bosch Twinguard into Home Assistant using the ZHA integration does not function for know (that's my knowledge for today, please correct me when I am false).
It is also known, that the integration with the Zigbee2mqtt integration can lead to some failures when you use the false Z2m version and it you use skyconnect (there are some different posts on github depending this topic)
And I can tell you that if you integrate the twinguard via Z2M (newest version) integration and you use a sonoff dongle that
All the stuff works fine.
So, I hope I could help you with this summery. 
Greetings to all 
[image]
[image]

sarangnemo, 2024-08-09:
So I dove in and took a stab at an initial ZHA quirk myself, see the above PR. It's inspired by the [Zigbee2MQTT implementation](https://github.com/Koenkk/zigbee-herdsman-converters/blob/6e96d847ddf633725108c08ddfcd6088a9fe3a71/src/devices/bosch.ts#L595).

If you want to test it yourself, you can drop _twinguard.py_ from my PR into your custom quirks directory.

It enables updating of temperature, humidity, illuminance, and raw VOC values. Smoke detection and alarm functions are not yet implemented. (I won't have time to work on that soon – if anybody else wants to have a crack, please feel free! They're implemented in the Z2M code, so it should be 'easy' to convert it.)

In my experience, the measurement values begin reporting some minutes after pairing. I also needed to restart HA to see the new sensors. Note that the default _Carbon Monoxide_ and _Pressure_ clusters also show up but won't update automatically. I'm still not clear if there's even a Carbon Monoxide sensor or derived calculation in the device...

I believe that what Z2M calls _pressure_ from the 0xE002 data cluster is actually something else. For me, it usually sits stable at its max value of 1000, but occasionally dips to the 900s or 800s for a few minutes. It doesn't seem directly connected to VOC or anything else that's obvious.

My PR doesn't at all address any pairing issues. In my case, I have a Conbee II stick, but the Twinguard never pairs directly with it, instead going through my Zigbee light switches (after doing the _ZHA permit_ procedure with the QR code, as described [above](https://github.com/zigpy/zha-device-handlers/issues/2561#issuecomment-2201141493). Somehow this usually results in losing pairing with one or more of my other light switches, so I often need to repair those after getting the Twinguard paired.

Home Assistant then reports successful pairing for me, while the Twinguard keeps blinking and eventually glows red (indicating that pairing failed on its side). But after holding down the button to get it back in pairing mode, it then quickly finds the network again and glows green. I believe the measurements might not get updated if you don't get the Twinguard to glow green...

Hope this helps out some people here!

Hedda, 2025-08-15:
> So I dove in and took a stab at an initial ZHA quirk myself, see the above PR. It's inspired by the [Zigbee2MQTT [...]

Anyone using/testing the initial Bosch Twinguard quirk (ZHA device handler ) from pull request that @sarangnemo is working on?

* https://github.com/zigpy/zha-device-handlers/pull/3293

FYI, you can find some more instructions here in this community how to test experimental qurks that have not yet been merged:

* https://community.home-assistant.io/t/zigbee-guide-how-to-add-setup-local-custom-zha-device-handlers-also-known-as-quirks-in-the-zha-integration/683473

sarangnemo, 2026-01-07:
@pefmeister, yes that's right, often the device doesn't start reporting for some time after pairing. Indeed i had not implemented smoke detection notifications or alarm triggers, so that won't work until somebody translates the z2m implementation. I've now moved to z2m myself, so I won't be able to implement any further updates on this.