zigpy/zha-device-handlers #2568: [Device Support Request] TS0201 _TZ3210_qkj7rujp

Nedjumiiro, 2023-09-05:
### Problem description

The device is successfully added to Home Assistant, but when you change the temperature and humidity readings in the room, the readings in Home Asistant do not change.

### Solution description

Would like the sensor work properly.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0402",
        "0x0405",
        "0xe002"
      ],
      "output_clusters": [
        "0x0004",
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_qkj7rujp",
  "model": "TS0201",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Xoc82, 2024-01-05:
I got the same device and problem, support would be great

TheJulianJES, 2024-01-05:
> but when you change the temperature and humidity readings in the room, the readings in Home Asistant do not change.

The screenshot section in the OP is missing, so can you post one to show how/if the values are shown in HA?
If they show some values at least, the sensors appear to be somewhat standard-compliant. Maybe binding and/or attribute-reporting isn't just set up correctly?

Did you try to "reconfigure" the sensor through the UI? (Although the sensor needs to be woken up to receive the messages. So it needs to have a button or something that can wake it up.)
Otherwise, does re-pairing help perhaps?

---

Regarding a quirk, the existing "ts0201" devices do not use any custom cluster for the temperature measurement cluster. They add a custom [TuyaTemperatureHumidityAlarmCluster](https://github.com/zigpy/zha-device-handlers/blob/6b9516dcf9b60326510caf223a80ec96a9d3b401/zhaquirks/tuya/ts0201.py#L34-L51) which has custom attributes that can be configured using the clusters UI (device page, three dots, Manage Zigbee device).
The `RelativeHumidity` cluster is also replaced to multiply the values sent by 10.

Looking at [zhaquirks/tuya/ts0201.py](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0201.py) (and using the stub [quirk-generator](https://github.com/zigpy/quirk-generator)), then modifying the generated `replacement` section with the two custom Tuya clusters (similar to other "ts0201" quirks), I've come up with the below quirk to test.

This very likely won't fix your problem though, as it just adds the custom Tuya cluster and multiplies new humidity values.

The below quirk can be installed as a [custom quirk](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274). On the device page (or in the device signature), you can see if the quirk applies.

<details>
<summary>Possible custom quirk (click to expand)</summary>

```python
"""Quirk for _TZ3210_qkj7rujp TS0201."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, PowerConfiguration, Time
from zigpy.zcl.clusters.measurement import RelativeHumidity, TemperatureMeasurement

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.ts0201 import (
    RelativeHumidityX10,
    TuyaTemperatureHumidityAlarmCluster,
)

class Tz3210Qkj7RujpTs0201(CustomDevice):
    """_Tz3210_Qkj7Rujp TS0201 custom device implementation."""

    signature = {
        MODELS_INFO: [("_TZ3210_qkj7rujp", "TS0201")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                    0xE002,
                ],
                OUTPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidityX10,
                    TuyaTemperatureHumidityAlarmCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

```

</details>

Xoc82, 2024-01-05:
This is how it is looking:
[image]
Sensors values are only updated directly after pairing and then never again (I pair it multiple times)
[image]
The reported values are correct, so the only problem seems to be that they are never updated again.

TheJulianJES, 2024-01-19:
Note: Perhaps https://github.com/zigpy/zha-device-handlers/pull/2888 could do the trick here too?

Xoc82, 2024-01-20:
Thanks, that really sounds like a solution. Do I have to wait until this integrated in Home Assistant or is there an easy way to try this out right now?

closed 2024-07-25 by github-actions[bot]