zigpy/zha-device-handlers #2569: [Device Support Request] TS0601 - _TZE200_p3dbf6qs - smart radiator thermostat (maybe me168?)

GeBaLeBa, 2023-09-05:
### Problem description

Tested with actual zha in home assistant. Creating own quirk based on ts0601_sas.py with _TZE200_p3dbf6qs. No success. 

Log says:
[0x875F:1:0xef00] Unknown cluster command 2 b’\x001 \x00\x00\x19\x05\x06\x00\x00\xd2\x08\x00\x00\xa0\x0c\x00\x00\xd2\x0e\x00\x00\xa0\x12\x00\x00\xd2\x16\x00\x00\xa0’
[0x875F:1:0xef00] Unknown cluster command 2 b’\x002!\x00\x00\x19\x06\x06\x00\x00\xd2\x08\x00\x00\xa0\x0c\x00\x00\xd2\x0e\x00\x00\xa0\x12\x00\x00\xd2\x16\x00\x00\xa0’
[0x875F:1:0xef00] Unknown cluster command 2 b’\x003"\x00\x00\x19\x07\x06\x00\x00\xd2\x08\x00\x00\xa0\x0c\x00\x00\xd2\x0e\x00\x00\xa0\x12\x00\x00\xd2\x16\x00\x00\xa0’
[0x875F:1:0xef00] Unknown cluster command 2 b’\x004#\x05\x00\x01\x00’
[0x875F:1:0xef00] Unknown cluster command 2 b’\x009\x05\x02\x00\x04\x00\x00\x00\xe6’

No Attributes, not usable. 

Then tested with custom quirk ts0601_sas.py with _TZE200_p3dbf6qs. Attributes of thermostat visible, but no data. No success.

### Solution description

find a quirk which works

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>
_TZE200_p3dbf6qs
```json
[Paste the diagnostic information here]
```

</details>

### Additional information

-

Raukze, 2023-09-29:
In the Zigbee2MQTT [issue](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6067) I read that it was kind of similar to _TZE200_bvu2wnxz. Thats why I tried the custom quirk from @jacekk015 named [here](https://github.com/zigpy/zha-device-handlers/issues/1818#issuecomment-1636880270) and added a line with the device signature of _TZE200_p3dbf6qs. After repairing I am now able to control my thermostat.
Maybe @jacekk015 you can add the signature to your repo.

jacekk015, 2023-10-07:
@Raukze That signature was added by PR
https://github.com/jacekk015/zha_quirks/pull/19
Thanks for testing!

dieterpapa, 2023-10-14:
@Raukze 

Could you please post the quirk with the included line? I have the similär Problem to add these TRVs to ZHA.....but may not be as sophisticated as you....

Thanks in advance

dieter

jacekk015, 2023-10-14:
@dieterpapa it's already there in my repo

Use file:
ts0601_trv_me167.py
from
https://github.com/jacekk015/zha_quirks

dieterpapa, 2023-10-15:
@jacekk015 
I've copied the above mentioned file to folder custom_zha_quirks, restartet HA, deleted the TRV in ZHA and had it reinstalled.
However, this did not work, see attached screenshot.
[image]

Any ideas?

Dieter

dieterpapa, 2023-10-16:
One more question:

Other TRVs as for example Shelly TRV have additionally to the climate entity a temperature sensor entity. This is not available on these Zigbee TRVs. Is this a matter of the firmware design or could it be implemented in the quirk?

Best regards

Dieter

jacekk015, 2023-10-16:
Could you show me an example like screenshot or some image?
Climate entity already has a room temp there.

dieterpapa, 2023-10-16:
[image]

See attached Screenshot of a Shelly TRV.

To use "Better Thermostat" in Home assistant you need a climate entity to adjust the TRV and a sensor entity to measure the teperature.

Dieter

jacekk015, 2023-10-16:
@dieterpapa Instead of messing with doubling items isn't just better do create a Helper template sensor??
State template:
`
{{state_attr('climate.tze200_qc4fpmcn_ts0601_b548f1fe_thermostat', 'current_temperature')}}
`

[image]

dieterpapa, 2023-10-17:
Thanks Jacek,

I'll try to get it done. To be honest - that's all new for me. I'm more than happy that I got my Alibaba TRVs running...step by step....and using "helpers" is now also something new for me......

Dieter

dieterpapa, 2023-10-17:
Wow - a Miracle....it is working

[image]

dieterpapa, 2023-10-24:
@jacekk015 
have to apologize...all my testing was with the older ones - which were _bvu2wnxz
Have now tried to install for the first time a _p3dbf6qs TRV. Installation is working, but the device becomes  unknown immediately.

[image]

jacekk015, 2023-10-24:
@dieterpapa Debug logs needed.
https://github.com/jacekk015/zha_quirks#in-case-of-errors-or-to-support-new-function
I will not tell you anything just from the picture.

dieterpapa, 2023-10-24:
@jacekk015 thanks for your support....I'll try to get the logs....is again something new for me :-)

jacekk015, 2023-10-24:
Nothing wrong with the quirk - ta least for now and what I see in the logs:
HA found the quirk for TRV and want to use it
```
2023-10-24 09:23:37.664 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_p3dbf6qs TS0601 (a4:c1:38:ef:6a:4d:cc:ae)
2023-10-24 09:23:37.664 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601_trv_me167.ME167'>
2023-10-24 09:23:37.664 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for a4:c1:38:ef:6a:4d:cc:ae: <class 'ts0601_trv_me167.ME167'>
```

Then HA says there's no connection with TRV - last seen over 12 hours ago
```
2023-10-24 09:23:47.894 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] [0x9AF3](_TZE200_p3dbf6qs TS0601) restored as 'unavailable', last seen: 12:34:17 ago, consider_unavailable_time: 21600 seconds
```

Since TRV doesn't respond HA marks it unavailable
```
2023-10-24 09:27:41.898 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x9AF3](TS0601): last_seen is 45491.379120111465 seconds ago and ping attempts have been exhausted, marking the device unavailable
2023-10-24 09:27:41.898 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x9AF3](TS0601): Update device availability -  device available: False - new availability: False - changed: False
```

Look like TRV doesn't have a Zigbee range or something like that.
It doesn't talk at all with HA
Even with no quirk there should be communication between HA and TRV.
Here's nothing.

jacekk015, 2023-10-24:
It's not loosing connection. There's always a delay when the HA marks it unavailable.
It try few times before marking.
Like you see form the logs. Device last seen 12 hours ago.

Bring it closer to coordinator antenna and check again. Or remove it and pair like a new.
Maybe TRV didn't bind to HA properly.
For now there's no Zigbee connection between HA and TRV. They don't talk.

MattWestb, 2023-10-24:
@dieterpapa Update you coordinator firmware to latest release then both TI and EZSP have some bugs with direct children. 
Also make sure having one good router around the TRVs then its sleeping end devices (not deep sleeper but is offline most of the time) and if its parent is not OK its going permanent offline then it have lost the contact with its parent.

dieterpapa, 2023-10-24:
See attached photos, the device paired correctly 
[image]
[image]
[image]

I am about 3m away from the ZHA antenna and there are plenty of router-devices installed.

However, I'll check my coordinator firmware - I have a ezsp stick. 

Dieter

jacekk015, 2023-10-24:
I would like to get HA logs from that pairing process.
So:
Remove TRV from HA.
Restart HA - to get a smaller log file.
Pair like new
Wait 10 minutes
Post HA full logs.

dieterpapa, 2023-10-24:
I'm now afraid a little bit....
I've had now serious problems with my heating system at all - all TRVs grouped via climate group gone crazy (3 Shelly TRV, one _bvu2wnxz). TRVs couldn't been adjusted, did change temperatures themselves....crazy behaviour. Had it yesterday and thought it may be HA 10.4 or 10.5 - did restore 10.3 then.
Today the same problem after playing around with the TRVs p3dbf6qs....went back to a backup made yesterday without one of those TRVs in the system - it is running again.
I'll watch my heating system first for a few days and once I have a stable system again I'll come back. 

Dieter

tomastyl, 2023-10-26:
Dear Jacek, thx for saving my money with your qurk file....is there any chance to get control over % of valve opening (e.g 0-100%)...I am looking fo this control: 
[image]

Thanks 
Tom

tomastyl, 2023-10-26:
ok, thanks anyway, so any tips on TRV tested in ZHA with that feature?

jacekk015, 2023-10-26:
@tomastyl From the quirks code I see that Moes an Rtitek have that attribute.
Can't give you 100% for that since those quirks are old and tested long ago.
I don't remember every TRV id functionality.

Question is what you wanna to achieve?
Even if you send 10% opening I'm still not sure if the TRV wan't change it in the next 5 minutes.

tomastyl, 2023-10-26:
I want to control % of valve opening based on other sensors (temperature, windows) etc. Basicaly TRV would be just motors - no need for their own termosats....

jacekk015, 2023-10-26:
Probably not possible. Are you sure that TRV won't override this after few minutes? Even in the manual mode.
If I remember correctly there were some guys that wanted to do this, but there was probably like only one TRV that could be really steered.

Remember that most of that devices have PID implemented.
LIDL TRV's I have aren't stupid devices like full open/close. They just move the engine a bit.
I can check valve open percentage, but only on TRV - they don't send it by Zigbee.

tomastyl, 2023-10-26:
ok, so it might be bad idea from the beggining :-) 
I am a bit new in this :
in automation I can see trigger HVAC changed to "HEAT" but in history of device I see status "HEATING"  - those two do not match - could it be cause my next steps aren't triggering?

jacekk015, 2023-10-26:
HVAC mode is probably a system_mode - I'm guessing what you see now.
We emulate TRV as a HVAC.
Real HVAC is an air conditioner. It can heat/cool/fan/dehydrate etc...
TRV can only HEAT or sometimes get OFF system mode. For sure it won't Cool.

Next is State: Heating or Idle.
Those two aren't the same.

Think of it like - I've turned AC to Heat system_mode, but if the temp is over degrees set, the AC goes from Heat state to Idle state and waits for the temperature to drop below temperature set.
But still AC got Heat system_mode

tomastyl, 2023-10-26:
ok, thanks.... 👍

TomcatPi, 2023-11-22:
Thanks for the customs quirks
it works but i've got a question

What are the signification of the 2 switchs on the TRV ?
I found that the 1st is the child lock, but I was unable to undestand what is the second one ?

Also i've got another parameter in °C that is at 0°C and can move frome -10°C to +10°C, what is for ?

[image]

jacekk015, 2023-11-22:
Switches are ChildLock and FrostProtection
Next one is a Temperature Calibration slider.

TomcatPi, 2023-11-22:
> Switches are ChildLock and FrostProtection Next one is a Temperature Calibration slider.

Thanks for the reply
I'm not sure to understand what the temperature calibration slider do exactly ?
could you explain to me ?

jacekk015, 2023-11-22:
Same as TRV manual says.
Calibrates room temp reading.

nicGoSan, 2023-11-30:
I have also issues with newly bought device ( AVATTO Tuya ZigBee3.0, tze200_p3dbf6qs_ts0601). Connection works with the quirk. I added an issue to your @jacekk015 github: [https://github.com/jacekk015/zha_quirks]. I have many other devices in ZHA and don't want to switch to Zigbee2MQTT .Your effort is much appreciated.

dredkin, 2024-01-19:
Hello @jacekk015 !
I just bought 6 TRVs with this identifier and I installed your quirk into the HA 2023.7.3.
It is working fine, so fist of all, thank you very much!

But the additional switches (like child lock and temperature calibration) were imported with gibberish names so now I have to manually rename all of them. It's so much pain.
Is there any way I can edit this quirk so these switches get their names and icons automatically?

jacekk015, 2024-01-19:
@dredkin That's unfortunately limitation of ZHA

dredkin, 2024-01-19:
I see. Such a pity. But thanks again anyway for your priceless work!

dolorosus, 2024-01-26:
I have noticed the attribute ME167_TEMP_CALIBRATION_ATTR is assumed to have the unit _decigree_:
```ME167_TEMP_CALIBRATION_ATTR = 0x022F  # (decidegree)```

However, as far as I can determine, the unit of this attribute is _degree_.  As soon as you try advanced thermostat controls (e.g. better thermostat, ...), the entire temperature control gets confused due to incorrect calibration values.  

Can someone take a look at this and correct it if necessary?

closed 2024-08-02 by github-actions[bot]

michalw2-alt, 2024-10-06:
I've noticed that whenever I try to change: "Min heat setpoint limit" or "Max heat setpoint limit" I get error: 
"Failed to perform the action number/set_value. 'NoneType' object is not iterable"
[image]

I'm wondering if this is known issue?

jneuhauser, 2025-02-13:
You do not need to use this custom quirk anymore....

Official support was added with https://github.com/zigpy/zha-device-handlers/commit/e61ecbb4f64291c663fd19d23f95f7db5d33e1f0