zigpy/zha-device-handlers #2572: [Device Support Request] MCCGQ12LM

Infrecon, 2023-09-06:
### Problem description

I want to connect my aqara MCCGQ12LM with a sky connect and zha

### Solution description

connection established

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2023-09-06:
Perhaps one Device signature and little Diagnostic information can helping more then only "I want" . . . .

Logs can also helping !!

TheJulianJES, 2023-09-17:
So you have issues connecting the sensor at all? Hold the reset button for longer while allowing joins in ZHA ("Add Device" button).

closed 2023-11-23 by github-actions[bot]