zigpy/zha-device-handlers #2573: [Device Support Request] TS011F Smart Plug (_TZ3000_cehuw1lw)

ozhound, 2023-09-07:
### Problem description

I have a frustrating issue with this power plug. The first lot of 5 that I installed worked great and the power measurement sensors work accurately and consitently

So 2 weeks later i purchased another 5 from the same seller (sixwgh store on Aliexpress) and when they came, they look and function identically. when installed in HA however, they are missing the Child Lock and the Backlight Settings, and the amperage is approximately 100 times what it is supposed to be (220 AMPS for a fridge?) power draw and Voltage seems accurate.

The difference i can see is that the new plugs mfg code is different and it isnt being picked up correctly by ZHA and the class is "zigpy.device.Device"

### Solution description

For it to work the same way as the existing TS011F Plugs that look indentical

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

Existing (working correctly) Plug
[image]

New - Partially Functioning Plug
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>
Partly-working Device Sig
```json
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4107, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0702",
        "0x0b04",
        "0x1000",
        "0x1888",
        "0xe000"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_0zfrhq4i",
  "model": "TS011F",
  "class": "zigpy.device.Device"
```
Correct Signature
```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_cehuw1lw",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
```
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

erkr, 2023-09-10:
@TheJulianJES 
I have a device with a totally different model info ("_TZ3000_xkap8wtb"), but it has exactly the same zigbee signature and the same issue of 100x higher current. 
I wrote a custom quirk for my device that works:

https://gist.github.com/erkr/5d4addf1b4c8e18d68d2adb154c6be93

I wonder if that quirk also works for this device, and if so can be added to the official distribution.

Best Eric

ozhound, 2023-09-13:
when is it likely this will be resolved? Ive just got 5 of these sitting here doing nothing

ozhound, 2023-09-14:
> I guess It will take a while to resolve this. Someone needs to test new
> code before it will included in the release.
>
> Why don't you try my custom quirk? Comment out the model part (regering my
> plug) and uncomment models_info set with your device info. It will help the
> developers to know if that works. As the zigbee signature is equal, that
> would not surprise me.
> Success Eric

i have no idea how to implement quirk.

erkr, 2023-09-16:
Step 1 is to enable custom quirks in your configuration.yaml:
https://www.home-assistant.io/integrations/zha#configuration---yaml

Step 2 is to copy the custom quirk file inthe configured folder

Step 3 is restart HA and check your log for errors.

If no errors, check in the zigbee info of the device is the quirk is loaded

TheJulianJES, 2023-09-17:
Please try this custom quirk:

<details>
<summary>Possible custom quirk to try</summary>

```python
"""TS011F plug."""

from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringClusterWithUnit,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class Plug_v3(EnchantedDevice):
    """Tuya TS011F plug. One plug is _Tz3000_0Zfrhq4I."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    LightLink.cluster_id,
                    0x1888,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    Time.cluster_id,
                    TuyaZBMeteringClusterWithUnit,
                    TuyaZBElectricalMeasurement,
                    LightLink.cluster_id,
                    0x1888,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }
```

</details>

TheJulianJES, 2023-09-17:
> when is it likely this will be resolved? Ive just got 5 of these sitting here doing nothing

Also note that this is all/mostly community-maintained. At least I'm doing this in my free time. So it's always possible that your devices may never be supported.
Tuya is notorious for explicitly making it hard to have their devices work on third-party hubs. If they followed Zigbee spec, it would *just work* without any quirks.

If you need help installing the custom quirk I've posted above, let me know.
(Basic steps are: put the content in a file called something like `tuya_plug.py`, put that file in your custom quirks directory that should be located in the `/config` directory of your HA installation. Then, make sure the custom quirk directory is set in your HA `configuration.yaml`, restart HA after that and see if the quirk applies to your device.
If everything works, let me know and we'll merge the quirk for the next major HA release. You should remove the custom quirk by then.)

ozhound, 2023-09-17:
Thankyou I will try it today. I've followed guides for creating quirks but I'm not strong on programming at all and whilst I get the general structure of the Jason file, the specifics just hurt my head. I appreciate this is community run and thank all that contribute.
  

  
Regards
  

  
Ozhound

ozhound, 2023-09-17:
@TheJulianJES 

I tried that quirk and the LOGS say that its loaded custom quirks but it isn't applied to that device.

[image]

Is there a reason we cant take an inbuilt quirk , create a custom quirk, changing the model to make it work? I have 5 or these exact same units that work beautifully and have a quirk applied to it automatically. These last 5, bought from the same supplied 2 weeks later have the different model code but are the same unit?

This is the working unit showing the quirk being applied by ZHA automatically

[image]

Thanks again for responding initially.

TheJulianJES, 2023-09-17:
I've updated my [above comment](https://github.com/zigpy/zha-device-handlers/issues/2573#issuecomment-1722370097). Please try the updated quirk.

> Is there a reason we cant take an inbuilt quirk , create a custom quirk, changing the model to make it work?

Yes, it's not the model that is different. The exposed clusters are different. It's basically an entirely different plug (under the hood) (or at least running different firmware).

ozhound, 2023-09-17:
@TheJulianJES Brilliant, that worked!. Power delivered looks right, the other sensors seem to have registered properly. Ill do some more testing by it looks like its working.

Thankyou so much.

henklietaert, 2023-09-17:
I can confirm that this [custom quirk](https://github.com/zigpy/zha-device-handlers/issues/2573#issuecomment-1722370097) mentioned earlier works perfectly, so should be integrated in the main code.
My plugs are the following: _TZ3000_0zfrhq4i TS011F

TheJulianJES, 2023-09-17:
Thanks for testing!

This PR adds support for the plug: https://github.com/zigpy/zha-device-handlers/pull/2592
It should be available in HA 2023.10.0

closed 2023-09-24 by TheJulianJES

phil-lipp, 2023-10-19:
Following https://github.com/home-assistant/core/issues/98624#issuecomment-1688222346 fixed this issue for me

alka79, 2024-02-28:
Hello. I have this power plug from another manufacturer : _TZ3000_okaz9tjs
I can turn it on and off, all the sensors are present, but :
-  the control is a light instead of a switch.
-  the measured current is deliriously high!

I don't mind the wrong current measurement, but is is annoying to be forced to use it as a light.

I tried the custom quirk linked above: no change.

<details><summary>Extract of the device diagnostics </summary>

```
 "data": {
    "ieee": "**REDACTED**",
    "nwk": 54310,
    "manufacturer": "_TZ3000_okaz9tjs",
    "model": "TS011F",
    "name": "_TZ3000_okaz9tjs TS011F",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4660,
    "power_source": "Mains",
    "lqi": 200,
    "rssi": -50,
    "last_seen": "2024-02-28T22:06:16",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x000a",
            "0x0702",
            "0x0b04",
            "0x1000",
            "0xe000",
            "0xe001"
          ],
          "output_clusters": []
        }
      },
      "manufacturer": "_TZ3000_okaz9tjs",
      "model": "TS011F"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "button.tz3000_okaz9tjs_ts011f_identifier",
        "name": "_TZ3000_okaz9tjs TS011F"
      },
      {
        "entity_id": "light.tz3000_okaz9tjs_ts011f_lumiere",
        "name": "_TZ3000_okaz9tjs TS011F"
      },
      {
        "entity_id": "select.tz3000_okaz9tjs_ts011f_comportement_au_demarrage",
        "name": "_TZ3000_okaz9tjs TS011F"
      },
      {
        "entity_id": "sensor.tz3000_okaz9tjs_ts011f_courant",
        "name": "_TZ3000_okaz9tjs TS011F"
      },
      {
        "entity_id": "sensor.tz3000_okaz9tjs_ts011f_tension",
        "name": "_TZ3000_okaz9tjs TS011F"
      },
      {
        "entity_id": "sensor.tz3000_okaz9tjs_ts011f_puissance",
        "name": "_TZ3000_okaz9tjs TS011F"
      },
      {
        "entity_id": "sensor.tz3000_okaz9tjs_ts011f_summation_delivered",
        "name": "_TZ3000_okaz9tjs TS011F"
      }
```

</details>

Anything else I could try ?

erkr, 2024-02-28:
@alka79 
I have several of these plugs and they work fine without custom quirk. On my system they use a released quirk `zhaquirks.tuya.ts011f_plug.Plug_v2`:
[image]

alka79, 2024-02-29:
Thanks. So there is hope :)

Whatever I do, it is always a light.
The quirk line is not displayed in the UI for this device. When looking in the diagnostics, I always see that no quirk was applied.
I also tried with no success the custom quirk you published here: https://gist.github.com/erkr/5d4addf1b4c8e18d68d2adb154c6be93

My plug must be different. I could see that the released quirk for TS011F has already 3 versions ! this might be the 4th. 

I have to figure out what the differences are and I don't really know where to look.

erkr, 2024-02-29:
I have exactly the same brand and firmware. Try to remove them. Restart en add them again

alka79, 2024-02-29:
Unbeliveable ! 
I tried twice already without custom quirk and also with the custom quirk you published and with the one publisher here by TheJulianJES. I'll try again later.
Maybe some traces are left from my previous tries. Is there something else I should do to clean other than remove the device in HA ?

alka79, 2024-03-01:
I tried again with no success. I started the log before pairing.

As I understand, your plug matches with the tuya.ts011f_plug.Plug_v2 quirk.

For my plug, the matching attempts fail with all the ts011f quirks, mostly "because endpoint list mismatch". 
For Plug_v2 the reason is a bit different : 
```
2024-02-29 21:15:57.353 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.tuya.ts011f_plug.Plug_v2'>	
2024-02-29 21:15:57.353 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
```

Attached is the full diagnostics of my plug. Maybe you could share the full diagnostics of one of yours matching the Plug_v2 quirk so that I can compare and possibly write a custom quirk based on Plug_v2 that would match my plug ?

[zha-5a0e0ab5b7ac4ee6cc62b84332364158-_TZ3000_okaz9tjs TS011F-8613104db9ceca9d04729a7b527ee8fa(13).json](https://github.com/zigpy/zha-device-handlers/files/14464236/zha-5a0e0ab5b7ac4ee6cc62b84332364158-_TZ3000_okaz9tjs.TS011F-8613104db9ceca9d04729a7b527ee8fa.13.json)

alka79, 2024-03-01:
I found a work around mentioned in the doc ! https://www.home-assistant.io/integrations/zha/#modifying-the-device-type/   
RTFM ;)
```yml
zha:
  device_config:
    84:71:27:ff:fe:93:17:24-1:    # format: {ieee}-{endpoint_id}
      type: "switch"              # corrected device type
```
The current metering is still wrong but at least it is recognized as a switch. Good enough for me.

erkr, 2024-03-01:
I'm running out of suggestions. Sorry

alka79, 2024-03-03:
the fully working quirk for my devices is here https://github.com/zigpy/zha-device-handlers/issues/1764#issuecomment-1946842958

aqxa1, 2024-06-23:
I seem to have a different variant of cehuw1lw which is missing the 242 endpoint. I've only tested it briefly but all looks correct with current and summation power returning sane values. Here's the code:

```python
"""TS011F plug."""

from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBMeteringClusterWithUnit,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class Plug_v3(EnchantedDevice):
    """Tuya TS011F plug. One plug is _Tz3000_0Zfrhq4I."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    LightLink.cluster_id,
                    0x1888,
                    TuyaZBE000Cluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    Time.cluster_id,
                    TuyaZBMeteringClusterWithUnit,
                    TuyaZBElectricalMeasurement,
                    LightLink.cluster_id,
                    0x1888,
                    TuyaZBE000Cluster,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }
```

ge33ek, 2024-08-22:
I’ve got this unit, but none of the quirks above seem to work, amp is the issue, reporting incorrectly as 1000x that is 84amps instead of 0.84

`1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
33
34
35
36
37
38
39
40
41
42
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4107,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0702",
        "0x0b04",
        "0x1000",
        "0x1888",
        "0xe000"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_cehuw1lw",
  "model": "TS011F",
  "class": "zigpy.device.Device"
} `

Anyone got suggestions?