zigpy/zha-device-handlers #2576: [Device Support Request] Third Reality button 3RSB22BZ

htstahl, 2023-09-08:
### Problem description

Third Reality Button shows up as Third Reality contact sensor.

Expected device: 3RSB22BZ

Device ZHA finds: 3RDS17BZ

### Solution description

Correctly detect and pair 3RSB22BZ to ZHA and have the single, double, and long press commands available

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4659, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0500"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Third Reality, Inc",
  "model": "3RDS17BZ",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheJulianJES, 2023-09-17:
When you listen to `zha_event` in the HA developer tools, do you see any events?

htstahl, 2023-09-17:
I do not, no. I think because it registers as a contact sensor it does not process a button press as an event

closed 2024-03-22 by github-actions[bot]